RIOT-OS/RIOT #11611: ifconfig has problem with stm32f4discovery

jhx2002, 2019-05-30:
setup:

- connect stm32f407vet6 and  LAN8720,   connect to ethernet

- `cd RIOT/examples/emcute_mqttsn`

- `make BOARD=stm32f4discovery all flash term`

- ifconfig , no ethernet interface info , see log: 
```
> help
2019-05-30 17:23:32,578 - INFO #  help
2019-05-30 17:23:32,581 - INFO # Command              Description
2019-05-30 17:23:32,584 - INFO # ---------------------------------------
2019-05-30 17:23:32,587 - INFO # con                  connect to MQTT broker
2019-05-30 17:23:32,592 - INFO # discon               disconnect from the current broker
2019-05-30 17:23:32,595 - INFO # pub                  publish something
2019-05-30 17:23:32,598 - INFO # sub                  subscribe topic
2019-05-30 17:23:32,603 - INFO # unsub                unsubscribe from topic
2019-05-30 17:23:32,606 - INFO # will                 register a last will
2019-05-30 17:23:32,609 - INFO # reboot               Reboot the node
2019-05-30 17:23:32,615 - INFO # ps                   Prints information about running threads.
2019-05-30 17:23:32,617 - INFO # ping6                Ping via ICMPv6
2019-05-30 17:23:32,623 - INFO # random_init          initializes the PRNG
2019-05-30 17:23:32,628 - INFO # random_get           returns 32 bit of pseudo randomness
2019-05-30 17:23:32,632 - INFO # nib                  Configure neighbor information base
2019-05-30 17:23:32,637 - INFO # ifconfig             Configure network interfaces
> ifconfig
2019-05-30 17:23:36,554 - INFO #  ifconfig
> 

```

PeterKietzmann, 2019-06-03:
1. If you want to use an externally connected device with the discovery board, you need to incorporate that module in your Makefile (`USEMOUDLE+=...`) and add the respective pin configuration.
2. If I see it correctly we don't have a  LAN8720 driver in RIOT. Do you have access to a enc28j60, encx24j600 or w5100 device?

jhx2002, 2019-06-13:
could i use enc28j60-spi module + stm32f4  ?

jhx2002, 2019-06-15:
I test stm32f4 + enc28j60 with  https://github.com/RIOT-OS/RIOT/tree/master/tests/driver_enc28j60,   got following error: 
```
Type '/exit' to exit.
2019-06-15 19:34:09,240 - INFO # main(): Thisgnrc_netif: netdev init failed: -1
2019-06-15 19:34:09,242 - INFO #  etdev init failed: -1
2019-06-15 19:34:09,243 - INFO # 9.04)
2019-06-15 19:34:09,246 - INFO # Test application for the enc28j60 driver
2019-06-15 19:34:09,246 - INFO # 
2019-06-15 19:34:09,251 - INFO # This test just pulls in parts of the GNRC network stack, use the
2019-06-15 19:34:09,258 - INFO # provided shell commands (i.e. ifconfig, ping6) to interact with
2019-06-15 19:34:09,261 - INFO # your enc28j60 device.
2019-06-15 19:34:09,262 - INFO # 
2019-06-15 19:34:09,262 - INFO # Starting the shell now...
> 
```

closed 2019-06-15 by jhx2002