RIOT-OS/RIOT #11613: uart_init seems not return

jhx2002, 2019-06-02:
- code : 

```

#include <stdio.h>
#include "periph/uart.h"

void uart_rx_cb(void *arg, uint8_t data) {
    int i = 0;
    if (arg == NULL) {
	i += 0;
    }
    //printf("uart data: %d", data);                                                                                                                                                                               
    data += 1;

    return; //TODO                                                                                                                                                                                                 
}

int main(void)
{
    puts("Hello World!");

    printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
    printf("This board features a(n) %s MCU.\n", RIOT_MCU);

    uart_t uart_id = 1;
    uint32_t baudrate = 115200;
    uint32_t key = 111;
    int ret = uart_init(uart_id, baudrate, uart_rx_cb, (void *)&key);
    if (ret != UART_OK) {
        printf("Init uart  %d , return error code %d", uart_id, ret);
    }
    else {
        printf("Init uart  %d", uart_id);
    }

    return 0;
}

```

- uart_init seems not return.  running log is : 

```
/home/huaxing/oproj/riot/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB0" -b "115200"
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-06-02 14:00:40,019 - INFO # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type '/exit' to exit.
2019-06-02 14:00:42,755 - INFO # main(): This is RIOT! (Version: 2019.04)
2019-06-02 14:00:42,756 - INFO # Hello World!
2019-06-02 14:00:42,760 - INFO # You are running RIOT on a(n) stm32f407vet6 board.
2019-06-02 14:00:42,764 - INFO # This board features a(n) stm32f407 MCU.

```

PeterKietzmann, 2019-06-03:
@jhx2002 please be a bit more precise in your description, i.e., use the issue form provided when opening the issue:

```
#### Description
<!--
Example: Cannot build gnrc_networking application for samr21-xpro board.
-->

#### Steps to reproduce the issue
<!--
Try to describe as precisely as possible here the steps required to reproduce
the issue. Here you can also describe your hardware configuration, the network
setup, etc.
-->

#### Expected results
<!--
Example: The gnrc_networking application builds on samr21-xpro.
-->

#### Actual results
<!--
Please paste or specifically describe the actual output.
-->

#### Versions
<!--
Operating system: Mac OSX, Linux, Vagrant VM
Build environment: GCC, CLang versions (you can run the following command from
the RIOT base directory: make print-versions).
-->
```

yegorich, 2019-06-03:
Have you tried to run the standard UART test app?

`make BOARD=stm32f407vet6 -C tests/periph_uart flash`

Once inside the shell try to initialize the first UART

`init 1 115200`

dylad, 2019-06-03:
@jhx2002 
Could you also share your periph_conf.h file to ensure configuration of second UART is correct please ?

jhx2002, 2019-06-06:
tests/periph_uart  is OK.    I'v solved this.  Thank you all !

closed 2019-06-06 by jhx2002

kaspar030, 2019-06-06:
>  I'v solved this.

Could you give a hint for future readers how you solved it?