RIOT-OS/RIOT #11644: MQTT-SN

cooldwq, 2019-06-06:
Hello Everyone, i am new here, hope to get Support.

1. .When I Subscribe[sub Topic] to a Topic from a RIOT Node, I don't get an Output same as [Publishing Topic] in the "make term" -Terminal, i have only 'Now subscribed to Topic''. But I noticed via Wireshark that a communication with the RIOT node has taken place, am i doing something wrong, or are there other ways to get the 'Topic' by Sub?  Please see Window 4.
[image]
 
[image]

2. If I want to create a connection to the MQTT broker (middle con [IPv6]) from 2 different RIOT Nodes, the MQTT-SN Broker - RSMB tells me when trying to connect the 2nd riot node with that there is already a connection with the client ( "duplicate connection from identifier gertrud") and disconnects the first connection, the second riot node is then connected to the broker, the first no longer there. I wanted to ask you if this behavior is known to you and you know a solution, or if this is more of a mistake with the broker.

aabadie, 2019-06-26:
> I wanted to ask you if this behavior is known to you and you know a solution, or if this is more of a mistake with the broker.

To connect multiple RIOT instances to the same RSMB broker, you need to use different clients ID. This can be done by editing the `EMCUTE_ID`, [here](https://github.com/RIOT-OS/RIOT/blob/master/examples/emcute_mqttsn/main.c#L32).

I don't understand your first question.

cooldwq, 2019-06-26:
I found this .c, is the current ID `gertrud`? Do I need to add a new ID? How can I modify this file? By executing which command I can connect the second RIOT instance to the same RSMB broker? By now I am using `make term` under `~/RIOT/examples/emcute_mqttsn$ `.

For the second question I mean how to use the shell command `sub`. As shown in "Fenster 4", after `sub heute`, i got only `Now subscribed to heute`, that's all? Can i get the subscribed" content of this topic? Or   from where can i subscribe the content with the same topic, the topic here is "heute"?

PeterKietzmann, 2019-07-16:
I just found #11747  which seem to address at least one of your question.

cooldwq, 2019-07-16:
What I mean for the first question maybe is, if I have multiple nodes connected to the broker at the same time, how can I on a Node subscribe the Message, which is come from another node?

closed 2020-03-02 by stale[bot]