RIOT-OS/RIOT #11669: Some problem about how to capture how many times high voltage appeared of a PWM wave in a certain time?

zpw199062, 2019-06-11:
#### Description
I tried to know how many times that high voltage appeared of a PWM in a certain time, so i modified one of the tests: periph_gpio. Like below.
static int read(int argc, char **argv)
{
    int port, pin;
    int flag=0;
    int lastflag=0;
    int cnt=0;
    if (argc < 3) {
        printf("usage: %s <port> <pin>\n", argv[0]);
        return 1;
    }

    port = atoi(argv[1]);
    pin = atoi(argv[2]);
   while(1)
   {
   if (gpio_read(GPIO_PIN(port, pin)))
   {                      
                          flag=1;
			 
			 // printf("cnt:%d\r\n",cnt);
        //printf("GPIO_PIN(%i.%02i) is HIGH\n", port, pin);
    }
    else {
        flag=0;
	// printf("cnt:%d\r\n",cnt);
        //printf("GPIO_PIN(%i.%02i) is LOW\n", port, pin);
    }
    
    if (lastflag==0&&flag==1)
    {
        cnt+=1;
    }
    lastflag=flag;
    printf("cnt:%d\r\n",cnt);   
    // xtimer_usleep(10 * US_PER_MS);  
   } 
    return 0;
}

I just wonder why the cnt number is not correct, i connect PA6 and PE9, which PE9 is set 1000HZ,500DutyCycle PWM, and the cnt is not correct, it didn't show 1000 increase in 1 second. but about 200 maybe.

What's the problem with this issue, can anybody help?！ Thank you so much！
### Useful links

PeterKietzmann, 2019-07-17:
@zpw199062 did try my proposal based on counting interrupts?

PeterKietzmann, 2019-08-19:
@zpw199062 ping

PeterKietzmann, 2019-08-20:
Author not responsive anymore and question was answered so I'm closing this one.

closed 2019-08-20 by PeterKietzmann