RIOT-OS/RIOT #11674: boards/cpu: model of 'common' directories

cladmi, 2019-06-11:
#### Description

Boards based on a `common` board sometime only use `Makefile.include` but not `Makefile.dep`, or do not process `Makefile.features` even if it exist.
Sometime a common `Makefile.dep` does not exist at the beginning but may be created in the future and not all boards get updated.
Some boards do not use the `board_common_abcd` module implemented.

This leads to asking is this really in common ?

Because of this, the current model for `common` boards and `common` cpu is not completely clear as some board could only partially use a common board for good or bad reasons.

#### Questions

##### Makefile including files

Is it something we would like to have, that if a board/cup uses any of the `Makefile.include|Makefile.dep|Makefile.features` it must also process the other files (if they exist) ?

If a board/cpu was always processing all files, we think of defining a board as

```
    board
       |-> board/common/1
       |-> board/common/2
           |-> board/common/3
       |-> cpu
           |-> cpu/common/1
               |-> cpu/common/2
```

If then this hierarchy can be defined recursively and be consistent, it could remove the need to re-include and add to `DIRS` all the common files in each `Makefile|Makefile.dep|Makefile.features|Makefile.include` and leave this to central files.

Also easily give "by which parts is my board/cpu implemented".
It could currently be deduced from `MAKEFILE_LIST` but would not mean all `features` or `dependencies` are used.

Some of the files not using `Makefile.features` were found during https://github.com/RIOT-OS/RIOT/pull/11477

* [ ] `common/msba2`, `common/remote`, `common/msb-430` not using a common `Makefile.features`. Which was problematic when moving `CPU/CPU_MODEL` to it. TODO PR, currently in #11477 
*

**TODO**:

Find a script that really shows if a board/cpu does not use all the files from the common. And so ignores the non existing common files.

```
git grep -l 'common/.*/Makefile.' boards/  | sed 's/Makefile.*//' | uniq -c | grep -v '3 '
      2 boards/arduino-leonardo/
      2 boards/avsextrem/
      2 boards/calliope-mini/
      1 boards/common/arduino-atmega/
      1 boards/common/kw41z/
      1 boards/common/nucleo144/
      1 boards/common/nucleo32/
      1 boards/common/nucleo64/
      2 boards/firefly/
      1 boards/frdm-k22f/
      1 boards/frdm-k64f/
      2 boards/jiminy-mega256rfr2/
      1 boards/mega-xplained/
      1 boards/msb-430/
      1 boards/msb-430h/
      2 boards/msba2/
      2 boards/nucleo-f410rb/
      2 boards/nucleo-f411re/
      2 boards/remote-pa/
      2 boards/remote-reva/
      2 boards/remote-revb/
      1 boards/waspmote-pro/
      2 boards/wsn430-v1_3b/
      2 boards/wsn430-v1_4/
```

A more verbose output that does not merge including two different common files, but even more false positive.

`git grep  'common/.*/Makefile.' boards/  | sed 's/Makefile.[a-z]*/Makefile.xxx/g' | uniq -c | sed 's/Makefile..*//'  | grep -v '3 '`

##### Features

##### Always defining each file

Question, should we move toward all the boards, cpu and commons always define `Makefile.dep/Makefile.features/Makefile.include` ?

My way of finding them was to look in the output of `git grep -l 'common/.*/Makefile.' boards/` and find inconsistencies. I do not have a magic correct script for it now.

A way at least it to find places where a board using a `common` includes it less than 3 times.

But it is not really accurate as it removes cases where a common does not have a `Makefile.dep` for example.

<details><summary><code>git grep  'common/.*/Makefile.' boards/  | sed 's/Makefile.[a-z]*/Makefile.xxx/g' | uniq -c | sed 's/Makefile..*//'  | grep -v '3 '</code></summary><p>

```

      1 boards/acd52832/
      2 boards/acd52832/
      2 boards/arduino-leonardo/
      2 boards/calliope-mini/
      1 boards/common/arduino-atmega/
      1 boards/common/kw41z/
      1 boards/common/nucleo144/
      1 boards/common/nucleo32/
      1 boards/common/nucleo64/
      1 boards/frdm-k22f/
      1 boards/frdm-k64f/
      1 boards/jiminy-mega256rfr2/
      1 boards/jiminy-mega256rfr2/
      1 boards/jiminy-mega256rfr2/
      1 boards/mega-xplained/
      2 boards/msb-430/
      2 boards/msb-430h/
      1 boards/nucleo-f030r8/
      2 boards/nucleo-f030r8/
      1 boards/nucleo-f031k6/
      2 boards/nucleo-f031k6/
      1 boards/nucleo-f042k6/
      2 boards/nucleo-f042k6/
      1 boards/nucleo-f070rb/
      2 boards/nucleo-f070rb/
      1 boards/nucleo-f072rb/
      2 boards/nucleo-f072rb/
      1 boards/nucleo-f091rc/
      2 boards/nucleo-f091rc/
      1 boards/nucleo-f103rb/
      2 boards/nucleo-f103rb/
      1 boards/nucleo-f207zg/
      2 boards/nucleo-f207zg/
      1 boards/nucleo-f302r8/
      2 boards/nucleo-f302r8/
      1 boards/nucleo-f303k8/
      2 boards/nucleo-f303k8/
      1 boards/nucleo-f303re/
      2 boards/nucleo-f303re/
      1 boards/nucleo-f303ze/
      2 boards/nucleo-f303ze/
      1 boards/nucleo-f334r8/
      2 boards/nucleo-f334r8/
      1 boards/nucleo-f401re/
      2 boards/nucleo-f401re/
      1 boards/nucleo-f410rb/
      1 boards/nucleo-f410rb/
      1 boards/nucleo-f411re/
      1 boards/nucleo-f411re/
      1 boards/nucleo-f412zg/
      2 boards/nucleo-f412zg/
      1 boards/nucleo-f413zh/
      2 boards/nucleo-f413zh/
      1 boards/nucleo-f429zi/
      2 boards/nucleo-f429zi/
      1 boards/nucleo-f446re/
      2 boards/nucleo-f446re/
      1 boards/nucleo-f446ze/
      2 boards/nucleo-f446ze/
      1 boards/nucleo-f722ze/
      2 boards/nucleo-f722ze/
      1 boards/nucleo-f746zg/
      2 boards/nucleo-f746zg/
      1 boards/nucleo-f767zi/
      2 boards/nucleo-f767zi/
      1 boards/nucleo-l031k6/
      2 boards/nucleo-l031k6/
      1 boards/nucleo-l053r8/
      2 boards/nucleo-l053r8/
      1 boards/nucleo-l073rz/
      2 boards/nucleo-l073rz/
      1 boards/nucleo-l152re/
      2 boards/nucleo-l152re/
      1 boards/nucleo-l432kc/
      2 boards/nucleo-l432kc/
      1 boards/nucleo-l433rc/
      2 boards/nucleo-l433rc/
      1 boards/nucleo-l452re/
      2 boards/nucleo-l452re/
      1 boards/nucleo-l476rg/
      2 boards/nucleo-l476rg/
      1 boards/nucleo-l496zg/
      2 boards/nucleo-l496zg/
      2 boards/ruuvitag/
      1 boards/ruuvitag/
      1 boards/waspmote-pro/
      2 boards/wsn430-v1_3b/
      2 boards/wsn430-v1_4/
```

</p></details>

##### common code without function namespacing

https://github.com/RIOT-OS/RIOT/wiki/Coding-conventions#naming

Some boards/common define `board_init` directly, and have `board.h` and `periph_conf.h` headers which leads to the board not being able to extend a common board without duplicating the files.

Where having the information in only one place updates all the boards at once.

##### Finding usages

#### Tracking updates

* [x] phynode-kw41z/Makefile.include does not use the `common/kw41z/Makefile.include` #11616
* [ ] ...

For the `phynode-kw41z` it was updated so that the `USEMODULE += boards_common_kw41z` is not put in common anymore but let on the board to include the module or not.

#### Related issues

This was currently found while moving `CPU/CPU_MODEL` to `Makefile.features`.

* Tracking: move CPU/CPU_MODEL to Makefile.features #11477
* phynode-kw41z/Makefile.include does not use the `common/kw41z/Makefile.include` #11616

fjmolinas, 2020-01-13:
Re-opening this is still valid IMO.

fjmolinas, 2020-08-18:
re-open