RIOT-OS/RIOT #11684: openmote-b: Cannot flash board with USB more than once

MrKevinWeiss, 2019-06-13:
If you are experiencing an issue with openmote-b boards only being able to flash with usb empty boards and not reflash, meaning either just arrived from purchase or erased with the JLink interface, please read the following:

This is an issue to provide information regarding some manufacturer problems with the openmote-b boards.  I believe this was the issue with #11031 as well.

The manufacturer stated:

> We had various people reporting the same behaviour and tracked it down to the firmware being executed by the MSP430 that manages the bootloading. 
Since programming the MSP430 is not intended for final users, I see two possible solutions:
1 - You ship back the devices, we reprogram them and ship them back to you
2 - We let you know the steps to reprogram the devices; you will need a MSP430-FET programmer and a special cable

For testing purposes I have found setting the `CCA_BACKDOOR_ACTIVE_LEVEL` to 1 instead of 0 allows for reflashing (however with battery power I think it could never execute main code) and if this is done on a board that is working it may lock it up.

cladmi, 2019-06-13:
Many boards have a `doc.txt`. It would be a better place to document this than in an issue.

cladmi, 2019-06-13:
Nice find though :)

MrKevinWeiss, 2019-06-25:
You would think there would be one for the openmote-b board... :disappointed:

MrKevinWeiss, 2019-06-25:
#11742

PR #11742 (2019-06-27): boards/openmote-b/doc: Add doc.txt to openmote-b
### Contribution description

It seems the doc was not added during the port.
This adds documentation similar to the openmote-cc2538.
This also expands on some flashing issues that have occurred.

### Testing procedure

`make doc` and read to see if it makes sense.

### Issues/PRs references
I don't know if it should close #11684
closed 2019-06-27 by smlng