RIOT-OS/RIOT #11691: murdock and tests using `utf-8` characters

cladmi, 2019-06-13:
#### Description

Tests using `utf-8` characters fails in murdock `native` but work my developer machine.

* tests/lua_loader: run the test in CI #11686
* tests/pkg_u8g2: run the test in CI #11687
* tests/pthread_rwlock: run the test in CI #11688 

For the boards tests, it looks like they work correctly on the nodes but the report is not described as an `utf-8` document so show invalid output in the reports by default.

Related to https://github.com/RIOT-OS/RIOT/pull/9095#issuecomment-470097586

#### Steps to reproduce the issue

Enable the given tests in CI and check the output.

#### Expected results

* Tests work in `native` correctly.
* Tests report show the `utf-8` characters by default on boards.

#### Actual results

* Tests using `utf-8` fail with `UnicodeEncodeError: 'ascii' codec can't encode characters in position` for native
* Encoding for file tests results is `utf-8`.

#### Versions

This is murdock specific so not RIOT.

kaspar030, 2019-06-13:
> Tests using `utf-8` characters fails in murdock `native` but work my developer machine.

Can you test on your developer machine if they work with  ```LANG_*``` set to C?

kaspar030, 2019-06-13:
And maybe test if adding "export LANG=C.UTF-8" to the top of ```./.murdock``` fixes the problem on CI?

kaspar030, 2019-06-13:
> For the boards tests, it looks like they work correctly on the nodes but the report is not described as an `utf-8` document so show invalid output in the reports by default.

Let's keep this issue seperate. It is a webserver configuration issue and not related to utf-8, but to non-utf-8 being present in the board's output.

cladmi, 2019-06-13:
`LC_ALL=C make -C tests/pthread_rwlock/ flash test` fails with the same error.
`LC_ALL=C.UTF-8 make -C tests/pthread_rwlock/ flash test` works.

I think `LANG` does not configure this, both `C` and `C.UTF-8` work without issues.

> And maybe test if adding "export LANG=C.UTF-8" to the top of `./.murdock` fixes the problem on CI?

Will try in `tests/pkg_u8g2: run the test in CI #11687`.

A good question being where should it be set by default ?
In the script, in the build system ?

> Let's keep this issue seperate. It is a webserver configuration issue and not related to utf-8, but to non-utf-8 being present in the board's output.

Agreed it is webserver configuration. I put it here if the locale configuration should have been something configured server side.

The issue is also present if only `utf-8` characters are present.

https://ci.riot-os.org/RIOT-OS/RIOT/11688/4879a310043103bcc23bb0077c0341f14d4fb10f/output/run_test/tests/pthread_rwlock/samr21-xpro:llvm.txt

cladmi, 2019-06-13:
Locally I successfully tested

```
LC_ALL=C  make -C tests/pthread_rwlock/ flash test
```
with 

``` diff
diff --git a/Makefile.include b/Makefile.include
index f9ed21b6a..149ac6d78 100644
--- a/Makefile.include
+++ b/Makefile.include
@@ -582,6 +582,7 @@ reset:
 .PHONY: test test/available
 TESTS ?= $(foreach file,$(wildcard $(APPDIR)/tests/*[^~]),\
                         $(shell test -f $(file) -a -x $(file) && echo $(file)))
+test: export LC_ALL=C.UTF-8
 test: $(TEST_DEPS)
        $(Q) for t in $(TESTS); do \
                $$t || exit 1; \
```

However on a board it fails to run `term` and I need to also add the variable for `term`.

cladmi, 2019-06-13:
> > And maybe test if adding "export LANG=C.UTF-8" to the top of `./.murdock` fixes the problem on CI?
> 
> Will try in `tests/pkg_u8g2: run the test in CI #11687`.

I should also still try to make the test fail to see how the error is handled.

kaspar030, 2019-06-13:
> A good question being where should it be set by default ?
> In the script, in the build system ?

I don't think we should override the user's locale system. Only obscure distributions do *not* configure utf-8 locales for users... Thus the build system is IMO not the right place.

Murdock is a different story. I think the build workers actually do configure utf-8, but on the RPi worker node, this doesn't happen. We could either set that up in the RPi systemd unit file, or just in ```.murdock```.
I think I'd prefer the latter, as I prefer the murdock environment as uniform as possible.
(I remember having this in there somewhere, in some context, but I don't know where...)

cladmi, 2019-06-13:
> I don't think we should override the user's locale system. Only obscure distributions do _not_ configure utf-8 locales for users... Thus the build system is IMO not the right place.

I wanted to overwrite it only for `test` and `term`.

If not overwritten, then we should add as requirement that the system running the tests is configured to use `UTF-8`, otherwise tests can fail.
Which would mean that the `CI` workers must be updated and not the `.murdock` script.

cladmi, 2019-07-11:
From local testing, it looks like it should be configured in the `riotdocker` image.

Currently the locale in the container is `"POSIX"` whatever that means

```
locale                                                                                                                                                                                                                                                                                    
LANG=                                                                                                                                                                                                                                                                                     
LANGUAGE=                                                                                                                                                                                                                                                                                 
LC_CTYPE="POSIX"                                                                                                                                                                                                                                                                          
LC_NUMERIC="POSIX"                                                                                                                                                                                                                                                                        
LC_TIME="POSIX"                                                                                                                                                                                                                                                                           
LC_COLLATE="POSIX"                                                                                                                                                                                                                                                                        
LC_MONETARY="POSIX"                                                                                                                                                                                                                                                                       
LC_MESSAGES="POSIX"                                                                                                                                                                                                                                                                       
LC_PAPER="POSIX"                                                                                                                                                                                                                                                                          
LC_NAME="POSIX"                                                                                                                                                                                                                                                                           
LC_ADDRESS="POSIX"                                                                                                                                                                                                                                                                        
LC_TELEPHONE="POSIX"                                                                                                                                                                                                                                                                      
LC_MEASUREMENT="POSIX"                                                                                                                                                                                                                                                                    
LC_IDENTIFICATION="POSIX"                                                                                                                                                                                                                                                                 
LC_ALL=                     
```

By doing a local test inspired from https://github.com/RIOT-OS/RIOT/pull/11220 it fails in the current `docker` image with a `samr21-xpro`.
I could not run `native` though due to a `native_cpu_init: getcontext: Operation not permitted` issue so ignored it.

```
BOARD=samr21-xpro make -C tests/lua_loader/ flash
DOCKER_FLAGS='--rm --privileged -i --group-add plugdev --group-add dialout --device=$(PORT)' BUILD_IN_DOCKER=1 DOCKER="sudo docker" BOARD=samr21-xpro make -C tests/lua_loader/ flash
```

The test fails due to the encoding errors.

```
print((require"m2").a)
2019-07-11 14:40:12,635 - INFO # chega de repente
print((require"c1").X)
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.6/logging/__init__.py", line 996, in emit
    stream.write(msg)
UnicodeEncodeError: 'ascii' codec can't encode character '\xe9' in position 48: ordinal not in range(128)
```

When adding `-e LC_ALL=C.UTF-8` it works:

[... cut ...]
        $(Q) for t in $(TESTS); do \
                $$t || exit 1; \
        done
+endif
 
 test/available:
        $(Q)test -n "$(strip $(TESTS))"
diff --git a/makefiles/docker.inc.mk b/makefiles/docker.inc.mk
index d7249b5f5..ffb05efd1 100644
--- a/makefiles/docker.inc.mk
+++ b/makefiles/docker.inc.mk
@@ -5,6 +5,7 @@ export DOCKER_FLAGS ?= --rm
 # List of Docker-enabled make goals
 export DOCKER_MAKECMDGOALS_POSSIBLE = \
   all \
+  test \
   buildtest \
   scan-build \
   scan-build-analyze \
```

kaspar030, 2019-07-11:
> From local testing, it looks like it should be configured in the `riotdocker` image.

+1, will PR a fix.

cladmi, 2019-09-16:
I added a PR to try re-enabling these tests https://github.com/RIOT-OS/RIOT/pull/12246

closed 2019-09-16 by miri64