RIOT-OS/RIOT #11706: how to set ISR stack size ?

jhx2002, 2019-06-17:
When I run tests/driver_enc28j60 on stm32f4discovery,  I got error :

```
Type '/exit' to exit.
2019-06-17 16:52:31,391 - INFO # main(): This is RIOT! (Version: 2019.04)
2019-06-17 16:52:31,394 - INFO # Test application for the enc28j60 driver
2019-06-17 16:52:31,395 - INFO # 
2019-06-17 16:52:31,400 - INFO # This test just pulls in parts of the GNRC network stack, use the
2019-06-17 16:52:31,406 - INFO # provided shell commands (i.e. ifconfig, ping6) to interact with
2019-06-17 16:52:31,408 - INFO # your enc28j60 device.
2019-06-17 16:52:31,409 - INFO # 
2019-06-17 16:52:31,411 - INFO # Starting the shell now...
ifconfig
2019-06-17 16:52:38,599 - INFO #  
2019-06-17 16:52:38,600 - INFO # Context before hardfault:
2019-06-17 16:52:38,602 - INFO #    r0: 0x1
2019-06-17 16:52:38,603 - INFO #    r1: 0x1
2019-06-17 16:52:38,605 - INFO #    r2: 0x800b668
2019-06-17 16:52:38,605 - INFO #    r3: 0x14
2019-06-17 16:52:38,606 - INFO #   r12: 0x0
2019-06-17 16:52:38,608 - INFO #    lr: 0xfffffffd
2019-06-17 16:52:38,608 - INFO #    pc: 0x80007e2
2019-06-17 16:52:38,610 - INFO #   psr: 0x21000037
2019-06-17 16:52:38,610 - INFO # 
2019-06-17 16:52:38,611 - INFO # FSR/FAR:
2019-06-17 16:52:38,613 - INFO #  CFSR: 0x8200
2019-06-17 16:52:38,613 - INFO #  HFSR: 0x40000000
2019-06-17 16:52:38,616 - INFO #  DFSR: 0x0
2019-06-17 16:52:38,616 - INFO #  AFSR: 0x0
2019-06-17 16:52:38,616 - INFO #  BFAR: 0x4c494146
2019-06-17 16:52:38,618 - INFO # Misc
2019-06-17 16:52:38,619 - INFO # EXC_RET: 0xfffffff1
2019-06-17 16:52:38,624 - INFO # Attempting to reconstruct state for debugging...
2019-06-17 16:52:38,624 - INFO # In GDB:
2019-06-17 16:52:38,627 - INFO #   set $pc=0x80007e2
2019-06-17 16:52:38,628 - INFO #   frame 0
2019-06-17 16:52:38,628 - INFO #   bt
2019-06-17 16:52:38,628 - INFO # 
2019-06-17 16:52:38,632 - INFO # ISR stack overflowed by at least 64 bytes.

```

PeterKietzmann, 2019-06-17:
@jhx2002 can you share your pin configuration?

jhx2002, 2019-06-17:
I use the default pin configuration of stm32f4discovery

PeterKietzmann, 2019-06-17:
What is the default configuration?

jhx2002, 2019-06-17:
Part configure is contained in:
https://github.com/RIOT-OS/RIOT/blob/master/tests/driver_enc28j60/Makefile

Another configure is contained in :
https://github.com/RIOT-OS/RIOT/blob/master/boards/stm32f4discovery/include/periph_conf.h

jhx2002, 2019-06-18:
Could any one help me to find out the solution?

aabadie, 2019-06-18:
Can you tell us on which pins and SPI you plugged the ethernet device ?

jhx2002, 2019-06-18:
Thank you , here is pins plugin
```
PA0 -------------- CS
PA5 -------------- SCK
PA6 -------------- SO
PA7 -------------- SI
PA1--------------INT
PA2---------------RESET
```

PeterKietzmann, 2019-06-18:
@jhx2002 I'm a little surprised about these pins. You pointed to the configurations above but didn't apply them? `SPI_DEV(1)` is on PB12-15, `INT` on  `PB11`and `RST` on PB10.

SPI:
```
{
        .dev      = SPI2,
        .mosi_pin = GPIO_PIN(PORT_B, 15),
        .miso_pin = GPIO_PIN(PORT_B, 14),
        .sclk_pin = GPIO_PIN(PORT_B, 13),
        .cs_pin   = GPIO_PIN(PORT_B, 12),
        .af       = GPIO_AF5,
        .rccmask  = RCC_APB1ENR_SPI2EN,
        .apbbus   = APB1
}
```

Other:
```
ifneq (,$(filter stm32f4discovery,$(BOARD)))
  ENC_SPI ?= SPI_DEV\(1\)
  ENC_CS  ?= GPIO_PIN\(PORT_B,12\)
  ENC_INT ?= GPIO_PIN\(PORT_B,11\)
  ENC_RST ?= GPIO_PIN\(PORT_B,10\)
endif
```

jhx2002, 2019-06-18:
There are two SPI_DEV conf in
``` 
https://github.com/RIOT-OS/RIOT/blob/master/boards/stm32f4discovery/include/periph_conf.h
```
Which one should I use ? 
```
static const spi_conf_t spi_config[] = {
    {
        .dev      = SPI1,
        .mosi_pin = GPIO_PIN(PORT_A, 7),
        .miso_pin = GPIO_PIN(PORT_A, 6),
        .sclk_pin = GPIO_PIN(PORT_A, 5),
        .cs_pin   = GPIO_PIN(PORT_A, 4),
        .af       = GPIO_AF5,
        .rccmask  = RCC_APB2ENR_SPI1EN,
        .apbbus   = APB2
    },
    {
        .dev      = SPI2,
        .mosi_pin = GPIO_PIN(PORT_B, 15),
        .miso_pin = GPIO_PIN(PORT_B, 14),
        .sclk_pin = GPIO_PIN(PORT_B, 13),
        .cs_pin   = GPIO_PIN(PORT_B, 12),
        .af       = GPIO_AF5,
        .rccmask  = RCC_APB1ENR_SPI2EN,
        .apbbus   = APB1
    }
};
```

PeterKietzmann, 2019-06-18:
> SPI_DEV(1) is on PB12-15

PeterKietzmann, 2019-06-18:
The `1` indexes the the `spi_config[]` array

jhx2002, 2019-06-18:
My board is stm32f407vet6 , almost same to stm32f4discovery. So I copy boards and cpu dirs of stm32f4discovery to my board. And modify the tests/driver_ENC28J60 as : 

```
# name of your application
APPLICATION = enc28j60

BOARD = stm32f407vet6
#BOARD = stm32f4discovery

# This has to be the absolute path to the RIOT base directory:
RIOTBASE ?= /home/huaxing/oproj/riot/RIOT
include $(RIOTBASE)/tests/Makefile.tests_common

BOARD_INSUFFICIENT_MEMORY := arduino-duemilanove arduino-mega2560 arduino-nano \
                             arduino-uno msb-430 msb-430h nucleo-f334r8 \
                             nucleo-l053r8 nucleo-f031k6 nucleo-f042k6 \
                             nucleo-f303k8 nucleo-l031k6 mega-xplained \
                             stm32f0discovery telosb waspmote-pro wsn430-v1_3b \
                             wsn430-v1_4 z1

#THREAD_STACKSIZE_DEFAULT = 20240
#THREAD_STACKSIZE_MAIN = 20240

USEMODULE += auto_init_gnrc_netif
USEMODULE += enc28j60
USEMODULE += gnrc_ipv6_router_default
USEMODULE += gnrc_icmpv6_echo
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps

#STDIO_UART_DEV ?= UART_DEV\(1\)

# set board specific peripheral configurations
#ifneq (,$(filter stm32f4discovery,$(BOARD)))
#  ENC_SPI ?= SPI_DEV\(1\)
#  ENC_CS  ?= GPIO_PIN\(PORT_B,12\)
#  ENC_INT ?= GPIO_PIN\(PORT_B,11\)
#  ENC_RST ?= GPIO_PIN\(PORT_B,10\)
#endif

# fallback: set SPI bus and pins to default values
ENC_SPI ?= SPI_DEV\(0\)
ENC_CS  ?= GPIO_PIN\(0,0\)  # PA0
ENC_INT ?= GPIO_PIN\(0,1\)  # PA1
ENC_RST ?= GPIO_PIN\(0,2\)  # PA2
# export SPI and pins
CFLAGS += -DENC28J60_PARAM_SPI=$(ENC_SPI)
CFLAGS += -DENC28J60_PARAM_CS=$(ENC_CS)
CFLAGS += -DENC28J60_PARAM_INT=$(ENC_INT)
CFLAGS += -DENC28J60_PARAM_RESET=$(ENC_RST)

# make sure we read the local enc28j60 params file
CFLAGS += -I$(CURDIR)

include $(RIOTBASE)/Makefile.include

```

jhx2002, 2019-06-18:
It's Ok now.  with add in Makefile:
```
STDIO_DEV = UART_DEV\(1\)
CFLAGS += -DSTDIO_UART_DEV=$(STDIO_DEV)

```

closed 2019-06-18 by PeterKietzmann