RIOT-OS/RIOT #11714: Does cc1101 can support gnrc networking in nucleo–f

hujiaodigua, 2019-06-18:
#### Description

### Useful links

hujiaodigua, 2019-06-18:
Does cc1101 can support gnrc networking in nucleo–f103rb

PeterKietzmann, 2019-06-18:
Please see #8754,  #10340 and #10373. I'm closing this because it's a duplicate.

closed 2019-06-18 by PeterKietzmann

maribu, 2019-06-19:
@hujiaodigua: Once #10340 is merged, it should work pretty much out of the box. I did a lot of testing with it on the BluePill, which uses an STM32F103 MCU. If you want to test that PR and report and bugs/rough edges you encounter, you are very welcome :-)

The current cc110x driver will also work, but only with small frames (everything above 64 bytes is buggy). Using that driver for non-868 MHz transceivers requires overwriting the content of some configuration registers; and calculation of their correct contents. So from an end user's point of view, #10340 will be much easier to use, where you only need to select the appropriate config ;-)

hujiaodigua, 2019-06-19:
@maribu Many thanks for your reply. With the help of your other previous replies, I tried gnrc_netowrking example(BOARD=nucleo-f103rb, actually stm32f103rct6 discovery ) successful by using cc110x (433MHz).
By reference to [](http://onewrt.com/cc1101-6lowpan-in-riot-os/))
Then, I found configure another address is very important.
`ifconfig 7 add fe80::ff:dc`
After that, _ping6_ and _udp send_ can be used. And the signal is unstable, most likely because of the antenna.

maribu, 2019-06-19:
> Then, I found configure another address is very important.

Hmm, this should only be the case if both nodes end up getting the same MAC address. This can be for two reasons: The board does not have a cpu ID to generate a MAC address from. (The STM32F1 family does have this). Or bad luck: With only 255 (0x01 to 0xff) possible addresses, with about 16 device you already have a chance of about 50% of a collision due to the birthday paradox.

Can you paste the output of `ifconfig` on both nodes after boot, so that I can get an idea what went wrong?

Keep in mind that the IPv6 address needs to match the L2 address. E.g. if you have an L2 address of `0x42`, you need to have `fe80::ff:fe00:42` as IPv6 address. You can set the L2 address using `ifconfig <ID> set addr <L2_ADDR_IN_HEX>`.

hujiaodigua, 2019-06-19:
Now let me describe it briefly.

After boot,  

Node 1,
Address , fe80::ff:fe00:dc

Node 2,
Address , fe80::ff:fe00:39

**Cannot ping6 each other.** ( I will upload the screenshot of the problem tomorrow. )

Then,  configure the address

Node 1,
`ifconfig 7 add fe80::ff:dc`

Node 2,
`ifconfig 7 add fe80::ff:39`

**Can ping6 each other.** (https://github.com/hujiaodigua/Screenshot/blob/master/ping6.png)

maribu, 2019-06-19:
@hujiaodigua: Are you using the cc110x driver from the current master / latest stable release, or the branch from the pull request in #10340?

The round trip time with 12 ms is about the time the old driver had, the new driver in #10340 has a round trip time of about 3 ms instead with the default configuration. Therefore, I assume you're using the old driver. The new driver is already more reliable than the old one, so you might want to give that one a try.

hujiaodigua, 2019-06-20:
@maribu I downloaded the current master. ( Merge pull request #11641 from fjmolinas/pr_riotboot_m4 ) 
The round trip time still with 12 ms. 
And I will use our [HATBED](https://cps-iotbench2019.ethz.ch/wp-content/uploads/2019/04/P4_new.pdf) ( hardware-assisted tracing testbed ) to test this example, I will give you feedback if there is any new discovery.

maribu, 2019-06-20:
Can you try it with the PR #10340? You can use `make doc` to generate the documentation, provided you have Doxygen installed. In the doc should be everything written you need to now to get it working.

hujiaodigua, 2019-06-20:
I downloaded the fork [version](https://github.com/maribu/RIOT). The situation has not improved. Now, I doubted something wrong with my 433MHz home-made antenna.

maribu, 2019-06-20:
@hujiaodigua: Don't forget to checkout the `cc110x_rewrite` branch of my Repo when using that. The master is identical to the official RIOT master. (Maybe I'm a few commit behind, as I update manually).