RIOT-OS/RIOT #11735: riotboot: no updates will be possible after 19.1.2038

benpicco, 2019-06-24:
The Riotboot Header currently uses a 32 bit `version` field.
The version is set to the Unix Timestamp at build time.
Riotboot will always boot the valid image with the largest `version`.

This means that no images produced after 19.1.2038, 03:14:07 UTC will be booted if a board has been flashed with an image produced before that date.

## Solution
Make `version` 64 bit.

jcarrano, 2019-06-25:
Why is `version` used for time? Shouldn't version be for version, i.e. a number that is manually assigned?

benpicco, 2019-06-25:
The default is [`APP_VER ?= $(shell date +%s)`](https://github.com/RIOT-OS/RIOT/blob/master/makefiles/boot/riotboot.mk#L17).

This can be of course be overwritten manually, but once you've used the default once, you end up with a rather big version number :wink: 

And I think most users will stick to the default since it just works™.

jcarrano, 2019-06-25:
My point is that if someone is updating firmware in a real-life setting, maybe they should give some more thought to the versioning scheme, rather than simply telling the device "load whatever was compiled later in time".

I'm not saying `APP_VER ?= $(shell date +%s)` is a bad idea - it is great for development. I wonder, though, wether it was smart to place a default in `riotboot.mk` rather than put it in the application's Makefile, and thus force the user to at least see what's going on.

kaspar030, 2019-06-25:
We're using *unsigned*, not signed 32bit integers.

Thus there's like 68 more years to update to a version that changes the logic, *after* 2038. If in the heat of 2038 panic someone finds out we're using only 32bits, we can still change the ```APP_VER ?= $(EPOCH)``` and have 2 billion more update version left.

And then there are easy solutions to reset the number. Only one image (let's say it has version 0xffffffff) needs to first flash a newer image with a lower version, validate it, then invalidate the active slot. Just in case our cute little 20 line bootloader is still around.

benpicco, 2019-06-25:
Oh I didn't notice the unix timestamp is usually *signed*!
Since we'll all be long dead before this becomes an issue then, I think I can close this.

closed 2019-06-25 by benpicco