RIOT-OS/RIOT #11756: E-paper driver samr21 using spi

talalong, 2019-06-27:
#### Description

I'm trying to control my 4.2inch e-paper display from waveshare using board samr21-xpro via SPI. There are libraries supporting this display, but they are all written in C++ and this feature #11078 is also not official yet and also did not work on my 400x300 display. So I decided to make my own '_driver_' with my little knowledge. If it works, it could be new pull request for driver of _samr21-xpro_ to control _epaper_ display :)

Here is my setup and little code:
[image]

Here is the wiring:
**Samr21**       <--->      **E-paper** 400x300
VCC-------------->     VCC
GND-------------->    GND
PB22------------->     DIN(MOSI)
PB23------------->     CLK
PA14-------------->    CS
PB02------------->     DC
PA16------------ ->     RST
PA08------------->     BUSY

```
int main(void)
{
    spi_t bus = SPI_DEV(1);
    spi_cs_t cs = (spi_cs_t)GPIO_PIN(PA, 14);
    spi_mode_t mode = SPI_MODE_1;
    spi_clk_t clk = SPI_CLK_10MHZ;
    gpio_t dc = (gpio_t)GPIO_PIN(PB,2);
    gpio_t reset = (gpio_t)GPIO_PIN(PA,16);

    spi_init(bus);
    spi_acquire(bus, cs, mode, clk);
    gpio_write(dc,0);
    gpio_write(reset,0);
    for (uint16_t i = 0; i < (32*32) - 1; i++) {
        spi_transfer_byte(bus, cs, true, 0xff);
    }
    spi_release(bus);
    char line_buf[SHELL_DEFAULT_BUFSIZE];
    shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);
    return 0;
}
```

No error has been found while compiling using `make flash term`. But nothing happened on my display. I thing I did miss something here, but I don't know what is missing.
i would very much appreciate for your helps.
### Useful links

https://www.waveshare.com/w/upload/6/6a/4.2inch-e-paper-specification.pdf

dylad, 2019-06-28:
Hi @talalong 
From what I see, there is some mistakes with your code.

First `spi_init()` should not be call by your application. This is done automatically by RIOT in `drivers/periph_common/init.c`
Instead, you should initialize your CS pin with `spi_init_cs()`

In addition, I took a really quick look at your datasheet, your reset is active low so `gpio_write(reset,0);`
will keep your device into reset mode.
If I understand correctly, you're sending data not command to your screen so you should also change `  gpio_write(dc,0);`
Last but not least, you should also initialize your reset and dc pins with `gpio_init()`

talalong, 2019-07-02:
@dylad Thanks for your answer.
I did make some changes that you pointed at. But the display still does nothing.
So my questions are:
These steps are to make spi transfer, are they correct?
Did I miss the initialization for e-paper display, so that's why I counld't interact with it?

```
#define DISPLAY_REFRESH           0x12
int main(void)
{
    spi_t bus = SPI_DEV(1);
    spi_cs_t cs = (spi_cs_t)GPIO_PIN(PA, 14);
    spi_mode_t mode = SPI_MODE_1;
    spi_clk_t clk = SPI_CLK_10MHZ;
    gpio_t dc = (gpio_t)GPIO_PIN(PB,2);
    gpio_t reset = (gpio_t)GPIO_PIN(PA,16);
    gpio_t busy = (gpio_t)GPIO_PIN(PA,8);

    /* initialization */
    spi_init_cs(bus,cs);
    if(gpio_init(dc,GPIO_OUT)<0){puts("initialization of pin DC failed!");}
    if(gpio_init(reset,GPIO_OUT)<0){puts("initialization of pin RESET failed!");}
    if(gpio_init(busy,GPIO_IN)<0){puts("initialization of pin BUSY failed!");}
    /* reset */
    gpio_write(reset, 0);
    xtimer_usleep(TWOMILISEC);  
    gpio_write(reset, 1);

    /*sending command POWER SETTING*/
     gpio_write(dc, 0);
    spi_transfer_byte(bus,cs,true,POWER_SETTING);

    /* sending command DISPLAY REFRESH */
    spi_acquire(bus, cs, mode, clk); 
    gpio_write(reset,1);
    gpio_write(dc,0);
    spi_transfer_byte(bus,cs,true, DISPLAY_REFRESH);

    spi_release(bus);
    char line_buf[SHELL_DEFAULT_BUFSIZE];
    shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);
    return 0;
}
```

dylad, 2019-07-02:
@talalong 
`spi_acquire()` must be call before any spi transaction. There is no need to call    `gpio_write(dc,0);` several times.
I also don't think your commands are right. You are sending the register you want to write but without any data...

talalong, 2019-07-06:
Okay. Thanks again for your answer. I am able to control my e-paper display, now. But sometimes my vagrant terminal freezes. So I have to reboot the system all the time. Does my code the reason why my vagrant freezes? Sorry if the question is extensive!

dylad, 2019-07-08:
> I am able to control my e-paper display, now.

That's a good news !

> Does my code the reason why my vagrant freezes?

I can't help you with Vagrant as I never used it. This might need another issue or a mail to the devel mailing list.

In any case, feel free to close your issue if your concerns were addressed

closed 2019-07-08 by talalong