RIOT-OS/RIOT #11763: spi_transfer_reg behavior doesn't follow doc

dylad, 2019-06-28:
#### Description
spi_transfer_reg implementation doesn't follow its documentation. 

```
/**
 * @brief   Transfer one byte to/from a given register address
 *
 * This function is a shortcut function for easier handling of SPI devices that
 * implement a register based access scheme.
 *
 * @param[in] bus       SPI device to use
 * @param[in]  cs       chip select pin/line to use, set to SPI_CS_UNDEF if chip
 *                      select should not be handled by the SPI driver
 * @param[in] reg       register address to transfer data to/from
 * @param[in] out       byte to send, set NULL if only receiving data
 *
 * @return              value that was read from the given register address
 */
uint8_t spi_transfer_reg(spi_t bus, spi_cs_t cs, uint8_t reg, uint8_t out);
```
We should be able to put a NULL into last argument to only receive data but last argument isn't a pointer.

Now the question is : Should we update the documentation to follow the current implementation OR should we update the current implementation to follow the documentation ?

#### Steps to reproduce the issue
put a NULL in last argument.

#### Expected results
put a NULL in last argument to only receive the content of a register.

#### Actual results
Compilation failed

#### Versions
master

benpicco, 2019-07-19:
I mean on SPI, for each byte you write you read back a byte at the same time.
That means you can not read a byte without writing a byte.

so you have

write | read
------|------
 reg  |   0
 0  | value

dylad, 2019-07-20:
@benpicco 
I agree with you but this is not the problem raised by this issue.
My issue is the inconsistency between RIOT's documentation and function behaviour, as the documentation states that a NULL can be pass to function to ignore data read but this argument has uint8_t type. You cannot put a NULL to a uint8_t argument without raising a compilation issue so either the doc or the function behavior is wrong here.

benemorius, 2019-07-20:
Is there a standard among SPI slaves for what data can/must be sent in the receive-only case? If we have a guarantee that sending 0x00 is always ok then we can change the function behavior, but otherwise the behavior is correct and it's the documentation that needs to change.