RIOT-OS/RIOT #11811: asymcute die because of dead lock?

jhx2002, 2019-07-06:
I write a program using asymcute.  After runing about 2 miniutes, the program die.

miri64, 2019-07-06:
Can you please be a bit more specific? What exactly do you mean with "die" (in the title you say it's a deadlock). How is your application structured?

jhx2002, 2019-07-06:
publish one data to a gateway on local area network  every  10ms.

jhx2002, 2019-07-06:
I copy and modify  asymcute_publish to asymcute_publish2.
The problem exist too. 

```
int asymcute_publish2(asymcute_con_t *con, asymcute_req_t *req,
                      const asymcute_topic_t *topic,
                      const void *data, size_t data_len, uint8_t flags)
{
    assert(con);
    assert(req);
    assert(topic);
    assert((data_len == 0) || data);

    int ret = ASYMCUTE_OK;

   /* assemble message */
    size_t pos = _len_set(req->data, data_len + 6);
    req->data[pos] = MQTTSN_PUBLISH;
    req->data[pos + 1] = (flags | topic->flags);
    byteorder_htobebufs(&req->data[pos + 2], topic->id);
    byteorder_htobebufs(&req->data[pos + 4], req->msg_id);
    memcpy(&req->data[pos + 6], data, data_len);
    req->data_len = (pos + 6 + data_len);

    sock_udp_send(&con->sock, req->data, req->data_len, &con->server_ep);

    return ret;
}

```

jhx2002, 2019-07-06:
Is sock_udp_send OK ?

jhx2002, 2019-07-06:
All threads are dead after about 2 ~ 5 minutes

jhx2002, 2019-07-07:
I also test emcute.  Problem is same. So What's the problem ?

jhx2002, 2019-07-08:
Could some one help me ?

jhx2002, 2019-07-08:
could you give some advices on how to debug this problem?

miri64, 2019-07-08:
10ms sounds like a very tight time frame. I'll investigate the reason for the deadlock, seems to me like a problem with the network stack rather.

If you want to investigate yourself: The `ps` shell command can help you find deadlocks. If it's a double mutex-lock (the table printed by `ps` will show `bl mutex` for the thread in question) I usually put some wrapper macros into `core/include/mutex.h` to find out where the double lock happens:

```C
/**
 * @brief Locks a mutex, blocking.
 *
 * @param[in] mutex Mutex object to lock. Has to be initialized first. Must not be NULL.
 */
static inline void __mutex_lock(mutex_t *mutex)
{
    _mutex_lock(mutex, 1);
}

/**
 * @brief Unlocks the mutex.
 *
 * @param[in] mutex Mutex object to unlock, must not be NULL.
 */
void __mutex_unlock(mutex_t *mutex);

#include <stdio.h>

#define mutex_lock(m) printf("%s:%d: lock %p\n", __FILE__, __LINE__, (void *)&m); __mutex_lock(m)
#define mutex_unlock(m) printf("%s:%d: unlock %p\n", __FILE__, __LINE__, (void *)&m); __mutex_unlock(m)
```

(of course `mutex_unlock()` to `__mutex_unlock()` needs also be renamed in `core/mutex.c`)

You then can identify a double lock with a lock to a pointer without an unlock in-between.

As a workaround for now: Are you bound to the 10ms time frame? If not I'd advice to try to go for something more relaxed like 500-1000 ms. With those I made good experiences in the past. I even [have demo](https://github.com/miri64/RIOT_playground/blob/6b53ae77a5eb5e02adca808f3aeec484930f59a5/lndw18_luke/display/main.c#L80-L116) were I did a notification for a CoAP GET OBSERVE (similar mechanism to MQTT-SN Publish QoS 0, just different protocol) every ~100ms _over a radio_ which ran for several hours.

Side-question: which QoS are you using? As you did not mention connection or registration I guess QoS -1 with a predefined topic ID or short topic?

jhx2002, 2019-07-08:
Qos is 0 . 
In my use case 10~20 ms  is needed.
topic name length < 32 bytes.
Thank you!

jhx2002, 2019-07-10:
I replace emcute.c with your fixes. It runs longer, but still die. 
Thank you @miri64

jhx2002, 2019-07-10:
I find that  the program die  when some line (lines between stm32f4 and enc28j60 ) broken in 0.1 second.  
It can not continue to run .

jhx2002, 2019-07-12:
problem:  multiple sub on same topic in  emcute.c.
Only last one  can receive.

miri64, 2019-07-12:
> problem: multiple sub on same topic in emcute.c.
> Only last one can receive.

I did not observe this behavior in #11823... Furthermore this definitely has nothing to do with the original title of this issue, so this should go into a separate issue.

jhx2002, 2019-07-13:
> > problem: multiple sub on same topic in emcute.c.
> > Only last one can receive.
> 
> I did not observe this behavior in #11823... Furthermore this definitely has nothing to do with the original title of this issue, so this should go into a separate issue.

See code in master, only send to first found:
```
/* find the registered topic */
    for (sub = subs; sub && (sub->topic.id != tid); sub = sub->next) {}
    if (sub == NULL) {
        buf[6] = REJ_INVTID;
        sock_udp_send(&sock, &buf, 7, &gateway);
        DEBUG("[emcute] on pub: no subscription found\n");
    }
    else {
        if (rbuf[pos + 1] & EMCUTE_QOS_1) {
            sock_udp_send(&sock, &buf, 7, &gateway);
        }
        DEBUG("[emcute] on pub: got %i bytes of data\n", (int)(len - pos - 6));
        size_t dat_len = (len - pos - 6);
        void *dat = (dat_len > 0) ? &rbuf[pos + 6] : NULL;
        sub->cb(&sub->topic, dat, dat_len);
    }
}

```

jhx2002, 2019-07-15:
Use high quality wire to connect enc28j60 and stm32.  The problem still exist. It die after a few miniutes.

jhx2002, 2019-07-16:
Test recv in 10HZ,  it die in 30seconds

jhx2002, 2019-07-16:
Anyone has good idea to findout the root reason.

miri64, 2019-07-16:
Hi sorry I was (and still am) occupied with other stuff, so I wasn't really able to look into it :(. Usually putting `printf()`s or `puts()` in strategic places helps with debugging these kind of race conditions.

miri64, 2019-07-18:
Superseded by https://github.com/RIOT-OS/RIOT/issues/11860

closed 2019-07-18 by miri64