RIOT-OS/RIOT #11826: RFC: Change of sensor and actoator API & auto initialization

maribu, 2019-07-11:
#### The Problem
Currently sensors and actuators are only auto-initialized when SAUL is used. This is not only inconvenient by itself, but also makes using both SAUL and the native driver API impossible. (One use case: Debugging via the `saul` shell command while the application uses the native interface for accessing additional device-specific features. Another use case: Different "apps" running as separate threads on the same device, where one app is portable and using SAUL, while the other needs to access device-specific features not available through SAUL.)

##### The Issue in Detail

E.g. in the auto init code of the `dht` driver the device descriptors are allocated here:

https://github.com/RIOT-OS/RIOT/blob/6b094a78a943bcbd74d21a7ce83840c1fdcdc0a6/sys/auto_init/saul/auto_init_dht.c#L35-L38

With them being `static`, it is impossible (without resorting to nasty hacks) to access the auto initialized descriptors. And having two device descriptors referring to the same device is something that will cause bad things to happen.

Even if we would remove the `static` qualifier from all device descriptor arrays, using `&dht_devs[0]` is kind of ugly.

### Proposed Solution

A sensor `foo` should provide to types: 

``` C
typedef struct {
     foo_params_t params;
     int state;
     ....
} foo_dev_t;

typedef uint8_t foo_t;
```

In the auto-init code we would have

``` C
foo_dev_t foo_devs[N_FOO_DEVS];
```

So `foo_dev_t` would contain the data/configuration/state of a device, and `foo_t` would be the index in `foo_devs`. The auto-init code must initialize all available sensors regardless of whether SAUL is used. If SAUL is used, it must additionally register the initialized sensors.

The API of foo should look than like this:

``` C
int foo_read(foo_t dev, uint16_t *dest);
....
```

This would also be more consistent with other RIOT APIs, e.g. `gpio_t`, `spi_t`, `i2c_t` etc. are also "simple types" that one can pass as value, while e.g. `bmx280_t` is something you really want to pass as reference. Also a POSIX file descriptor is an `int`, the newly proposed device descriptor would be much better match to POSIX descriptors than a potentially large `struct`.

maribu, 2019-07-11:
@haukepetersen, @gschorcht, @dylad, @smlng, @aabadie, @keestux, @gebart , @MichelRottleuthner, @SemjonKerner, @leandrolanzieri, @basilfx: I found your names repeatedly in `sys/auto_init/saul`, so I'd be happy if you could join this discussion :-)

jcarrano, 2019-07-12:
I believe we should find a way to have _less_ things in auto_init instead of _more_.

Right there, in the example you cite (DHT) there is code in `sys/auto_init`. That code interacts with `sys/saul_reg` and `drivers/dht`. So not only does it cross module boundaries, it crosses the sys-drivers boundary. This is not specific to DHT and is a general feature of auto_init code.

Would having access to the lowel-level device handle through saul work, or do we also require that the module be auto initialized when saul is not being used?

I think the first one is a not-pretty but acceptable work-around. The second would be adding a tons of stuff to auto_init and while it would not be better or worse than what is now, it would add to pile that we should clean up some day.

maribu, 2019-07-12:
> I believe we should find a way to have *less* things in auto_init instead of *more*.

People tend to expect that an OS initializes its drivers. And I think for very fine reasons:

1. It reduces the boiler plate code in applications
2. The initialization code is well tested and easier to maintain, than having the same functionality and code duplicated for each and every user of the driver
3. It is simpler to use and simpler to use **correctly**
    a) It prevents applications from using the driver uninitialized
    b) It prevents two threads accessing a device from double-initialize a driver
    c) Even if the two threads would try to prevent double-initializing by checking if the driver is already initialized, this is inherently complex. E.g. the average application developer will likely write racy code that works by pure luck 99.9% of the time, with 0.1% ending up in two threads concurrently initializing the driver

> So not only does it cross module boundaries, it crosses the sys-drivers boundary.

This could be solved by moving driver initialization code to `drivers` (e.g. `drivers/auto_init`). But a boot sequence will always cross module-boundries. E.g. the startup code lives in `cpu`, which will e.g. call `board_init()` in `boards` and `periph_init` in `drivers`, and `kernel_init()` in `core`.

That said, I have nothing against moving driver initialization into `drivers`. But I'm strongly in favor of automatic initialization of the drivers (regardless of SAUL being used). (And of automatic registration of sensors and actuators to SAUL, if SAUL is used.)

maribu, 2019-07-12:
@jcarrano: Maybe you want to open a separate thread to discuss the architecture of `auto_init` in general? I'd personally would like to keep the discussion here focused on:

1. Is auto-initialization of sensor and actuator drivers is desired?
2. Should using the native sensor/actuator API and SAUL no longer be mutually exclusive?
3. And, if the first two questions are answered with yes: How should the sensor/actuator APIs be changed to achieve this?

(@jcarrano: I couldn't figure out your opinion on the above. Care to share it?)

SemjonKerner, 2019-07-12:
IMO initialisation should be part of the application and not of auto_init.
1. In case of SAUL the initialisation is obvsly necessary, in most other cases it is neither obvious nor necessary to do it automatically.
2. People that expect the OS to take care of stuff like that are probably not the target audience of Riot.
3. Auto_init comes with default values - changing those defaults with a special header file is more complicated than calling init functions in all cases that I know of.
4. Sometimes initialisation is a recurring part of the application, because for some drivers its the only function that changes specific values.

maribu, 2019-07-12:
> 1. In case of SAUL the initialisation is obvsly necessary, in most other cases it is neither obvious nor necessary to do it automatically.

Keep in mind that this does increase the complexity of the application. E.g. if code is written to always initialize the driver into a local device descriptor, bad things can easily happen:

1. Someone compiles with `USEMODULE=saul_default make`
2. Some other shell command relying on auto-init (e.g. the `weather` shell command) is added
3. Another thread is also using the driver. It would need to block until the other thread has completed the initialization, or other ways of synchronizing the driver need to be found.

> 2. People that expect the OS to take care of stuff like that are probably not the target audience of Riot.

The distinctive point between other operating systems for IoT devices and RIOT is, that RIOT is more user friendly: It supports the common POSIX APIs application developers are used to. Historically (before RIOT was renamed to RIOT), async and event driven APIs were used. But application developers (and at that time mostly students) are much more comfortable with having blocking APIs and rely on multiple threads.

Citing [RIOT's homepage](http://riot-os.org/#features): 

> ### RIOT is Developer Friendly
>
> Program like you are used to. Do not waste time with complex or new environments. 

These people are ***exactly*** the target audience of RIOT!

> 3. Auto_init comes with default values - changing those defaults with a special header file is more complicated than calling init functions in all cases that I know of.

Citing from [RIOT's guide for device driver development](https://github.com/RIOT-OS/RIOT/wiki/Guide%3A-Writing-a-device-driver-in-RIOT#default-device-configuration):

>  We can override one or more of the parameter from the makesystem, e.g.
>
> CFLAGS="-DTMPABC_PARAM_ADDR=0x23" make all

There is also the configuration task force (e.g. @miri64 and @jia200x are working on that) working on ways to make compile time configuration easier.

> 4. Sometimes initialisation is a recurring part of the application, because for some drivers its the only function that changes specific values.

So then lets fix these APIs ;-)

miri64, 2019-07-12:
Just a sidenote for this discussion as I was mentioned: Please keep in mind that `auto_init` is optional. You always can exclude it with `DISABLE_MODULE`.

SemjonKerner, 2019-07-14:
I don't agree with your argumentation. Of course it is not my intention to obfuscate driver usage by not initializing sensors. But this is not at all what I said.
Riot is not a copy-paste, arduino-like OS (not my intention to insult ardunio :D).
Init functions are always documented and the usage of those is straight forward, so we are not making anything overly complicated for programmers by not initializing. Moreover we are giving them the freedom to do it if and when needed.
IMHO anyone who fails to use an init function, who is also not able to fix the issue within the time you'd need to find a missing semicolon is **_NOT_** our target audience.

I don't really get why you gave this excursion, since it'd only support your point in case it was the standard for OSs to initialize drivers and thus everyone is used to this behaviour. I don't what to get all rhetoric, but even considering standards for OS driven sensor initialisation seems weird to me.

> So then lets fix these APIs ;-)

You know yourself how hard this is and sometimes not even wanted. I think, if you are really convinced of this idea you should not push in this direction right away, or it will give you a hard time discussing the basics.

> 1. Someone compiles with USEMODULE=saul_default make
> 2. Some other shell command relying on auto-init (e.g. the weather shell command) is added

Anyone interested in using auto_init for their sensors should not be interested in managing the sensor themselves. If this is a concern then there should be a note in the documentation (probably both in auto_init calling modules and in the drivers that support auto_init for a module).

> 3. Another thread is also using the driver. It would need to block until the other thread has completed the initialization, or other ways of synchronizing the driver need to be found.

Making sure a driver is properly initialized before usage is always appreciated, but again IMHO not necessarily the responsibility of the OS.
Even if this is a wanted feature it might as well be a part of the driver (e.g. status flags/mutex guarding function calls). This seems overly complicated as well, but most sensors are not expected to be threadsafe anyway.

maribu, 2019-07-14:
> I don't really get why you gave this excursion, since it'd only support your point in case it was the standard for OSs to initialize drivers and thus everyone is used to this behaviour. I don't what [...]

It is commonly considered beneficial to be aware of core design goals of a software project. It is commonly considered beneficial to know the reasons and historic context behind core design goals. And it is also commonly considered as a strong argument to point out that one design decision is aligned with a core design goal and another is not.

All POSIX compliant OS have their drivers initialized during boot up. (Try `cat /sys/class/thermal/thermal_zone<NUM>/temp` on a Linux machine. The temperature sensors are *always* initialized once the userspace API (the special file) is accessible.)

A core design goal of RIOT is that it *can* be used similar to a POSIX compliant OS. (This does not mean RIOT cannot also provide non-POSIX-like alternatives - e.g. RIOT's sock API can be used instead of the POSIX socket API.)

> Anyone interested in using auto_init for their sensors should not be interested in managing the sensor themselves.

I don't get this. Anyone wanting to *use* a driver has a *strong* interest in having it initialized. E.g. SAUL is not *manageing* the drivers, but *using* them. And SAUL *allows* the user to auto_initialize and register the drivers with `saul_auto_init` (which is a dependency of `saul_default`). The user can just as well not use `saul_auto_init` and do the initialization and registration himself/herself.

But you are forgetting one important point: If there is a module to auto initialize sensor / actuator drivers, the user can decide to use it or still initialize them manually. If there is no such module, the decision to manually initialize the driver is forced upon the user.

The saul auto init module is failing to provide users access to the driver (as the device descriptors are marked as `static`). E.g. when writing a test application for a sensor driver, you cannot expose both the internal API  and the SAUL API of the driver at the same time (at least when using `saul_auto_init`, which is worth to test as well). The current architecture/API prevents me from providing a simple test application that covers the whole feature set of a sensor driver (that is: the native API, the SAUL API and the `saul_auto_init` hook). I consider this a bug that needs fixing.

> Even if this is a wanted feature it might as well be a part of the driver (e.g. status flags/mutex guarding function calls).

Quoting the [RIOT maintainer guideline](https://github.com/RIOT-OS/RIOT/blob/master/MAINTAINING.md#1---review-the-fundamentals):
> Is the solution presented in the PR as simple as possible to satisfy the requirements, but no simpler?

Adding to *every* function of the driver a check if the driver is already initialized and a conditional call to the init function increases complexity and code size, compared to providing an auto-initialize. Just ask yourself: From where should e.g. `foo_read_temperature(foo_t *dev, int16_t *temp)` take the `foo_params_t`?

>  >  So then lets fix these APIs ;-)
>
> You know yourself how hard this is

Designing good APIs is hard, let's go shopping?!?

> 1.  Is auto-initialization of sensor and actuator drivers is desired?
> 2.  Should using the native sensor/actuator API and SAUL no longer be mutually exclusive?
> 3.  And, if the first two questions are answered with yes: How should the sensor/actuator APIs be changed to achieve this?

Being completely honest: I considered the questions 1. and 2. to be rhetorical. Every use case relying on manual initialization is not harmed by having a module that *can* auto-initialize drivers: That app will just not use e.g. `USEMODULE += auto_init_sensors` the same way as it was not using `USEMODULE += auto_init_saul` so far. And 90% of the users will be happy to have the drivers ready to use with the default parameters by simply adding the auto init module.

MichelRottleuthner, 2019-07-15:
@maribu thanks for starting this RFC. I agree that its worth having this discussion and there are problems we should try to solve.
Adding my opinion:
> 1. Is auto-initialization of sensor and actuator drivers is desired?

Yes sure! But optional, IMO even on a per device level.

> 2. Should using the native sensor/actuator API and SAUL no longer be mutually exclusive?

Yes!

> And, if the first two questions are answered with yes: How should the sensor/actuator APIs be changed to achieve this?

Collecting viable solutions and comparing their ups and downs should help here ;)
As you already pointed out, other APIs handle it similarly with a dev index. A relevant difference between those cases is that the number of SPI etc. peripherals can be considered fixed for one specific platform, while external devices could be added in basically arbitrary numbers. Arguably though, you would probably know your setup beforehand...
One problem I see is that this requires you to have all devices in this one centralized array, whereas a reference to the dev is more flexible and makes it easier to (dynamically) allocate devices wherever you want them.
An option to access the low-level device handle thru SAUL, as @jcarrano mentioned above also sounds interesting.

Generally speaking: just because SAUL is simplifying/generalizing the use of devices it shouldn't keep you from accessing device specific functionality if _you know what you are doing_™. 

Recently I also had a case where my application could have benefited a lot from using SAUL with the exception of one sensor where I needed low-level access. For this case, our SAUL/auto_init/build-config procedure was missing flexibility to specify exactly what should be initialized, and what not.

Only partly related, but just in case there are overlaps of both topics we should also keep an eye on the SAUL/phydat issues regarding flexibility (e.g. dimensions) and resolution, as already described in [#10157
](https://github.com/RIOT-OS/RIOT/issues/10157)

maribu, 2019-07-15:
@MichelRottleuthner: Just to check if I understood you correctly: You're in favor of making that `static` array of device descriptors in SAUL's auto_init accessible from outside, right?

This would solve the issue to no longer have SAUL and the native API mutually exclusive (when not manually doing SAUL registration) and would not require an API change (unlike my proposal). I suggest to additionally have a `static inline foo_t *foo_get_dev(unsigned id)` function that checks if the `id` is valid. E.g. a user of that API would call `foo_read_temp(foo_get_dev(0), &dest)` instead of `foo_read_temp(foo_devs[0], &dest)`. I consider both (direct array access and helper function) as equally readable, I assume the compiler will generate the same code whenever `id` is known at compile time. But the helper function could check for out of bounds reads/writes, while directly accessing the array will not.

But I personally would still like to have auto initialization of sensors/actuators and usage of SAUL not to be linked so closely like they currently are. E.g. we could have a the modules `auto_init_sensor` and `auto_init_actuator` which would auto-initialize all sensors and actuators used (if added to `USEMODULE`). The module `auto_init_saul` would then depend on those two and would additionally also auto-register the initialized drivers to SAUL.

I like the idea of being able to blacklist certain sensors and actuators from auto-init. But using modules will add a ton of modules, so another mechanism would likely be better. How about `CFLAGS += -DNO_AUTO_INIT_FOO` to blacklist and `#ifndef NO_AUTO_INIT_FOO` ... `#endif` in the auto-init code? (Technical detail: The no-empty-compilation-unit work around should be used, obviously.) preprocessor-macros are not as beginner friendly as using modules, but the use case of using auto_init for sensors/actuators in general except for a specific driver is unlikely to jump in the way of a newbie.

jcarrano, 2019-07-15:
@maribu 
> I couldn't figure out your opinion on the above.

First, regardless of whether we want auto-init, the current design is flawed, so by adding more to auto-init we make the task of fixing harder.

> 1. Is auto-initialization of sensor and actuator drivers is desired?

I side with @SemjonKerner , it is not desired.

> 2. Should using the native sensor/actuator API and SAUL no longer be mutually exclusive?

Every time one has a "simple" and "full" interface the "simple" one should not forbid the usage of the complex one. Think of this: you write an app that uses the simple interface. At some point you realize there is something you cannot do unless you use the complex interface. How would you like it if you had to rewrite everything?

Everyone:

> Adding to every function of the driver a check if the driver is already initialized and a conditional call to the init function increases complexity and code size, compared to providing an auto-initialize

Please, please, please stop waving the code size argument, it is simply not true and it is optimizing ahead of time. Complexity is actually reduced this way: have you seen the hairy mess of `#ifdef` and build system magic that is required for auto init?

A conditional initialization on first use **may not be the best solution** but it is **not complex** and does **not increase code size** considerably:

- RAM: You need to store the initialization state: at most a word (4 bytes) and at least 0 (if you can fit a flag in some unused bit in the driver structure)
- FLASH: first, ROM is not even a problem most of the time. Second, loading a variable with the init state and conditionally calling a function can be done in about 4 to 5 thumb instructions in the worst case. That's 8 to 10 bytes but all this does not make sense since I'm not comparing it with the alternative.

On top of that, I believe that we should have a protection against initialing a driver twice, even if we do not initialize automatically on first use.

kaspar030, 2019-07-15:
> Please, please, please stop waving the code size argument, it is simply not true and it is optimizing ahead of time. 

Do you realize  **how often we wave it***? And everytime it is "just 8-10 bytes"?

>Complexity is actually reduced this way: have you seen the hairy mess of `#ifdef` and build system magic that is required for auto init?

It is using ```#ifdef MODULE_foo```. So what? That's not complex.

> A conditional initialization on first use **may not be the best solution** but it is **not complex** and does **not increase code size** considerably:

No. In RIOT, we have a pretty clear convention on when to use any object pointer or periphal: it is only allowed to do so if it has been initialized through the corresponding ```_init()``` function, if that exists.
We're not going to add NULL pointer checks, ```driver_foo_is_initialized(ptr)```, ... to *every driver function call*. That is just dead code for any deployment.

kaspar030, 2019-07-15:
For the auto-init question, how about this:

- drivers provide auto-init as seperate (pseudo-)module auto_init_driver_foo
- auto_init_default depends on all auto_init_driver_foo
- saul depends on auto_init_driver_foo for every sensor it abstracts, expects the corresponding drivers to be initialized
- auto_init_driver_foo can be disabled

So, *by default*, any driver(-instance) is automatically initialized. Saul can depend on that. Powerusers can disable driver auto init. Everyone happy, and RIOT can still be considered friendly.

MichelRottleuthner, 2019-07-15:
>@MichelRottleuthner: Just to check if I understood you correctly: You're in favor of making that static array of device descriptors in SAUL's auto_init accessible from outside, right?

Yes.

> I suggest to additionally have a static inline foo_t *foo_get_dev(unsigned id) (...)

I'm not sure if I understand. This is a foo-device specific function, right? Then how does this work together with app allocated dev/param variables? Or is this only meant for (currently) auto-init allocated devs?

>But I personally would still like to have auto initialization of sensors/actuators and usage of SAUL not to be linked so closely like they currently are.

I agree. But I'm against expressing parts of this configuration as modules and other parts with custom blacklist defines like `#ifndef NO_AUTO_INIT_FOO`. As currently modules are the way to include specific functionalities I'd say it's would be more consistent to stick with that. What would be the big problem with having "tons of modules" anyway? I don't think that this will add excessive complexity to the build.

>I side with @SemjonKerner , it is not desired.

To clarify: You think this shouldn't even be available as an option? (why? because you consider it wrong to do that?)

>Every time one has a "simple" and "full" interface the "simple" one should not forbid the usage of the complex one.

I strongly agree. But also, to me an auto-init of sensors/actuators would be such a simplification where I would expect that it is still possible to NOT use it and keep full control (i.e. application init) if you need it.

>(..) hairy mess of #ifdef and build system magic that is required for auto init

To what exactly are you referring here? I don't think that the relevant code/preprocessor stuff is beautiful but I also wouldn't call it hairy mess or magic. Did I maybe miss something?

>I believe that we should have a protection against initialing a driver twice, even if we do not initialize automatically on first use.

In which case is that important? How would one ever expect that an application reliably works, if you are not sure when and how often an init call is or may be executed?

maribu, 2019-07-15:
> I'm not sure if I understand. This is a foo-device specific function, right? Then how does this work together with app allocated dev/param variables? Or is this only meant for (currently) auto-init allocated devs?

Excatly. If the use wants to access an auto-initialized device by its index, I would like to provide a static inline function to access the array element. (For non-auto initialized devices, the user will pass the pointer to the manually initialized device descriptor directly, instead of calling `foo_get_dev()`.)

> But I'm against expressing parts of this configuration as modules and other parts with custom blacklist defines like #ifndef NO_AUTO_INIT_FOO. As currently modules are the way to include specific [...]

You're right. I didn't think it through: E.g. `make info-modules` will only list the modules actually *used*. So my fear that too many *available* modules will be confusing/overwhelming is pointless.

maribu, 2019-07-15:
I implemented the current state of the discussion for the SHT1X driver here: https://github.com/maribu/RIOT/commits/sht1x_auto_init

Sometimes it helps to see an actual implementation of a proposed change in order to realize if it works out as expected or if some things could be handled better.

maribu, 2019-07-19:
So with the discussion having become silent, I assume everyone agreed upon [@kaspar030's proposal](https://github.com/RIOT-OS/RIOT/issues/11826#issuecomment-511358113).

Please check if the implementation of that proposal in #11871 matches how everyone interpreted that proposal. I also provided a separate PR in https://github.com/RIOT-OS/RIOT/pull/11872 that adapts the SHT1X driver to that API, so that discussion and testing of the new auto init based on an actual user of the new interface.