RIOT-OS/RIOT #11852: scan-build errors found during 2019.07 testing

cladmi, 2019-07-16:
#### Description

I am running `scan-build` checking for the 2019.07 release https://github.com/RIOT-OS/Release-Specs/issues/128

I will try to list the issues I found here.

```
# Same boards as `RIOT` building for `llvm`
BOARDS="iotlab-m3 native nrf52dk mulle nucleo-f401re samr21-xpro slstk3402a"

export BUILD_IN_DOCKER=1
export TOOLCHAIN=llvm

# Cannot currently be done through `RIOT_MAKEFILES_GLOBAL_POST` with `docker`
# So hack it
echo 'CFLAGS += -Wno-deprecated-declarations' >> Makefile.include

for board in ${BOARDS}
do
    ./dist/tools/compile_and_test_for_board/compile_and_test_for_board.py --compile-targets scan-build-analyze --no-test . ${board} || true
done
```

The parsed result will be available here tomorrow: https://ci-ilab.imp.fu-berlin.de/job/RIOT%20scan-build-analyze/18/

The previous one without `deprecated-declarations' removed: https://ci-ilab.imp.fu-berlin.de/job/RIOT%20scan-build-analyze/16/

#### List of found issues:

TODO

#### Versions

2019.07-RC1

miri64, 2019-09-19:
For the `gnrc_ipv6` related parts I provided https://github.com/RIOT-OS/RIOT/pull/12274.

miri64, 2019-09-20:
Sorry, wasn't careful enough about my wording in #12274.

PR #12274 (2019-09-20): gnrc_ipv6: fix scan-build errors

### Contribution description
Fixes #11852 for `gnrc_ipv6`

### Testing procedure
Run

```C
TOOLCHAIN=llvm make -C examples/gnrc_networking scan-build-analyze
```

for `native`. All `gnrc_ipv6` related files should now not trigger an error anymore (`sys/include/net/ipv6/addr.h` is not part of `gnrc_ipv6` ;-)).

CI should still pass.

### Issues/PRs references
Addresses parts of #11852.