RIOT-OS/RIOT #11860: send data with UDP at 10HZ, the program die

jhx2002, 2019-07-18:
setup:

- hardware:  stm32f4+enc28j60

- core code: 
```
void send(sender_t* dev) {
    assert(dev != NULL);
    char* buf = "hi";
    size_t res = sock_udp_send(NULL, buf, strlen(buf), &gw);                                                                                                                                                           
    printf("send %d\n", res);
}

static void *main_task(void *arg)
{
    assert(arg != NULL);

    sender_t* dev = (sender_t*)arg;                                                                                                                                                                

    /* main loop */
    int count = 0;
    while(1) {
        send(dev);
        count ++ ;
        xtimer_usleep(100000); //100ms                                                                                                                                                                             
    }

    return NULL;    /* should never be reached */
}

```

- logs, program dead after sending 8 package
```
2019-07-18 10:56:58,513 - INFO # gateway: fe80::f58c:2fb7:adad:1
2019-07-18 10:56:58,516 - INFO # local: fe80::f58c:2fb7:adad:10
2019-07-18 10:56:58,517 - INFO # send 2
> 2019-07-18 10:56:58,617 - INFO #  send 2
2019-07-18 10:56:58,718 - INFO # send 2
2019-07-18 10:56:58,818 - INFO # send 2
2019-07-18 10:56:58,919 - INFO # send 2
2019-07-18 10:56:59,020 - INFO # send 2
2019-07-18 10:56:59,120 - INFO # send 2
2019-07-18 10:56:59,221 - INFO # send 2
2019-07-18 10:56:59,322 - INFO # send 2
2019-07-18 10:56:59,422 - INFO # send 2
ps

```

PeterKietzmann, 2019-07-18:
- Did you confirm the Ethernet frames were actually on cable?
- Can you share a branch that contains your application so we can easily reproduce the issue?

miri64, 2019-07-18:
[I tried to recreate](https://github.com/RIOT-OS/RIOT/compare/master...miri64:tests/enh/gnrc_sock_udp_high_freq) your test application as close as possible from the snippet you gave us. I was not able to reproduce this on `iotlab-m3` (`stm32f3`-based) with `ethos` (Ethernet over serial) as Ethernet interface:

```
----> ethos: sending hello.
----> ethos: activating serial pass through.
----> ethos: hello reply received
send [fe80::949c:a8ff:fe12:aa37]:1337
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
hsend 2
esend 2
lsend 2
send 2
psend 2
send 2

help
Command              Description
---------------------------------------
send                 starts sending buffer every 100ms
reboot               Reboot the node
random_init          initializes the PRNG
random_get           returns 32 bit of pseudo randomness
nib                  Configure neighbor information base
ifconfig             Configure network interfaces
> send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
send 2
[…]
```

(using help `because` I did not include the `ps` module in the version I pushed, but I was not able to reproduce with `ps` either).

miri64, 2019-07-18:
@PeterKietzmann can you try with the hardware @jhx2002 specified, please?

PeterKietzmann, 2019-07-18:
I will, once @jhx2002 shred his branch and I have access to the hardware (not before tomorrow or monday)

miri64, 2019-07-18:
You can use my branch. It should have the same behavior ;-).

jhx2002, 2019-07-18:
I push the test code here:  
```
https://github.com/jhx2002/test_riot/tree/master/udp
```

kaspar030, 2019-07-18:
> I push the test code here:
> 
> ```
> 
> ```

Your thread is using printf in ```send()```. That might blow the stack. Can you try increasing it [here](https://github.com/jhx2002/test_riot/blob/a2086ff813a06cd5db5138cd9f791fffd8c91ccb/udp/sender/sender.c#L13), for testing to maybe "THREAD_STACKSIZE_DEFAULT*2"?

jhx2002, 2019-07-18:
> @jhx2002 is this related to #11811 or a separate issue?

It's related. I simplify the debug using this test case.

jhx2002, 2019-07-18:
> @jhx2002 thanks. Last thing: whats the RIOT commit you are working on?

RELEASE VERSION  2019.04

jhx2002, 2019-07-18:
> > I push the test code here:
> > ```
> > 
> > ```
> 
> Your thread is using printf in `send()`. That might blow the stack. Can you try increasing it [here](https://github.com/jhx2002/test_riot/blob/a2086ff813a06cd5db5138cd9f791fffd8c91ccb/udp/sender/sender.c#L13), for testing to maybe "THREAD_STACKSIZE_DEFAULT*2"?

It's same after setting stacksize to  "THREAD_STACKSIZE_DEFAULT*2"

miri64, 2019-07-18:
> > @jhx2002 is this related to #11811 or a separate issue?
> 
> It's related. I simplify the debug using this test case.

Ok, so is this superseding #11811 or are you unsure if other reasons for hanging might be the reason for #11811. Mainly asking to know if we can close #11811.

jhx2002, 2019-07-18:
> > > @jhx2002 is this related to #11811 or a separate issue?
> > 
> > 
> > It's related. I simplify the debug using this test case.
> 
> Ok, so is this superseding #11811 or are you unsure if other reasons for hanging might be the reason for #11811. Mainly asking to know if we can close #11811.

Yes, #11811 can be closed.

PeterKietzmann, 2019-07-22:
I found your pin configuration [here](https://github.com/jhx2002/test_riot/blob/master/udp/Makefile#L41) which won't work because PA2/PA3 are used for UART / STDIO on that board (see [periph_conf.h](https://github.com/RIOT-OS/RIOT/blob/master/boards/stm32f4discovery/include/periph_conf.h#L66)). Please correctz that and report if your issue persists

PeterKietzmann, 2019-07-22:
Can you please paste the output of `ifconfig` here?

jhx2002, 2019-07-22:
> I found your pin configuration [here](https://github.com/jhx2002/test_riot/blob/master/udp/Makefile#L41) which won't work because PA2/PA3 are used for UART / STDIO on that board (see [...]

I'v reconfig to P8 and P9 , refer to https://github.com/jhx2002/test_riot/blob/master/udp/Makefile

PeterKietzmann, 2019-07-22:
Can you please point to the line where PA8/9 are set? I don't see it. Or do you do that when calling `make`?

PeterKietzmann, 2019-07-22:
And please paste output from `ifconfig` here

jhx2002, 2019-07-22:
> Can you please point to the line where PA8/9 are set? I don't see it. Or do you do that when calling `make`?

37 line: 
```
# stdio
# uart port: PD8, PD9
STDIO_DEV = UART_DEV\(1\)
CFLAGS += -DSTDIO_UART_DEV=$(STDIO_DEV)
```

jhx2002, 2019-07-22:
> And please paste output from `ifconfig` here

```
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-07-22 16:31:02,137 - INFO # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type '/exit' to exit.
2019-07-22 16:31:04,846 - INFO # main(): This is RIOT! (Version: 2019.04)
2019-07-22 16:31:04,849 - INFO # Vehicle control execute 
2019-07-22 16:31:04,849 - INFO # 
2019-07-22 16:31:04,852 - INFO # gateway: fe80::f58c:2fb7:adad:1
2019-07-22 16:31:04,854 - INFO # local: fe80::f58c:2fb7:adad:10
> ifconfig
2019-07-22 16:31:08,448 - INFO #  ifconfig
2019-07-22 16:31:08,453 - INFO # Iface  5  HWaddr: 4A:16:2B:13:1B:53  Link: up 
2019-07-22 16:31:08,458 - INFO #           L2-PDU:1500 MTU:1500  HL:64  RTR  
2019-07-22 16:31:08,458 - INFO #           RTR_ADV  
2019-07-22 16:31:08,461 - INFO #           Source address length: 6
2019-07-22 16:31:08,464 - INFO #           Link type: wired
2019-07-22 16:31:08,470 - INFO #           inet6 addr: fe80::4816:2bff:fe13:1b53  scope: local  VAL
2019-07-22 16:31:08,477 - INFO #           inet6 addr: fe80::f58c:2fb7:adad:10  scope: local  VAL
2019-07-22 16:31:08,479 - INFO #           inet6 group: ff02::2
2019-07-22 16:31:08,481 - INFO #           inet6 group: ff02::1
2019-07-22 16:31:08,487 - INFO #           inet6 group: ff02::1:ff13:1b53
2019-07-22 16:31:08,491 - INFO #           inet6 group: ff02::1:ffad:10
2019-07-22 16:31:08,491 - INFO #           
> 

```

PeterKietzmann, 2019-07-22:
I can't reproduce your issue. I get:
```
2019-07-22 17:22:26,498 - INFO # main(): This is RIOT! (Version: 2019.04)
2019-07-22 17:22:26,500 - INFO # Vehicle control execute 
2019-07-22 17:22:26,500 - INFO # 
2019-07-22 17:22:26,504 - INFO # gateway: fe80::f58c:2fb7:adad:1
2019-07-22 17:22:26,506 - INFO # local: fe80::f58c:2fb7:adad:10
2019-07-22 17:22:26,506 - INFO # send 2
```
and `send 2` continues 'endlessly'. Any ideas @jhx2002?

What is your compiler version? Did you double check that there are no other (unstaged) changes in your repository?

jhx2002, 2019-07-23:
> I can't reproduce your issue. I get:
>
> ```
> 2019-07-22 17:22:26,498 - INFO # main(): This is RIOT! (Version: 2019.04)
> 2019-07-22 17:22:26,500 - INFO # Vehicle control execute
> 2019-07-22 17:22:26,500 [...]

how long did you run the test case

jhx2002, 2019-07-23:
The case has ramdonness.  It may die at different duration.

PeterKietzmann, 2019-07-23:
- Yes I used "exactly" your code except the `RIOTBASE` variable that I hat to change.
- The program ran (currently runs) for minutes. And it didn't stop – I just stopped watching.
- Where does the test include randomness?
- Now I built with docker too but didn't face any issues either.

I assume your setup is as fiddly as mine with many cheap jumper cables. Double check that all connections are good and try to wobble the cables a bit while you  are connected to the console.

jhx2002, 2019-07-23:
May you 

> * Yes I used "exactly" your code except the `RIOTBASE` variable that I hat to change.
>
>     * The program ran (currently runs) for minutes. And it didn't stop – I just stopped watching.
>
>     * Where [...]

Can you test it with high freqency, such as 100HZ？
Can you share your pin mapping between enc28j60 and stm32f4 ?
If I add xtimer_sleep(50)  in the end of  while loop in udp、 ipv6 and enc28j60 ,  the program will run much longger.  Some times it could  run about 15 miniutes , and it still die.

PeterKietzmann, 2019-07-23:
- I don't use your docker setup or build script if that makes any difference? Why don't you use the easy RIOT integration `BUILD_IN_DOKCER=1 make` ? (See [here](https://github.com/RIOT-OS/RIOT/wiki/Use-Docker-to-build-RIOT) for more detail)
- You do already set 100Hz [here](https://github.com/jhx2002/test_riot/blob/master/udp/sender/sender.c#L49) by sleeping 10ms. However, I set sleep to 1ms and still works
- I use a stm32f4discovery which contains a STM32F407VG (not STM32407VET6)
- My pins are connected as you set up in the Makefile. Furthermore, 
VCC - 3V
GND -GND
CLKOUT - open
WOL - open
- I figured some other weird thing right now. Sometimes when press reset while running, the program doesn't start running again until I wobble a little bit the jumper cables (I assume it is VCC)

PeterKietzmann, 2019-07-23:
BTW: I did not yet check frames on the line. So far this is just about continuously running the loop in you demo program.

jhx2002, 2019-07-24:
> * I don't use your docker setup or build script if that makes any difference? Why don't you use the easy RIOT integration `BUILD_IN_DOKCER=1 make` ? (See [...]

- can you test how long it could continuously  run ? 

> 

> * I use a stm32f4discovery which contains a STM32F407VG (not STM32407VET6)
>
>     * My pins are connected as you set up in the Makefile. Furthermore,
>       VCC - 3V
>       GND -GND
>       CLKOUT - [...]

PeterKietzmann, 2019-07-24:
After I forgot that I started it this morning I can say that the board kept printing "send 2" every 10ms for almost 5 hours now.

How do we proceed now? We should make sure to have **very similar** (not to say exact) setups. For that:
- Do you have access to a stm32f4discovery board?
- Can you setup the RIOT docker so we utilize the same compiler
- Maybe we should go with a minimal example such [this one](https://riot-os.org/api/group__net__sock__udp.html)

jhx2002, 2019-07-24:
- I'll change hardware to test. 
- I test in a real local network.  One ubuntu server  connect to a ethernet switch, and enc28j60 connect to the same ehternet switch.  Udp package is sent to ubuntu server .

PeterKietzmann, 2019-07-25:
Give me a sign once you have your stm32f4disco connected

jhx2002, 2019-07-26:
I'v tested on another two enc28j60 on release version 2019.07.   The issue still exist. 
 Discovery board is on the way.

jhx2002, 2019-07-27:
I tested on stm32f4discovery board.  The issue still exist.

jhx2002, 2019-07-27:
I guess the problem root is in schedule algorithm.  It's complicate, it involves {ISRs, threads, msgs }.

PeterKietzmann, 2019-07-29:
> I test in a real local network. One ubuntu server connect to a ethernet switch, and enc28j60 connect to the same ehternet switch. Udp package is sent to ubuntu server . Udp receiver could receive package.

Can you please not do that in first place? With your stm43f4disco + encj28j60 **only** does your setup still hang after some iterations?

jhx2002, 2019-07-29:
> > I test in a real local network. One ubuntu server connect to a ethernet switch, and enc28j60 connect to the same ehternet switch. Udp package is sent to ubuntu server . Udp receiver could receive package.
> 
> Can you please not do that in first place? With your stm43f4disco + enc28j60 **only** does your setup still hang after some iterations?

If I disconnect enc28j60 from ethernet switch and there is no actual package sent, the program not die , keep running.

jhx2002, 2019-07-29:
You could use Udp receiving program to reproduce the issue.
Build and run this on ubuntu  https://github.com/jhx2002/test_riot/tree/master/udp_recv

PeterKietzmann, 2019-07-30:
> If I disconnect enc28j60 from ethernet switch ...

So we've been testing different things the whole time... Right now I don't have an extra switch lying around but I'll try to get one to set it up .

Please enable `DEBUG` messages [here](https://github.com/RIOT-OS/RIOT/blob/master/drivers/enc28j60/enc28j60.c#L35) to see whats the last event that happens in the Ethernet shield driver. 

On the sender I recommend to transmit packets that contain unique identifiers. You could simply add a counter value to the payload [here](https://github.com/jhx2002/test_riot/blob/master/udp/sender/sender.c#L27). On the receiver, I'd recommend to look at the packets via wireshark and then try to find out which one was the last.

jhx2002, 2019-07-31:
DEBUG will slower the sending, and can not reproduce the issue, I'v tried this.

I'v made another test :  Directly connect enc28j60 and linux ethernet interface, the program die also.

jhx2002, 2019-08-02:
You can reproduce the issue without switch.

PeterKietzmann, 2019-08-09:
Just to confirm your issue and to make it easier reproducible to others: Could you run *tests/gnrc_networking* and use the `udp send` command and latest RIOT release to run into the deadlock? Don't forget to add you pin configuration as well as `USEMODULE=enc28j60` to the Makefile. 

Unfortunately I face several issues right now (could be wiring, side traffic on my switch, buggy drivers .... ) and I currently don't have the time to sort them all.

@maribu, @gschorcht IIRC you have used the enc28j60 recently. Did everything work as expected? Can you share some insights on HW/SW setup and scenario?

PeterKietzmann, 2019-08-09:
Additionally, can you confirm that the issue does not relate to #9784?

jhx2002, 2019-08-10:
> Just to confirm your issue and to make it easier reproducible to others: Could you run _tests/gnrc_networking_ and use the `udp send` command and latest RIOT release to run into the deadlock? Don't [...]

- You mean https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_networking ? This code only send one package when got a command.  It is the continuously sending that cause program die.

jhx2002, 2019-08-10:
> Additionally, can you confirm that the issue does not relate to #9784?

- Not relate

gschorcht, 2019-08-10:
I have not used my hardware setup with `esp8266`/`esp32` and `enc28j60` in the last few months. So I tried @jhx2002's `test_riot/udp` program with release `2019.04` as well as current master.

In both configurations, the `send 2` output stops after an arbitrary time. Sometimes it takes hours, sometimes only minutes. If debugging is enabled for the driver `enc28j60`, I can observe that only the output of `send 2` seems to stop while the communication stack continues to work. I can then continue to observe the following message.
```
[enc28j60] isr: packet received
[enc28j60] recv: size=86 next=3886 buf=0 len=0
[enc28j60] recv: size=86 next=3886 buf=0x3ffef03c len=86
[enc28j60] isr: packet transmitted
```
Furthermore, I can still ping the node from the LAN.

jhx2002, 2019-08-13:
So , how to debug enc28j60 driver ?

jhx2002, 2019-08-17:
Is it ok ?  if I change enc28j60 to w5100 ?

gschorcht, 2019-08-17:
> Is it ok ? if I change enc28j60 to w5100 ?

????

gschorcht, 2019-08-17:
> Thank you @george-hopkins , Have you have any idea on how to debug enc28j60 driver ?

Probably you have used wrong user name! The only way to debug enc28j60 driver is to set `ENABLE_DEBUG` to 1 and to add additional `printf`s.

jhx2002, 2019-08-17:
> > Is it ok ? if I change enc28j60 to w5100 ?
> 
> ???
If it could run reliably  with w5100, I could replace enc28j60 with w5100 network device.
But I have not w5100 on hand.

jhx2002, 2019-08-17:
> > Thank you @george-hopkins , Have you have any idea on how to debug enc28j60 driver ?
> 
> Probably you have used wrong user name! The only way to debug enc28j60 driver is to set `ENABLE_DEBUG` to 1 and to add additional `printf`s.

Sorry and thank you @gschorcht

gschorcht, 2019-08-17:
> > > Is it ok ? if I change enc28j60 to w5100 ?
> > 
> > 
> > ???
> > If it could run reliably  with w5100, I could replace enc28j60 with w5100 network device.
> > But I have not w5100 on hand.

I can't give any statement about `w5100` driver since I'm having the hardware.

PeterKietzmann, 2019-08-19:
> This code only send one package when got a command

It includes the `ping6` command which which lets you set the number of pings, the payload size and an interval.

> Furthermore, I can still ping the node from the LAN.

That is weird. @jhx2002 can you confirm that? I would start analyzing the ISR and how the hardware interrupt is being reset.

> So maybe it's even related to the enc28j60 driver, but I have no idea how.

I'd be surprised if it was a layer above. BTW thanks @gschorcht for the detailed analysis.

@jhx2002 I do not currently find the time to debug on that level. Do you have a logic analyzer or oscilloscope at hand?

jhx2002, 2019-08-19:
> It includes the `ping6` command which which lets you set the number of pings, the payload size and an interval.
>
>
> That is weird. @jhx2002 can you confirm that? I would start analyzing the ISR and [...]

I'v no  logic analyzer and oscilloscope, But w5100 is on road. I'll test it with w5100.

maribu, 2019-08-19:
> Just to confirm your issue and to make it easier reproducible to others: Could you run _tests/gnrc_networking_ and use the `udp send` command and latest RIOT release to run into the deadlock? Don't [...]

Sorry for the late reply. I also experienced random hangs in the `enc28j60`. Some were caused by the missing drop feature of the `netdev_driver_t::recv()` function, which was fixed by @gschorcht. With that fixed, the mean time to failure increased to about a minute under load for me. I didn't have time to further investigate and have no longer a use case for it.

The `w5100` worked more reliable to me, but fails to initialize on a warm boot. It reads the TMSR register **before** resetting the device and expects it to have the default value. But the TMSR value is set to a value different from the default during initialization, which persists on reboot of a RIOT node. I also did not yet have time to fix that bug, but maybe someone else has. The issue is easy to reproduce and the solution should be as simple as moving the reset before the connectivity test.

jhx2002, 2019-08-21:
Is there any difference of pin wiring and ip address setting  between w5100 and enc28j60  ?  @maribu 
I try to test w5100, but can not ping6 successfully.

jhx2002, 2019-08-26:
Can you share the fixing code of netdev_driver_t::recv() function ?  @gschorcht

gschorcht, 2019-08-26:
The fix was merged into the master with PR #9806. That is, it is already included in the current driver code.

jhx2002, 2019-08-27:
The issue is difficult to debug, I'm crazy :)

jhx2002, 2019-08-30:
Do you have any plan to debug?  @PeterKietzmann

PeterKietzmann, 2019-08-30:
Not really. I don't find any time for this atm. As indicated above: if the issue is at a very low layer (and this seems to be the case) then you should connect a logic analyzer and find where exactly the driver hangs-up. You trace the code to this point and eventually check the the last state transitions. Take the reference manual and compare if all transitions were valid.   That would be my first approach

[12 more replies]

closed 2023-05-18 by maribu