RIOT-OS/RIOT #11861: msba2: floating point tests fail

miri64, 2019-07-18:
#### Description

The floating point unittests fail on `msba2`

#### Steps to reproduce the issue
Get your hands on an `msba2`, check out current master (I was on 8150b014b17b25dc when testing), compile, flash, and run the floating point unittests to the board:

```sh
BOARD=msba2 make -C tests/unittests/ tests-scanf_float flash-only test
```

#### Expected results
Unittests should pass

```
main(): This is RIOT! (Version: 2019.10-devel-70-g8150b)
......
OK (6 tests)
```

#### Actual results
Tests fail:

```
2019-07-18 12:24:21,531 - INFO # main(): This is RIOT! (Version: 2019.10-devel-70-g8150b)
2019-07-18 12:24:21,531 - INFO # .
2019-07-18 12:24:21,532 - INFO # scanf_float_tests.test_f (tests/unittests/tests-scanf_float/tests-scanf_float.c 37) exp 0 was 1
2019-07-18 12:24:21,532 - INFO # .
2019-07-18 12:24:21,533 - INFO # scanf_float_tests.test_e (tests/unittests/tests-scanf_float/tests-scanf_float.c 38) exp 0 was 1
2019-07-18 12:24:21,533 - INFO # .
2019-07-18 12:24:21,534 - INFO # scanf_float_tests.test_E (tests/unittests/tests-scanf_float/tests-scanf_float.c 39) exp 0 was 1
2019-07-18 12:24:21,534 - INFO # .
2019-07-18 12:24:21,535 - INFO # scanf_float_tests.test_g (tests/unittests/tests-scanf_float/tests-scanf_float.c 40) exp 0 was 1
2019-07-18 12:24:21,535 - INFO # .
2019-07-18 12:24:21,536 - INFO # scanf_float_tests.test_sign (tests/unittests/tests-scanf_float/tests-scanf_float.c 45) exp 0 was 1
2019-07-18 12:24:21,536 - INFO # .
2019-07-18 12:24:21,538 - INFO # scanf_float_tests.test_scientific (tests/unittests/tests-scanf_float/tests-scanf_float.c 53) exp 0 was 1
2019-07-18 12:24:21,538 - INFO #
2019-07-18 12:24:21,538 - INFO # run 6 failures 6
```

#### Versions

RIOT: 8150b014b17b25dc

```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "19.04 (Disco Dingo)"
                 Kernel: Linux 5.0.0-20-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: clang version 8.0.0-3 (tags/RELEASE_800/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.13.4
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.7.3 on Linux
                    git: git version 2.20.1
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0+dev-00910-g4dbcb1e7 (2019-06-17-16:24)
                 python: Python 2.7.16
                python2: Python 2.7.16
                python3: Python 3.7.3
             coccinelle: missing
```

maribu, 2019-07-19:
I can reproduce the issue on my system. The unit tests fails as `sscanf()` does not return the correct number of items parsed. Even `printf("pi is approx. %f\n", 3.14159);` will not work on the MSB-A2. The reason is that newlib seems to not include floating point support in stdio by default for ARM7 MCUs. (Floating point support can be configured at compile time of newlib.)

I'm not sure how we should handle this. A user of the MSB-A2 who really *wants* to use stdio with floating point support could set up the toolchain that way.  But there will be users of the MSB-A2 who will actually prefer to not have float support in stdio to preserve ROM.

The best we could do (if we really *want* float support in stdio) would be to update the riotbuild docker to use a newlib version that provides this.

I'm personally not a fan of stdio at all. I consider it a poor API design that has proved a [source](https://www.openwall.com/lists/musl/2019/03/14/1) [of](https://nvd.nist.gov/vuln/detail/CVE-2018-1000101) [security](https://www.freebsd.org/security/advisories/FreeBSD-SA-14:27.stdio.asc) [relevant](https://www.cvedetails.com/cve/CVE-2012-3406/) [bugs](https://bugzilla.redhat.com/show_bug.cgi?id=1548934), a [common attack vector](https://en.wikipedia.org/wiki/Uncontrolled_format_string), and bloat for ages. If that would get finally removed from the C standard my scream of joy would likely be audible in all of the the northern hemisphere.

Some people seem to share that view, e.g. in MISRA C stdio is [forbidden in production code](https://rules.sonarsource.com/c/RSPEC-988) :-)

And RIOT already has an alternative: E.g. [`fmt_float()`](http://api.riot-os.org/group__sys__fmt.html#ga9578cea845911b8599501348ad8e0eec) allows to only link in float support if an application uses it.

So how about a wont-fix on this one?

miri64, 2019-07-19:
I'm just the messanger, but the ARM Cortex builds also use newlib and don't have that problem. Maybe we can have a look how it is solved there. IIRC there were pseudo-modules for that. Maybe the include of such a module is just missing for test for the ARM7 platforms?

maribu, 2019-07-19:
> I'm just the messanger

Please don't feel adressed by my stdio rants.

> IIRC there were pseudo-modules for that

That's a nice solution: It will make users of stdio happy without adding the bloat to everyone else :-)

As a wild guess I would think that the nano variant is used for linking unless those stdio modues are pulled in. I'll have a look how it's done for Cortex and will try to replicate for ARM7.

maribu, 2019-07-22:
It is btw even handled better than I would have guessed :-) The nano version is still used even with the pseudo modules `scanf_float` and `printf_float` used. It sets linker flags to add float support without switching away from nano :-)

The problem is that the ARM7 build system needs some love. I tried to apply some love in https://github.com/RIOT-OS/RIOT/pull/11882, but being honest, I only have a very limited understanding (basic at best) of RIOT's build system.

maribu, 2019-07-22:
Fixed in https://github.com/RIOT-OS/RIOT/pull/11883. Thanks for reporting!

And sorry for not taking care to point out that my stdio rant was directed against the API and not any person.

closed 2019-07-22 by maribu

miri64, 2019-07-22:
> And sorry for not taking care to point out that my stdio rant was directed against the API and not any person.

Not understood as such so no harm done :relaxed: