RIOT-OS/RIOT #11862: [Missing Feature ?]TCP API Echo Example Does not Build for ESP8266

mmaxus35, 2019-07-18:
I had an issue about UDP Echo Server and Client. It is already solved because of module includes.
However this time i am sure that i have included necessary modules.
Board is esp8266-esp-12x.  I  thought that there might be a problem on TCP port for esp8266?

The code is offical example in TCP Api in [https://riot-os.org/api/group__net__sock__tcp.html#ga0144778bc074e291fcb0d43bd35c0179](url)

My makefile is given below
 name of your application
APPLICATION = tcp-simple-server
 If no BOARD is found in the environment, use this default:
BOARD ?= esp8266-esp-12x
 This has to be the absolute path to the RIOT base directory:
RIOTBASE ?= $(CURDIR)/../..
BOARD_INSUFFICIENT_MEMORY := arduino-duemilanove arduino-leonardo \
                             arduino-mega2560 arduino-nano \
                             arduino-uno blackpill bluepill calliope-mini \
                             chronos hifive1 i-nucleo-lrwan1 mega-xplained \
                             microbit msb-430 msb-430h \
                             nucleo-f031k6 nucleo-f042k6 nucleo-f303k8 \
                             nucleo-l031k6 nucleo-f030r8 nucleo-f070rb \
                             nucleo-f072rb nucleo-f103rb nucleo-f302r8 \
                             nucleo-f334r8 nucleo-l053r8 saml10-xpro \
                             saml11-xpro spark-core stm32f0discovery \
                             stm32l0538-disco telosb \
                             waspmote-pro wsn430-v1_3b wsn430-v1_4 z1
 Include packages that pull up and auto-init the link layer.
 NOTE: 6LoWPAN will be included if IEEE802.15.4 devices are present
USEMODULE += gnrc_netdev_default
USEMODULE += auto_init_gnrc_netif
 Activate ICMPv6 error messages
USEMODULE += gnrc_icmpv6_error
 Specify the mandatory networking modules for IPv6 and UDP
USEMODULE += gnrc_ipv6_router_default
USEMODULE += gnrc_sock_check_reuse # gnrc_udp, gnrc_sock_udp , gnrc_ipv6 
USEMODULE += gnrc_sock_tcp  
USEMODULE += gnrc_tcp
USEMODULE += gnrc_ipv6_default
USEMODULE += gnrc_ipv6
 Add a routing protocol
USEMODULE += gnrc_rpl
USEMODULE += auto_init_gnrc_rpl
This application dumps received packets to STDIO using the pktdump module
USEMODULE += gnrc_pktdump
 Additional networking modules that can be dropped if not needed
USEMODULE += gnrc_icmpv6_echo
 Add also the shell, some shell commands
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps
USEMODULE += netstats_l2
USEMODULE += netstats_ipv6
USEMODULE += netstats_rpl
...
....
`
And output of the fail build is given below ;

`/home/riot/RIOT-master/examples/tcp-simple-server/server.c:6:12: error: array type has incomplete element type
 sock_tcp_t sock_queue[SOCK_QUEUE_LEN];
            ^
/home/riot/RIOT-master/examples/tcp-simple-server/server.c: In function 'main':
/home/riot/RIOT-master/examples/tcp-simple-server/server.c:15:22: error: storage size of 'queue' isn't known
     sock_tcp_queue_t queue;
                      ^
/home/riot/RIOT-master/examples/tcp-simple-server/server.c:30:9: error: too few arguments to function 'sock_tcp_accept'
         if (sock_tcp_accept(&queue, &sock) < 0) {
         ^
In file included from /home/riot/RIOT-master/examples/tcp-simple-server/server.c:2:0:
/home/riot/RIOT-master/sys/include/net/sock/tcp.h:474:5: note: declared here
 int sock_tcp_accept(sock_tcp_queue_t *queue, sock_tcp_t **sock,
     ^
/home/riot/RIOT-master/examples/tcp-simple-server/server.c:15:22: error: unused variable 'queue' [-Werror=unused-variable]
     sock_tcp_queue_t queue;
                      ^
cc1: all warnings being treated as errors
/home/riot/RIOT-master/Makefile.base:83: recipe for target '/home/riot/RIOT-master/examples/tcp-simple-server/bin/esp8266-esp-12x/application_tcp-simple-server/server.o' failed
make[1]: *** [/home/riot/RIOT-master/examples/tcp-simple-server/bin/esp8266-esp-12x/application_tcp-simple-server/server.o] Error 1
/home/riot/RIOT-master/examples/tcp-simple-server/../../Makefile.include:478: recipe for target '/home/riot/RIOT-master/examples/tcp-simple-server/bin/esp8266-esp-12x/application_tcp-simple-server.a' failed
make: *** [/home/riot/RIOT-master/examples/tcp-simple-server/bin/esp8266-esp-12x/application_tcp-simple-server.a] Error 2
make: Leaving directory '/home/riot/RIOT-master/examples/tcp-simple-server'`

Thanks for your help @gschorcht

crest42, 2019-07-19:
Possible duplicate of https://github.com/RIOT-OS/RIOT/issues/10664

mmaxus35, 2019-07-19:
Definetely duplicate of #10664 i close this issue. Hope that these features get implemented in soon.

closed 2019-07-19 by mmaxus35