RIOT-OS/RIOT #11867: Cannot build hello world example

vasav123, 2019-07-19:
#### Description

I made a fresh checkout of the latest RIOT repo, and used vagrant to setup the enviornment on a vm. Then I went to examples, and did just make. However I was unable to compile.

#### Steps to reproduce the issue

1) checkout a fresh version of RIOT 
2) vagrant up
3) vagrant ssh
4) go to examples-> hello world
5) make
#### Expected results

#### Actual results

Building application "default" for "native" with MCU "native".

/usr/bin/env: 'python3\r': No such file or directory
/bin/sh: 1: /home/vagrant/RIOT/dist/tools/genconfigheader/genconfigheader.sh: not found
/home/vagrant/RIOT/examples/default/../../Makefile.include:732: recipe for target '/home/vagrant/RIOT/examples/default/bin/native/riotbuild/riotbuild.h' failed
make: *** [/home/vagrant/RIOT/examples/default/bin/native/riotbuild/riotbuild.h] Error 127

#### Versions

Windows, with virtual box, and vagrant.

vasav123, 2019-07-19:
I think the issue was the files were checkout as windows files which have different line endings. So i fixed it by configuring the checkout format in github to: ```git config --global core.autocrlf true``` and then it was able to compile.

closed 2019-07-19 by vasav123

muhammadadeelinfo, 2020-01-31:
> Windows, with virtual box, and vagrant.

Actually, the current vagrant image is based on ubuntu 16.04 which is now going to be updated to ubuntu 18.04 in next RIOT release. (Not supported with windows os)
if you try the current vagrant image with with ubuntu 16.04 and gcc version 7.4.0 or above then it will compile. I have just tried now.
#13015