RIOT-OS/RIOT #11884: NIB only sending NS when using global IP

shr70, 2019-07-22:
When trying to ping from one RIOT node to another one while using gloabl IPs, the nodes get stuck on NS messages. NA will never be sent.
Both nodes run the default `examples/gnrc_networking` application. One node will get IP `fdaa:bb:cc:dd::3` by calling `ifconfig 7 add fdaa:bb:cc:dd::3` the other one will respectively receive `fdaa:bb:cc:dd::4`.
When trying to ping one node from the other both will start to transmit NS messages. When pinging from a Pi to RIOT and vice versa everything works as expected (NS->NA->Ping).

Here are the NIB Logs:

Node1 `fdaa:bb:cc:dd::3`:
```
ping6 fdaa:bb:cc:dd::4␍␊
icmpv6: Building ICMPv6 message with type=128, code=0␊
nib: get next hop link-layer address of fdaa:bb:cc:dd::4%7␊
nib: Getting on-link node entry (addr = fdaa:bb:cc:dd::4, iface = 7)␊
  No suitable entry found␊
nib: fdaa:bb:cc:dd::4 is off-link, resolve route␊
nib: get route fdaa:bb:cc:dd::4 for packet 0x20001a84␊
nib: get match for destination fdaa:bb:cc:dd::4 from NIB␊
nib: fdaa:bb:cc:dd::/64 => (nil)%7 matches with 125 bits␊
nib: best match (125 bits)␊
nib: no route to fdaa:bb:cc:dd::4 found or is prefix list entry, search neighbor cache␊
nib: prefix list entry => taking dst as next hop␊
nib: Getting on-link node entry (addr = fdaa:bb:cc:dd::4, iface = 7)␊
  No suitable entry found␊
nib: resolve address fdaa:bb:cc:dd::4 by probing neighbors␊
nib: Allocating on-link node entry (addr = fdaa:bb:cc:dd::4, iface = 7)␊
  0x200016b0 is an exact match␊
nib: Adding to neighbor cache (addr = fdaa:bb:cc:dd::4, iface = 7)␊
nib: queueing (addr = fdaa:bb:cc:dd::4, iface = 7) for potential removal␊
nib: lookup ctx = 0x200016b0, type = 4fc1␊
icmpv6: Building ICMPv6 message with type=135, code=0␊
icmpv6: NDP message received. Handle with gnrc_ipv6_nib␊
nib: Handle packet (icmpv6->type = 135)␊
nib: Received valid neighbor solicitation:␊
     - Target address: fdaa:bb:cc:dd::3␊
     - Source address: fdaa:bb:cc:dd::4␊
     - Destination address: ff02::1:ff00:3␊
nib: Storing SL2AO for later handling␊
icmpv6: Building ICMPv6 message with type=136, code=0␊
icmpv6: no one interested in type 135␊
nib: get next hop link-layer address of fdaa:bb:cc:dd::4%7␊
nib: Getting on-link node entry (addr = fdaa:bb:cc:dd::4, iface = 7)␊
  Found 0x200016b0␊
nib: fdaa:bb:cc:dd::4 is on-link or in NC, start address resolution␊
nib: resolve address fdaa:bb:cc:dd::4 by probing neighbors␊
nib: lookup ctx = 0x200016b0, type = 4fc1␊
nib: host unreachable␊
```

Node 2 `fdaa:bb:cc:dd::4`:
```
nib: Handle packet (icmpv6->type = 135)␊
nib: Received valid neighbor solicitation:␊
     - Target address: fdaa:bb:cc:dd::4␊
     - Source address: fdaa:bb:cc:dd::3␊
     - Destination address: ff02::1:ff00:4␊
nib: Storing SL2AO for later handling␊
nib: get next hop link-layer address of fdaa:bb:cc:dd::3%7␊
nib: Getting on-link node entry (addr = fdaa:bb:cc:dd::3, iface = 7)␊
  Found 0x200016b0␊
nib: fdaa:bb:cc:dd::3 is on-link or in NC, start address resolution␊
nib: resolve address fdaa:bb:cc:dd::3 by probing neighbors␊
nib: Allocating on-link node entry (addr = fdaa:bb:cc:dd::3, iface = 7)␊
  0x200016b0 is an exact match␊
nib: Adding to neighbor cache (addr = fdaa:bb:cc:dd::3, iface = 7)␊
nib: lookup ctx = 0x200016b0, type = 4fc1␊
nib: host unreachable␊
```

This will repeat over and over. Pinging link-local IPs works fine. Am I misunderstanding how the Neighbor Discovery is supposed to work or is it an error in RIOT?

miri64, 2019-07-30:
What kind of hardware are you on. If you are using 6LoWPAN, the neighbor discovery will work in accordance with RFC-6775

shr70, 2019-07-31:
I'm using the nucleo-l073rz board together with the openlabs 802.15.4 transceiver (at86rf233)

miri64, 2019-07-31:
And what do you use as a border router?

shr70, 2019-07-31:
As for now I am not using a border router. Is this why global IPs won't work? I'm confused, because when I connect the transceiver to the Raspberry PI (also configured as Node, not as border router) global IPs are working.

miri64, 2019-07-31:
> As for now I am not using a border router. Is this why global IPs won't work?

Yes, the global prefix distribution in a 6LoWPAN typically starts from the border router (see [RFC 6775, section 8.1](https://tools.ietf.org/html/rfc6775#section-8.1). As Linux does not implement 6Lo-ND (aka RFC 6775), it just uses [classic ND](https://tools.ietf.org/html/rfc4861) and [SLAAC](https://tools.ietf.org/html/rfc4862) and so it just starts sending Router Advertisements containing the prefix information (when configured as such) without waiting for a border router.

miri64, 2019-07-31:
As for why NS only are sent for global IPs: For link-local addresses [it is assumed](https://tools.ietf.org/html/rfc6775#section-5.6) that they are generated from the link layer address and thus the generation process is just reversed, so no address resolution and thus no NS/NA exchange is needed.

shr70, 2019-07-31:
Ok, thank you for clarifying

closed 2019-07-31 by shr70