RIOT-OS/RIOT #11885: arm7: `printf()` with `float`/`double` not working

maribu, 2019-07-22:
#### Description

Similar to https://github.com/RIOT-OS/RIOT/issues/11861 `printf()` is not working as well, but due to a different yet to determine reason. This was figured out in https://github.com/RIOT-OS/RIOT/pull/11883

#### Steps to reproduce the issue

```
BOARD=msba2 make --no-print-directory -C tests/unittests/ tests-printf_float  flash-only test

Type '/exit' to exit.
2019-07-22 17:10:02,014 - INFO # main(): This is RIOT! (Version: 2019.10-devel-106-g2d5ed-pr/newib/remove_use_nano_specs_variable)
2019-07-22 17:10:02,015 - INFO # .
2019-07-22 17:10:02,019 - INFO # pkt_tests.sfprintf_float (tests/unittests/tests-printf_float/tests-printf_float.c 40) exp "2016.034900" was "-0.000000"
2019-07-22 17:10:02,019 - INFO # 
2019-07-22 17:10:02,020 - INFO # run 1 failures 1
```

#### Expected results
Unit test passes

#### Actual results
Unit test fails

#### Versions
Current RIOT master

cladmi, 2019-07-23:
Note, it currently is the only `unittests`  remaining failing test:

```
2019-07-23 13:32:21,764 - INFO # main(): This is RIOT! (Version: 2019.10-devel-109-gb1bab)
2019-07-23 13:32:21,767 - INFO # .........................................................................................................................................................................................................
2019-07-23 13:32:21,768 - INFO # pkt_tests.sfprintf_float (tests/unittests/tests-printf_float/tests-printf_float.c 40) exp "2016.034900" was "-0.000000"
2019-07-23 13:33:25,269 - INFO # ........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
2019-07-23 13:33:25,270 - INFO # run 945 failures 1
```

And when changing:

```diff
diff --git a/core/kernel_init.c b/core/kernel_init.c
index 1b153b38a..bb61c5dd9 100644
--- a/core/kernel_init.c
+++ b/core/kernel_init.c
@@ -74,7 +74,7 @@ static void *idle_thread(void *arg)
 const char *main_name = "main";
 const char *idle_name = "idle";

-static char main_stack[THREAD_STACKSIZE_MAIN];
+static char main_stack[THREAD_STACKSIZE_MAIN] = {0xff};
 static char idle_stack[THREAD_STACKSIZE_IDLE];

 void kernel_init(void)
```
it works.

Some other `printf` cases work:

```
ssh -t vaduz.imp.fu-berlin.de 'source /srv/ilab-builds/workspace/workspace.rc && BOARD=msba2 QUIET=0 make --no-print-directory -C /srv/ilab-builds/boards term'
/srv/ilab-builds/boards/RIOT/dist/tools/pyterm/pyterm -tg -p "/dev/riot/ttyMSBA2"
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-07-23 13:44:07,605 - INFO # Connect to serial port /dev/riot/ttyMSBA2
Welcome to pyterm!
Type '/exit' to exit.
2019-07-23 13:44:08,609 - INFO # main(): This is RIOT! (Version: 2019.10-devel-109-gb1bab)
2019-07-23 13:44:08,609 - INFO # Hello World!
2019-07-23 13:44:08,610 - INFO # You are running RIOT on a(n) msba2 board.
2019-07-23 13:44:08,610 - INFO # This board features a(n) lpc2387 MCU.
2019-07-23 13:44:08,611 - INFO # What is the approximate value of Pi: 3.141593
2019-07-23 13:44:08,611 - INFO # What is the float answer: 42.000000
```

cladmi, 2019-07-23:
It may be `unittests` related too. Or float context handling or something.
This does not print correctly:

```diff
diff --git a/tests/unittests/tests-core/tests-core.c b/tests/unittests/tests-core/tests-core.c
index 3551286e5..34eed4d5f 100644
--- a/tests/unittests/tests-core/tests-core.c
+++ b/tests/unittests/tests-core/tests-core.c
@@ -6,10 +6,12 @@
  * directory for more details.
  */

+#include <stdio.h>
 #include "tests-core.h"

 void tests_core(void)
 {
+    printf("Tests-core float %f\n", 3.141592654);
     TESTS_RUN(tests_core_atomic_tests());
     TESTS_RUN(tests_core_bitarithm_tests());
     TESTS_RUN(tests_core_cib_tests());
```

```
USEMODULE=printf_float BOARD=msba2 make --no-print-directory -C tests/unittests/ tests-core  flash term

2019-07-23 13:51:15,831 - INFO # main(): This is RIOT! (Version: 2019.10-devel-109-gb1bab)
2019-07-23 13:51:15,832 - INFO # Tests-core float 156104357234891300905575417519308883293391578287969291970140075052881684524533620118692204698927104.000000
2019-07-23 13:51:15,982 - INFO # .............................................................................................
2019-07-23 13:51:15,983 - INFO # OK (93 tests)
```

maribu, 2019-07-23:
When  running:

```diff
diff --git a/examples/hello-world/Makefile b/examples/hello-world/Makefile
index 258d8e9baf..447d78fbbb 100644
--- a/examples/hello-world/Makefile
+++ b/examples/hello-world/Makefile
@@ -15,4 +15,6 @@ DEVELHELP ?= 1
 # Change this to 0 show compiler invocation lines by default:
 QUIET ?= 1
 
+USEMODULE += printf_float
+
 include $(RIOTBASE)/Makefile.include
diff --git a/examples/hello-world/main.c b/examples/hello-world/main.c
index f51bf8c0a0..deccd5f7cf 100644
--- a/examples/hello-world/main.c
+++ b/examples/hello-world/main.c
@@ -21,6 +21,8 @@
 
 #include <stdio.h>
 
+extern double pi;
+
 int main(void)
 {
     puts("Hello World!");
@@ -28,5 +30,7 @@ int main(void)
     printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
     printf("This board features a(n) %s MCU.\n", RIOT_MCU);
 
+    printf("Pi is approx. %f\n", pi);
+
     return 0;
 }
diff --git a/examples/hello-world/pi.c b/examples/hello-world/pi.c
new file mode 100644
index 0000000000..9e599303be
--- /dev/null
+++ b/examples/hello-world/pi.c
@@ -0,0 +1 @@
+double pi = 3.141592653589793238;
```

I get

```
$ make BOARD=msba2 term
/home/maribu/Repos/software/RIOT/dist/tools/pyterm/pyterm -tg -p "/dev/ttyUSB0"
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-07-23 13:58:06,380 - INFO # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type '/exit' to exit.
2019-07-23 13:58:07,388 - INFO # main(): This is RIOT! (Version: 2019.10-devel-109-gb1bab)
2019-07-23 13:58:07,389 - INFO # Hello World!
2019-07-23 13:58:07,391 - INFO # You are running RIOT on a(n) msba2 board.
2019-07-23 13:58:07,392 - INFO # This board features a(n) lpc2387 MCU.
2019-07-23 13:58:07,392 - INFO # Pi is approx. 3.141593
2019-07-23 13:58:08,788 - INFO # Exiting Pyterm
``

closed 2022-10-13 by maribu