RIOT-OS/RIOT #11891: ESP32 Build image is corrupted

mmaxus35, 2019-07-23:
I switch to ESP32 in order to use tcp,udp with lwip. For esp8266 lwip gives memory error.

However, i face with more problems in ESP32. @gschorcht , i use esptool 2.7, i have build my esp32 code using your docker image. I also tested with esptool 2.7 converts elf to bin successfully or not. I tried different images. I successfully upload micropython binary files to esp32. Also i tested .elf to .bin conversion in esptool 2.6 and 2.7. Then last step is make problems. I use make BOARD=esp32-wroom-32 to make a hello world. Even it gave me this output

> 

> `rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3f400020,len:6424
ets Jun  8 2016 00:22:57`

Does esp32 has a known problem for this ?

gschorcht, 2019-07-23:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:QIO, clock div:1 load:0x3f400020,len:6424 ets Jun 8 2016 00:22:57
```
is just the reset message which always comes before boot loader loads the app from flash.

What command did you use to flash?

mmaxus35, 2019-07-23:
This command to make.
USEMODULE=esp_wifi \
CFLAGS='-DESP_WIFI_SSID=\"FiberHGW_TPBD9E_2.4GHz\" -DESP_WIFI_PASS=\"dguKuYPc\"' \
make -C examples/gnrc_networking BOARD=esp32-wroom-32

This command to elf to bin 

sudo ~/.local/bin/esptool.py --chip esp32 elf2image gnrc_networking.elf

This command to flash

sudo ~/.local/bin/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset  write_flash -z --flash_mode qio --flash_freq 80m --flash_size detect 0x1000 gnrc_networking.bin

gschorcht, 2019-07-23:
As described in the [documentation](https://riot-os.org/api/group__cpu__esp32.html#esp32_riot_docker_toolchain) in section <i>Make Process with Docker Image</i>, the usual way to flash an image that was made with docker is to use `make flash-only` on the host system, for example:
```
make flash-only BOARD=esp32-esp-12x -C examples/hello-world
```
Could you try this?

mmaxus35, 2019-07-23:
The following output is like this. 

`riotbuild@fd5661eed946:~/RIOT-master$ make flash-only BOARD=esp32-wroom-32 -C examples/hello-world`

```
make: Entering directory '/data/riotbuild/RIOT-master/examples/hello-world'
ESP32_SDK_DIR should be defined as /path/to/esp-idf directory
ESP32_SDK_DIR is set by default to /opt/esp/esp-idf
/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image -fm dout -fs 2MB -ff 40m    -o /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/hello-world.elf.bin /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/hello-world.elf; echo "" > /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.csv; echo "nvs, data, nvs, 0x9000, 0x6000" >> /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.csv; echo "phy_init, data, phy, 0xf000, 0x1000" >> /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.csv; echo -n "factory, app, factory, 0x10000, " >> /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.csv; ls -l /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/hello-world.elf.bin | awk '{ print $5 }' >> /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.csv; python /data/riotbuild/RIOT-master/cpu/esp32/gen_esp32part.py --disable-sha256sum --verify /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.csv /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.bin
```
```
esptool.py v2.5.0
Parsing CSV input...
/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash -z -fm dout -fs detect -ff 40m    0x1000 /data/riotbuild/RIOT-master/cpu/esp32/bin/bootloader.bin 0x8000 /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/partitions.bin 0x10000 /data/riotbuild/RIOT-master/examples/hello-world/bin/esp32-wroom-32/hello-world.elf.bin
esptool.py v2.5.0
Serial port /dev/ttyUSB0
Traceback (most recent call last):
  File "/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2849, in <module>
    _main()
  File "/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2842, in _main
    main()
  File "/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2549, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/opt/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 213, in __init__
    self._port = serial.serial_for_url(port)
  File "/data/riotbuild/.local/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/data/riotbuild/.local/lib/python2.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
/data/riotbuild/RIOT-master/examples/hello-world/../../Makefile.include:557: recipe for target 'flash-only' failed
make: *** [flash-only] Error 1
make: Leaving directory '/data/riotbuild/RIOT-master/examples/hello-world'
```

By the way i noticed that esptool version is 2.5.0 for docker image. Could it be possible to after that version 2.6 and 2.7 has elf->bin problems?

gschorcht, 2019-07-23:
The output of `make flash-only` is completly ok with the exception that your device is obviously not connected to `/dev/ttyUSB0`:
```
Serial port /dev/ttyUSB0
Traceback (most recent call last):
...
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
```
Please define the port at which your device is connected, for example:
```
make flash-only BOARD=esp32-wroom-32 -C examples/hello-world PORT=/dev/ttyUSB1
```

> Update: I also notice that there are bootloader areas as parameters for this make flash command

Of course, and it generates a partition table which is required to load the right application.

mmaxus35, 2019-07-23:
I have make it work. I just give admin rights to all ttyUSB. I am able to
run examples now. Thank you. Let me ask you a short question then close
this issue. Using lwip can i run tcp and udp examples in offical page.
Simply does esp32 port has implementations for tcp,udp,ip based on lwip?

şunu yazdı:

gschorcht, 2019-07-23:
> Simply does esp32 port has implementations for tcp,udp,ip based on lwip?

It's not a question of whether the ESP32 supports lwip, but whether RIOT does. RIOT at least comes with package called lwip which provides an interface to lwip, please refer the tests in `test/lwip_*`. But, I have no knowledge about this lwip package. The only thing I can say is that ESP32 boards are not blacklisted in the makefiles of these tests. Thus, they are at least compilable. Whether they really work, ..., I don't know.

mmaxus35, 2019-07-23:
Thank for your answer. 
Best regards.

closed 2019-07-23 by mmaxus35