RIOT-OS/RIOT #11894: start a udp server that listens only to one client and port

talalong, 2019-07-23:
I am trying to create `udp` server with `gnrc_networking` base on [this](https://riot-os.org/api/group__net__sock__udp.html?#ga89562920ad89fe0e098fc989e3b064ec) for my samr21-xpro.

This code below works properly, but I want the UDP server to listen to only one client (ipv6 address) and its port. Therefore I have two questions:

1. Is it possible to force the server listening to only one ipv6 address and its port?
2. How can I get the client's address and port as string using `remote` variable?

```
sock_udp_ep_t local = SOCK_IPV6_EP_ANY;
    sock_udp_t sock;
    local.port = 8080;
    puts("starting udp serser at 8080");
    if (sock_udp_create(&sock, &local, NULL, 0) < 0)
    {
        puts("Error creating UDP sock");
        return NULL;
    }
    while (1)
    {
        sock_udp_ep_t remote;
        ssize_t res;
        if ((res = sock_udp_recv(&sock, buf_server, sizeof(buf_server), SOCK_NO_TIMEOUT,
                                 &remote)) >= 0)
        {
            puts("Received a message from ???); // remote ???
             if ((res > 0) && (res < 99))
            {
                buf_server[res] = '\0';   /* add the null byte so buffer now holds a string */
                puts((char *)buf_server); /* write this string to the display */
            }
            if (sock_udp_send(&sock, buf_server, res, &remote) < 0)
            {
                puts("Error sending reply");
                return NULL;
            }
        }
    }
```

benemorius, 2019-07-27:
> 1) Is it possible to force the server listening to only one ipv6 address and its port?

Yes. Populate a `sock_udp_ep_t` for the remote endpoint and pass it to `sock_udp_create()` in place of `NULL`. IIRC both the port and address must be specified and the address may not be `::`. When a datagram is received that doesn't match `remote`, `sock_udp_recv()` will return `-EPROTO`.
```c
    sock_udp_ep_t remote = SOCK_IPV6_EP_ANY;
    remote.port = 9000;
    char remote_ip[IPV6_ADDR_MAX_STR_LEN] = "2001:db8:ffff:ffff::2";
    ipv6_addr_from_str((ipv6_addr_t *)&remote.addr.ipv6, remote_ip);

    int res;
    sock_udp_t sock;
    if ((res = sock_udp_create(&sock, &local, &remote, 0)) < 0) {
        LOG_ERROR("[udp server] error initializing socket (%i)\n", res);
        return res;
    }
```

> 2) How can I get the client's address and port as string using `remote` variable?

You could use `ipv6_addr_to_str()` like this:
```c
    while (1) {
        sock_udp_ep_t remote;
        ssize_t res;
        if ((res = sock_udp_recv(&sock, server_buffer, sizeof(server_buffer),
            SOCK_NO_TIMEOUT, &remote)) < 0)
        {
            LOG_WARNING("sock_udp_recv() returned %li\n", res);
            continue;
        }

        char remote_ip[IPV6_ADDR_MAX_STR_LEN];
        ipv6_addr_to_str(remote_ip, (const ipv6_addr_t *)remote.addr.ipv6,
                         sizeof(remote_ip));
        LOG_DEBUG("received datagram from [%s]:%" PRIu16 "\n",
                  remote_ip, remote.port);
    }
```

talalong, 2019-07-28:
Hi @benemorius Thank you for you answer.
The function `ipv6_addr_to_str `for my second question works well.
But I did make a mistake while asking the first question. Actually I wanted to filter out only the client's address, and ignore the port number, because I have no control over the port number from client side, so it could be blocked from other process. Since you said: _IRC **both** the port and address must be specified_ . Is there another way to ignore the port number? because commenting the line `remote.port = 9000;` will create error.

Here is my server code:

    
    sock_udp_ep_t local = SOCK_IPV6_EP_ANY;
    sock_udp_t sock;
    local.port = 8080;
    sock_udp_ep_t remote = SOCK_IPV6_EP_ANY;
    //remote.port = 9000; 
    char remote_ip[IPV6_ADDR_MAX_STR_LEN] = "fd00:dead:beef::1";
    ipv6_addr_from_str((ipv6_addr_t *)&remote.addr.ipv6, remote_ip);
    puts("starting udp serser at 8080");
    int result;
    if ((result = sock_udp_create(&sock, &local, &remote, 0)) < 0)
    {
        printf("[udp server] error initializing socket (%i)\n", result);
        return result;
    }
    while (1)
    {
        ssize_t res;
        if ((res = sock_udp_recv(&sock, buf_server, sizeof(buf_server), SOCK_NO_TIMEOUT,
                                 &remote)) >= 0)
        {
            char remote_ip[IPV6_ADDR_MAX_STR_LEN];
            ipv6_addr_to_str(remote_ip, (const ipv6_addr_t *)remote.addr.ipv6,
                             sizeof(remote_ip));
            printf("received datagram from [%s]:%" PRIu16 "\n",
                   remote_ip, remote.port);
         }
     }

benemorius, 2019-07-28:
I don't think that's possible with gnrc_sock_udp.

In `sock_udp_create()`, `port` must be specified if `remote` is not null:
https://github.com/RIOT-OS/RIOT/blob/a9ab9d37e2fe4650e4ccac8d4d650311b3180a0f/sys/net/gnrc/sock/udp/gnrc_sock_udp.c#L91

And in `sock_udp_recv()` if a remote was set then the port must match to avoid returning `-EPROTO` and losing the packet data:
https://github.com/RIOT-OS/RIOT/blob/a9ab9d37e2fe4650e4ccac8d4d650311b3180a0f/sys/net/gnrc/sock/udp/gnrc_sock_udp.c#L208

So instead I would proceed to filter the packets in your application code based on your question 2). You can use `memcmp()` directly so you don't have to convert and compare strings:
https://github.com/RIOT-OS/RIOT/blob/a9ab9d37e2fe4650e4ccac8d4d650311b3180a0f/sys/net/gnrc/sock/udp/gnrc_sock_udp.c#L213

miri64, 2019-07-28:
> I don't think that's possible with gnrc_sock_udp.

AFAIK this goes not only beyond the functionallity of `sock_udp` (`gnrc_sock_udp` is just *one* implementation), but also beyond sockets in general. Usually, the task to restrict (or grant) access from certain hosts goes to a firewall (which we don't have in RIOT, so it needs to be deployed externally, e.g. on the border router).

talalong, 2019-07-29:
Thanks you guys for supporting.
@benemorius Yes the question 2 was the backup, if the first one does not work.
My questions are answered now.

closed 2019-07-29 by talalong