RIOT-OS/RIOT #11913: drivers/sx127x: Enforces reset pin to be defined

leandrolanzieri, 2019-07-25:
#### Description
After #11678 the reset sequence in the sx127x devices has been fixed, but having a reset pin is now mandatory. This is not the case of the `sensebox_samd21` board, which generates a reset pulse by adding a delay to the power line (see [schematics](https://github.com/watterott/senseBox-MCU/blob/master/hardware/senseBox-MCU_v12.pdf)). This currently makes the driver initialization fail on that board. How should we handle this?

I propose to remove the check of the reset pin being undefined from the driver, and only apply the reset sequence if the pin if defined. The side effect of that is that `sx127x_reset` will not actually reset the module in the `sensebox_samd21`. Although, the radio will reset on a software reset of the MCU. If we agree on that, I can provide the fix on a PR.

#### Steps to reproduce the issue
Build and flash `tests/driver_sx127x` application on a `sensebox_samd21`, or set `SX127X_PARAM_RESET` to `GPIO_UNDEF`.

#### Expected results
Successful initialization, output with debug on:
```
2019-07-25 12:16:04,856 - INFO # main(): This is RIOT! (Version: 2019.10-devel-111-g7eb57)
2019-07-25 12:16:04,860 - INFO # [sx127x] netdev: initializing driver...
2019-07-25 12:16:04,863 - INFO # [sx127x] SPI_1 initialized with success
2019-07-25 12:16:04,895 - INFO # [sx127x] initializing radio settings
2019-07-25 12:16:04,900 - INFO # [sx127x] netdev: initialization done
2019-07-25 12:16:04,905 - INFO # Initialization successful - starting the shell now
```

#### Actual results
Driver fails to initialize:
```
2019-07-25 12:57:28,527 - INFO # main(): This is RIOT! (Version: 2019.10-devel-111-g7eb57)
2019-07-25 12:57:28,528 - INFO # [sx127x] netdev: initializing driver...
2019-07-25 12:57:28,530 - INFO # [sx127x] SPI_0 initialized with success
2019-07-25 12:57:28,531 - INFO # [sx127x] error: No reset pin defined.
2019-07-25 12:57:28,533 - INFO # [sx127x] netdev: initialization failed
2019-07-25 12:57:28,534 - INFO # Failed to initialize SX127x device, exiting
```

fjmolinas, 2019-07-25:
I don't have any argument against that. If it doesn't have a reset pin and we wan't to support the board then I don't see why not go back to the previous behavior. We can either remove the check or modify it for `sensebox` so it prints a debug message but doesn't return `-SX127X_ERR_GPIOS`.

MichelRottleuthner, 2019-07-25:
>(..) only apply the reset sequence if the pin if defined

I'd be fine with that.

IMO, as an alternative we could also add a separate power pin to the driver that could be used for low-level supply control. If no reset but a power control pin is available it could be used as a fallback to trigger the reset by power-cycling. Actually it wouldn't seem very unlikely that in a very low power setup you would like to use a FET to avoid the radio's sleep current of up to 1 µA. Especially considering that a lot of MCUs can sleep with currents in the nA range.

fjmolinas, 2019-07-25:
> IMO, as an alternative we could also add a separate power pin to the driver that could be used for low-level supply control. If no reset but a power control pin is available it could be used as a [...]

With @jia200x  we did exactly that for a LoRa project. And that concept could be applied to any driver, if a DRIVER_SW_OFF_PIN then turn off the radio. But in the case of the sensebox I couldn't fin any on the datasheet.

leandrolanzieri, 2019-07-25:
> IMO, as an alternative we could also add a separate power pin to the driver that could be used for low-level supply control.

I think it would be nice to have.

> But in the case of the sensebox I couldn't fin any on the datasheet.

@fjmolinas in the case of `sensebox` the pin that controls this is defined as `XBEE1_EN_PIN`, 'D22_XB1_PWR in the schematic. Do you think it should be optional as `SX127X_USE_TX_SWITCH`?

fjmolinas, 2019-07-25:
> @fjmolinas in the case of sensebox the pin that controls this is defined as XBEE1_EN_PIN, 'D22_XB1_PWR in the schematic. Do you think it should be optional as SX127X_USE_TX_SWITCH?

I think yes. Maybe we can define a `sx127x_on/off()` function and in the case of the sensebox when calling `sx127x_reset()`, instead of the normal reset sequence it would call `sx127x_on/off()` with the appropriate timing and delays (if not the same for cold boot than a hardware reset).

 `sx127x_on/off()` can be guarded depending if the SX127X_PWR_SWITCH is defined or not. An application aiming for the smallest power consumption could turn of the driver when going to sleep and save that 1uA as @MichelRottleuthner suggested.

jcarrano, 2019-07-26:
Aside from the sensobox issue, I don't think checking for a valid config everywhere is a good design:
https://github.com/RIOT-OS/RIOT/blob/cb57c6ff1a730a740d9298c312d429cbf201e8cc/drivers/sx127x/sx127x.c#L98-L101

That code should go (i.e. it should just return without error). Otherwise, what is it guarding against? An unitialized value? Then it is highly unlikely that the value is exactly GPIO_UNDEF. If the value is GPIO_UNDEF then the user set it to that value and there must be a reason.

Following that logic, the following `#ifdefs` can go too, by allowing rx/tx_switch_pin to be undefined:
https://github.com/RIOT-OS/RIOT/blob/cb57c6ff1a730a740d9298c312d429cbf201e8cc/drivers/sx127x/sx127x.c#L105-L114

leandrolanzieri, 2019-07-26:
> Following that logic, the following `#ifdefs` can go too, by allowing rx/tx_switch_pin to be undefined:

In this case it is different, as the `sx127x_params_t` structure does only has those members if the macros are defined.

PR #12390 (2019-10-10): drivers/sx127x: skip reset sequence if pin is undefined

### Contribution description

This PR is an attempt to fix #11913: the sx127x driver reset sequence fails if the reset pin is undefined but some board don't provide it.

As initially proposed in #11913, it just don't return an error if the reset pin is not undefined but just skip the reset sequence.

Note that I moved the block initializing TX/RX switch pins before the reset sequence. I'm not sure what consequence this could have.

### Testing procedure

- Reset sequence should still work for board providing the reset pin
- The driver sh [... cut]
closed 2019-10-10 by leandrolanzieri