RIOT-OS/RIOT #11931: NRF52 NFC and GPIO

dcellucci, 2019-07-29:
#### Description
For the NRF5X family of chips, the pins reserved for the NFC can only be used as GPIO by changing a register in the UICR (User Information Configuration Registers). This is a nontrivial process because the UICR is stored in non-volatile memory, that defaults to the NFC configuration if the entire chip is erased. Currently, attempting to use these reserved pins as GPIO fails silently. 

My proposed feature will 

1. check if the pin being initialized is one of these UICR-limited pins and send an error message if the UICR not set properly. 
2. allow the use of a preprocessor define that will write this value to UICR at startup, in a similar manner as the Nordic SDK

### Useful links
https://infocenter.nordicsemi.com/topic/ps_nrf52840/uicr.html?cp=3_0_0_3_4#register.NFCPINS

SemjonKerner, 2019-07-30:
I've read up here, https://infocenter.nordicsemi.com/pdf/nRF52840_OPS_v0.5.pdf chapter **41.3**. This is what you mean?
Reminds me of https://github.com/RIOT-OS/RIOT/tree/master/dist/tools/nrf52_resetpin_cfg

Both your proposals have drawbacks when use of NFC is intended. Maybe a tool like above mentioned could be utilized to set up pins as desired. Of course this means, users have to be aware of this tool beforehand - which in my opinion is unlikely.

SemjonKerner, 2019-07-30:
btw thx for the ping @miri64

closed 2020-03-02 by stale[bot]