RIOT-OS/RIOT #11934: Make RIOT_VERSION available in external applications when using BUILD_IN_DOCKER=1

MrKevinWeiss, 2019-07-30:
#### Description
While writing an external test that uses RIOT as a git submodule I noticed using `BUILD_IN_DOCKER=1` does not have the correct RIOT_VERSION.  Insead I get `Version: UNKNOWN (builddir: /data/riotbuild/riotbase)`

### Useful links
example of riot as a submodule [here](https://github.com/RIOT-OS/RobotFW-tests)

MrKevinWeiss, 2019-07-30:
@cladmi knows it best!

cladmi, 2019-09-13:
Is it reasonable to think that the host system has `git` installed and so is responsible to get the RIOT_VERSION ?
It would remove the need for doing the hacky mounts for the git repository inside and would solve everything.
(except for static tests, but it is another issue.)

cladmi, 2019-09-13:
One issue I know is that this is executed in the wrong directory for your case:

https://github.com/RIOT-OS/RIOT/blob/ab8cdace1206aa271c69b4e1ebcc92c4b6817d89/makefiles/docker.inc.mk#L262

But the handling may start being complicated between external RIOT/worktree/submodules if it is only for the `RIOT_VERSION` that could be calculated before.

MrKevinWeiss, 2019-09-24:
I think we can assume that people have git if they are using RIOT.

closed 2020-04-27 by stale[bot]