RIOT-OS/RIOT #11941: cpu/esp32: instabilities in multiheap memory management

gschorcht, 2019-07-31:
#### Description

There seems to be an instability in the multiheap implementation of ESP32. If the WiFi interface is used with module `esp_now` or `esp_wifi`, the multiheap memory management of the SDK is used extensively. After some time the ESP32 stops to work and a short error message can be observed in the terminal
```
> CORRUPT
```
before it blocks completely. After that ESP32 can be only reactivated by hardware reset.

This error message is caused by an assertion in the mutiheap implementation and should normally looks like:
```
CORRUPT HEAP: multi_heap.c:428 detected at 0x3FC00123
```

#### Steps to reproduce the issue

Just compile and flash `examples/gnrc_networking`, for example:
```
USEMODULE=esp_now make BOARD=esp32-wroom-32 -C examples/gnrc_networking flash
```

gschorcht, 2019-08-06:
@MichelRottleuthner Unfortunatly, there is still a problem with the make system. The last change in commit https://github.com/RIOT-OS/RIOT/pull/11967/commits/e4198542d13ea7df119824836a8afae0e64b903c solved the problem of multiple definitions for `cpp11_*` applications, but led to a problem if `esp_wifi` module is enabled. 

The problem is that module `esp_idf_heap` (which requires to override memory management functions with wrappers) is enabled in `cpu/esp32/Makefile.dep` depending on whether module `esp_wifi` is enabled or not. However, since `cpu/esp32/Makefile.dep` is read after `cpu/esp32/Makefile.include`, the conditional definition of the linker options
```diff
+ifneq (,$(filter esp_idf_heap,$(USEMODULE)))
+    LINKFLAGS += -Wl,-wrap,_malloc_r
+￼    LINKFLAGS += -Wl,-wrap,_calloc_r
￼+    LINKFLAGS += -Wl,-wrap,_realloc_r
￼+    LINKFLAGS += -Wl,-wrap,_free_r
+endif
```
in `cpu/esp32/Makefile.include` doesn't work for `esp_wifi`. This leads to crashes of applications that are using the WiFi interface :worried:

Since WiFi application don't work anymore, I will provide a quick fix.

MichelRottleuthner, 2019-08-06:
Hmm that's nasty. Feel free to ping me when you have a PR ready. (side note: I think your comment was actually meant to go in #11967?)

gschorcht, 2019-12-07:
The problem was already fixed by PR #11947 and PR #11971.

closed 2019-12-07 by gschorcht