RIOT-OS/RIOT #11948: examples: ccn-lite-relay example broken

haukepetersen, 2019-08-01:
#### Description
The standard CCN-lite example in RIOT does not behave as expected: when sending interests

#### Steps to reproduce the issue
Run the `examples/ccn-lite-relay` example application on `native`, its enough to do so on a single node.

Next create some contents and afterwards send interests asking for these contents, and also send interests for some random names, where no content exists.

#### Expected results
The interests send for the contents we created should return the actual content we put in the content store. The interest for the random names should time out at some point.

#### Actual results
Any interest we send simply returns the first content we added to the content store:
```
main(): This is RIOT! (Version: 2019.10-devel-185-g2be13)
Basic CCN-Lite example
> ccnl_cs
ccnl_cs
> ccnl_cs /hello hello
ccnl_cs /hello hello
> ccnl_cs /moin moin
ccnl_cs /moin moin
> ccnl_cs
ccnl_cs
CS[0]: /moin [-1]: moin
CS[1]: /hello [-1]: hello
> ccnl_int /moin
ccnl_int /moin
PKTDUMP: data received:
~~ SNIP  0 - size:   4 byte, type: GNRC_NETTYPE_CCN_CHUNK (2)
Content is: moin
~~ PKT    -  1 snips, total size:   4 byte
> ccnl_int /hello
ccnl_int /hello
PKTDUMP: data received:
~~ SNIP  0 - size:   4 byte, type: GNRC_NETTYPE_CCN_CHUNK (2)
Content is: moin
~~ PKT    -  1 snips, total size:   4 byte
> ccnl_int /blubbblabla
ccnl_int /blubbblabla
PKTDUMP: data received:
~~ SNIP  0 - size:   4 byte, type: GNRC_NETTYPE_CCN_CHUNK (2)
Content is: moin
~~ PKT    -  1 snips, total size:   4 byte
> ccnl_int /hello
ccnl_int /hello
PKTDUMP: data received:
~~ SNIP  0 - size:   4 byte, type: GNRC_NETTYPE_CCN_CHUNK (2)
Content is: moin
~~ PKT    -  1 snips, total size:   4 byte
```
similar:
```
main(): This is RIOT! (Version: 2019.10-devel-185-g2be13)
Basic CCN-Lite example
> ccnl_cs
ccnl_cs
> ccnl_cs /hello hello
ccnl_cs /hello hello
> ccnl_cs /moin moin
ccnl_cs /moin moin
> ccnl_cs
ccnl_cs
CS[0]: /moin [-1]: moin
CS[1]: /hello [-1]: hello
> ccnl_int /blubb
ccnl_int /blubb
PKTDUMP: data received:
~~ SNIP  0 - size:   4 byte, type: GNRC_NETTYPE_CCN_CHUNK (2)
Content is: moin
~~ PKT    -  1 snips, total size:   4 byte
```

#### Versions
Build under Linux Mint 18.3, `gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609`

PeterKietzmann, 2019-08-02:
We fixed this in CCN-lite a while ago but did not yet bump the version in RIOT. Can you try with [this branch](https://github.com/PeterKietzmann/RIOT/tree/pr_ccnl_update_and_bump) please

haukepetersen, 2019-08-02:
yes, using your branch fixes the problems state above. Would you please create a PR so we can get this fix into master?!

PR #11952 (2019-08-05): pkg/ccn-lite: bump version and adapt shell functions accordingly
### Contribution description

This PR bumps the CCN-lite pkg version and adapts to recent API changes in the `sc_ccnl` shell command handler. I've tested the new version briefly in small multi-hop setups under native and in the FIT IoT-LAB testbed on iotlab-m3 nodes. Thereby I did some requests to differently named content items that (i) exist (ii) don't exist (iii) are generated during retransmission procedure. Everything went as expected.

### Testing procedure
(follow description in #11948)

### Issues/PRs references
Fixes #11 [... cut]
closed 2019-08-05 by miri64