RIOT-OS/RIOT #11951: nordic_softdevice_ble: Update to allow modern kernels for IPSP

chrysn, 2019-08-02:
There is an interoperability cut in 6lo-over-bluetooth (Bluetooth's IPSP) that happened in Linux around 4.12 and in nRF SDK between the currently-used "nRF IoT SDK" and the "nRF SDK" (which supports it at least since the [14.1 release](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v14.1.0%2Fiot_sdk_user_guides_linux_commands.html))

The current [documentation in RIOT](https://github.com/RIOT-OS/RIOT/blob/master/pkg/nordic_softdevice_ble/README-BLE-6LoWPAN.md) says it requires the old behavior by indicating the maximum supported Linux version as 4.12.

Please consider updating the upstream version, allowing to set `BLE_6LOWPAN_LEGACY_MODE = 0`, and updating the documentation.

(Note this is all from reading current documentation; I did not test the behavior as I don't have suitable hardware yet, so the fix might be as easy as stating in the documentation that RFC7668 mode can be enabled).

miri64, 2019-08-02:
I'm not sure about the future of the nordic softdevice, however for IPSP there is also #11578 as an alternative.

haukepetersen, 2019-08-02:
AFAIK there is nobody working on the Nordic Softdevice integration in RIOT currently, and not sure if anyone plans to do so in the near future...

The (soon to be) default BLE stack in RIOT is NimBLE, where all our efforts go into. As @miri64 stated, #11578 is the first step towards IP-over-BLE support using NimBLE. However, I did not test this agains Linux (yet), and there are most likely some more adaptions in GNRC needed to make it fully compliant to RFC7668 and the upcoming draft-ietf-6lo-blemesh-05. I am actively working on this as we speak :-)

miri64, 2019-08-10:
See also https://github.com/RIOT-OS/RIOT/pull/9473.

miri64, 2020-02-11:
@chrysn is this still up-to-date given that we now have nimble support?

chrysn, 2020-02-11:
Yes, but solving it is easier. A notice about, depending on what's the case, "This module is deprecated, use NimBLE" or "This module is supported, but has severe shortcomings in interoperability, use NimBLE if you need to interoperate with modern Linux kernels" would do.

I can put the documentation in a PR, but would need some direction on where nortic_softdevice_ble is going.

miri64, 2020-02-11:
> I can put the documentation in a PR, but would need some direction on where nortic_softdevice_ble is going.

@kaspar030 @haukepetersen @pekkanikander and others might better know about this. How about opening that PR and then discuss it there.

kaspar030, 2020-02-11:
> where nortic_softdevice_ble is going.

IMO, let's try to not use it anymore. It was a closed source blob. I'd rather concentrate on Nimble.

chrysn, 2020-02-11:
Opened as #13334. Actually I'm closing this here right away, as it *is* a duplicate of #9473 (and #13334 is just documenting that those exist and are not expected to move forward).

PR #13334 (2020-02-11): pkg/nordic_softdevice_ble: Imminent deprecation warning
### Contribution description

This makes the nRF softdevice docs accessible via doxygen and explains the outdatedness situation.

The goal of this is to bring some form of closure to #11951, whose underlying issue is that the first thing you find when searching for RIOT and Linux IPSP connectivity is that "it works only with kernels up to 4.12" (ie. still didn't solve the ages-old bug there).

### Issues/PRs references

This won't technically close #11951, but put it in a state where I'd be comfortable closing it with "won't fix unless so [... cut]
closed 2020-02-11 by miri64