RIOT-OS/RIOT #11978: Loramac Example Kernel Panic

dcellucci, 2019-08-07:
#### Description
Cannot 

#### Steps to reproduce the issue
I used the LoRaMac example on the AcSiPNode board (it's an STM32L073RZ with an SX1276)
Note that I have to use an 8-frequency subband of the full US915 range, because my base station.

#### Expected results
Confirmation of transmitted packets

#### Actual results
My terminal output is as follows (after successful OTAA join):
```
[semtech-loramac] loramac cmd msg
[semtech-loramac] send frame 1234
[semtech-loramac] MCPS request: confirmed TX
[semtech-loramac] MCPS request: OK
[semtech-loramac] Transmission completed
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] RX timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] unexpected netdev event received: 1
[semtech-loramac] MCPS confirm event
[semtech-loramac] MCPS indication event
[semtech-loramac] MCPS confirm msg received
[semtech-loramac] MCPS confirm event OK
[semtech-loramac] MCPS confirm event: CONFIRMED
[semtech-loramac] forward TX status to sender thread
[semtech-loramac] MCPS indication msg received
[semtech-loramac] MCPS indication Unconfirmed
[semtech-loramac] MCPS indication: ACK received
Cannot send message '1234', ret code: 6
[semtech-loramac] received something
Received ACK from network
```
After a few failed transmissions, I then get
```
sys/random/random.c:80 => 0x8000f25
*** RIOT kernel panic:
FAILED ASSERTION.

*** halted.
```

Weird right?

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.13.6
                 Kernel: Darwin 17.7.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Apple LLVM version 9.1.0 (clang-902.0.39.2)
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: Apple LLVM version 9.1.0 (clang-902.0.39.2)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.15.1
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.15.2 (Apple Git-101.1)
                   make: GNU Make 3.81
                openocd: missing
                 python: Python 3.7.4
                python2: Python 2.7.15 :: Anaconda, Inc.
                python3: Python 3.7.4
             coccinelle: missing
```

Best, 
DC

aabadie, 2019-08-08:
The message 'Cannot send message '1234', ret code: 6' is strange, since you received an ACK from the server.

Can you try #11983 and see if it also fixes your issue ?

rdagher, 2019-08-09:
> The message 'Cannot send message '1234', ret code: 6' is strange, since you received an ACK from the server.

@aabadie, I am having a similar behaviour in the AU915 band when using the latest release of RIOT. BTW, I checked that the gateway received and confirmed the message.
```
Sending message 0: This is RIOT!
[semtech-loramac] loramac cmd msg
[semtech-loramac] send frame This is RIOT!
[semtech-loramac] MCPS request: confirmed TX
[semtech-loramac] MCPS request: OK
[semtech-loramac] MAC timer timeout
[semtech-loramac] Transmission completed
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] RX timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] RX timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] Transmission completed
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] RX timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] MAC timer timeout
[semtech-loramac] RX started
[semtech-loramac] MCPS confirm event
[semtech-loramac] MCPS indication event
[semtech-loramac] MCPS confirm msg received
[semtech-loramac] MCPS confirm event OK
[semtech-loramac] MCPS confirm event: CONFIRMED
[semtech-loramac] forward TX status to sender thread
[semtech-loramac] MCPS indication msg received
[semtech-loramac] MCPS indication Unconfirmed
[semtech-loramac] MCPS indication: ACK received
Cannot send message 'This is RIOT!', ret code: 6:  Transmission completed 
```

@dcellucci would you please test with the release 2019.04 and see if you have this ACK problem ?
The kernel panic can be temporarily (for testing purposes) corrected as in the PR #11983, but that is a different issue than the ACK problem.

rdagher, 2019-08-13:
> The message 'Cannot send message '1234', ret code: 6' is strange, since you received an ACK from the server.
> 
I think this seems to happen when the same thread is used for TX and RX.

dcellucci, 2019-08-16:
I still get the kernel panic after changing
```
return (int32_t) (random_uint32_range(0, max - min) + min);
```
to
```
return (min == max) ? min : (int32_t) (random_uint32_range(0, max - min) + min);
```
in `randr`

dcellucci, 2019-08-16:
Can confirm that 11983 fixes the fault.

I'll take a look at the TX/RX problem.

PR #12383 (2019-10-07): pkg/semtech-loramac: fix failed assertion in randr function

### Contribution description

This PR is an alternative to #11983 to fix #11978, as a first step, before the package is migrated to the latest release.

### Testing procedure

Connecting to a LoRaWAN network in EU868 or US915 regions should work. In master US915 region doesn't work, as described in #11978.

Note: for EU868, this cannot be tested on iotlab right now, where there seem to have an issue on the TTN gateway...

### Issues/PRs references

fixes #11978
closes #11983
closed 2019-10-07 by miri64