RIOT-OS/RIOT #11980: Sending a packet to the Multicast Address causes Segmentation fault with multiple interfaces

benpicco, 2019-08-08:
#### Description
On RIOT, if there is more than once network interface, `ping6 ff02::1` will cause a segmentation fault.

#### Steps to reproduce the issue
In `examples/gnrc_networking`:

```
$ ../../dist/tools/tapsetup/tapsetup
$ make GNRC_NETIF_NUMOF=2 CFLAGS=-DNETDEV_TAP_MAX=2
$ bin/native/gnrc_networking.elf tap0 tap1
> ping6 ff02::1

Program received signal SIGSEGV, Segmentation fault.
0x5656dc02 in _fill_ipv6_hdr (netif=0x5659dcd8 <_netifs+312>, ipv6=0x5659de68 <_pktbuf+40>)
    at /home/benpicco/dev/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:308
308	    hdr->len = byteorder_htons(gnrc_pkt_len(ipv6->next));
(gdb) p hdr
$1 = (ipv6_hdr_t *) 0x40
```

#### Expected results
The packet should be send over both interfaces.
Also, `ipv6->data` should point to a valid address.

#### Actual results
RIOT crashes. This happens both on native and on actual hardware.

#### Versions
RIOT master (ca2b00875a9e91458ba361e293cc4e97d281ecf8)

miri64, 2019-10-20:
I was also able to reproduce this with a `netdev_tap` interface and a `socket_zep` interface.

miri64, 2019-10-20:
This slipped somehow under my radar and I only noticed it by looking at the issue list for the current release milestone. This is an easy fix, I will pull request ASAP.

miri64, 2019-10-20:
Fixed in #12512.

miri64, 2019-10-21:
Still not fixed for the release.

miri64, 2019-10-21:
Fixed by #12521.

commit ce14ee1f (2019-10-21): gnrc_ipv6: fix SEGFAULT when multicasting with multiple interfaces

When writing to the IPv6 header the implementation currently doesn't
take the packet with the (potentially) duplicated header, but the
packet with the original one, which leads to the packet sent and then
released in `gnrc_netif_ethernet.c` first and then accessed again in
further iterations of the "writing to the IPv6 header" loop, which
causes access to an invalid pointer, causing a crash.

Fixes #11980
commit e95bb176 (2019-10-21): gnrc_ipv6: fix SEGFAULT when multicasting with multiple interfaces

When writing to the IPv6 header the implementation currently doesn't
take the packet with the (potentially) duplicated header, but the
packet with the original one, which leads to the packet sent and then
released in `gnrc_netif_ethernet.c` first and then accessed again in
further iterations of the "writing to the IPv6 header" loop, which
causes access to an invalid pointer, causing a crash.

Fixes #11980

(cherry picked from commit ce14ee1f7702fec43b682c850da66285ca164ff0)
closed 2019-10-24 by benpicco