RIOT-OS/RIOT #12003: bootloaders/riotboot: broken with BUILD_IN_DOCKER and wrong flashfile

cladmi, 2019-08-13:
#### Description

The application in `bootloaders/riotboot` has the following issues, from a first look I think I also found the culprits:

* Introduced by bb71e979f
  * the file use when flashing is not the bootloader firmware but an extended bootloader with a slot0 bootloader. So `riotboot/flash-bootloader` would overwrite `slot0` and `slot1` header.
* Introduced by aaa187e87
  * it does not compile with `BUILD_IN_DOCKER` anymore
  * it uses the local toolchain

#### Steps to reproduce the issue

<details><summary>FLASHFILE is extended file</summary>

```
BOARD=iotlab-m3 make --no-print-directory -C bootloaders/riotboot/ info-debug-variable-FLASHFILE
/home/harter/work/git/RIOT/bootloaders/riotboot/bin/iotlab-m3/riotboot-slot0-extended.bin
```
</details>

<details><summary>It does not build with `docker`</summary>

```
BUILD_IN_DOCKER=1 DOCKER="sudo docker" BOARD=frdm-k64f make --no-print-directory -C bootloaders/riotboot
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/cpu/fcfield.o: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/cpu/vectors.o: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/newlib_syscalls_default/syscalls.o: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/application_riotboot.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/board.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/checksum.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/core.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/cortexm_common.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/cortexm_common_periph.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/cpu.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/newlib_syscalls_default.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/periph.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/riotboot.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/stdio_uart.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/sys.a: No such file or directory
arm-none-eabi-gcc: error: /home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/periph_common.a: No such file or directory
/home/harter/work/git/RIOT/makefiles/boot/riotboot.mk:35: recipe for target '/home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/riotboot-slot0.elf' failed
```
</details>

<details><summary>Building uses the local toolchain</summary>

```
PATH=${DEFAULT_PATH} BUILD_IN_DOCKER=1 DOCKER="sudo docker" BOARD=frdm-k64f make --no-print-directory -C bootloaders/riotboot/  
/bin/sh: 1: arm-none-eabi-gcc: not found
/home/harter/work/git/RIOT/makefiles/boot/riotboot.mk:35: recipe for target '/home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/riotboot-slot0.elf' failed
make: *** [/home/harter/work/git/RIOT/bootloaders/riotboot/bin/frdm-k64f/riotboot-slot0.elf] Error 127
```
</details>

#### Expected results

* `FLASHFILE` is `${BINDIR}/riotboot.elf`
* It compiles correctly in docker without local toolchain

#### Versions

Found on e5fe8684eae4f8e139800b94b4689f0e238eb816

Commit introducing the issues bb71e979f and aaa187e87

cladmi, 2019-08-13:
Currently running `buildtest` with `BUILD_IN_DOCKER=1` would work as https://github.com/RIOT-OS/RIOT/issues/11842 is hiding the issue.

fjmolinas, 2019-09-25:
I opened #12303 to address the second point. 

I can think of a couple solutions for the first issue:

- define a `riotboot_pseudomodule` that would be in `FEATURES_REQUIRED` only for riotboot application, proof of concept in https://github.com/fjmolinas/RIOT/tree/pr_riotboot_booloader

```
ifeq (,$(filter riotboot_bootloader,$(FEATURES_USED)))
# Applications that use 'riotboot' feature default to actually using it,
# except for the bootloader itself which will use 'riotboot.elf'
# when beeing flashed stand-alone.
# It also makes 'flash' and 'flash-only' work without specific command.
FLASHFILE = $(RIOTBOOT_EXTENDED_BIN)
endif
```

- use `APPLICATION = riotboot` to achieve the same behavior, this means less code addition but means checking a variable we don't check in RIOT so far. IMO this is actually a good solution but no application can be named riotboot other than the bootloader. Does this seem wrong regarding the bootloader? @cladmi 

```
ifeq (,$(filter riotboot,$(APPLICATION)))
# Applications that use 'riotboot' feature default to actually using it,
# except for the bootloader itself which will use 'riotboot.elf'
# when beeing flashed stand-alone.
# It also makes 'flash' and 'flash-only' work without specific command.
FLASHFILE = $(RIOTBOOT_EXTENDED_BIN)
endif
```

Maybe @kaspar030 has thoughts on this one as well..

cladmi, 2019-09-30:
This one is not closed yet #12302 took care of one part only for the moment.

fjmolinas, 2019-09-30:
Yes, I though prefixing with something different than fix would not close the issue, re-opened!

cladmi, 2019-09-30:
It was the top one in the pr comment that closed it, I only paid attention to the last one.

cladmi, 2019-09-30:
I updated the issue description to reflect the fixed compilation and failing test.

cladmi, 2019-10-17:
While reviewing #12446 I found out another issue with compiling the tool on the raspberry PI.

```
compiling /tmp/dwq.0.20579331210052065/5c38a93b2e9b62fea1be9c23e3b3971c/dist/tools/riotboot_gen_hdr/bin/genhdr...
genhdr.c: In function 'genhdr':
genhdr.c:71:49: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (errno != 0 || *p != '\0' || app_ver_arg > UINT32_MAX) {
                                                 ^
genhdr.c:79:52: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (errno != 0 || *p != '\0' || start_addr_arg > UINT32_MAX) {
                                                    ^
genhdr.c:87:76: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (errno != 0 || *p != '\0' || hdr_len_arg % HDR_ALIGN || hdr_len_arg > UINT32_MAX) {
                                                                            ^
```

When testing a change on this, the `test caching` must also be disabled as of course these files are ignored for it.

fjmolinas, 2019-10-17:
```
While reviewing #12446 I found out another issue with compiling the tool on the raspberry PI.

compiling /tmp/dwq.0.20579331210052065/5c38a93b2e9b62fea1be9c23e3b3971c/dist/tools/riotboot_gen_hdr/bin/genhdr...
genhdr.c: In function 'genhdr':
genhdr.c:71:49: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (errno != 0 || *p != '\0' || app_ver_arg > UINT32_MAX) {
                                                 ^
genhdr.c:79:52: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (errno != 0 || *p != '\0' || start_addr_arg > UINT32_MAX) {
                                                    ^
genhdr.c:87:76: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (errno != 0 || *p != '\0' || hdr_len_arg % HDR_ALIGN || hdr_len_arg > UINT32_MAX) {
                                                                            ^
When testing a change on this, the test caching must also be disabled as of course these files are ignored for it.
```

I'm guessing those are pre-compiled on murdock right? But you are right, this is also an issue.

cladmi, 2019-10-17:
It is an issue only on `32bit` hosts, so not on our machines or murdock that are 64bit.

PR #12302 (2019-09-30): Makefile.include: add BUILD_FILES variable that holds all files to be built
### Contribution Description

This PR adds a `BUILD_FILES` variable to hold all files that need to be linked. This is an attempt to fix #12003. 

By default `BUILD_FILES` will hold `$(ELFFILE)`, `$(FLASHFILE)` but more dependencies can be added as for example `SLOT_RIOT_ELFS`. This ensures the added targets are also built in docker when required.

There are other possible solutions, e.g:

```
ifneq (1, $(RIOTNOLINK))
  ifneq (1, $(BUILD_IN_DOCKER))
  link: $(SLOT_RIOT_ELFS)
  endif
endif
```

The first approach is bett [... cut]
commit aa5c6173 (2020-01-13): makefiles/boot/riotboot.mk: exclude if RIOTBOOT_BUILD is set

- riotboot targets should not be needed for riotboot application
  so dont include it.
- also fixes #12003 by not setting FLASHFILE = $(RIOTBOOT_EXTENDED_BIN)
  when compiling riotboot application
closed 2020-01-13 by kaspar030