RIOT-OS/RIOT #12027: Missing environment variables with BUILD_IN_DOCKER=1

gschorcht, 2019-08-17:
#### Description

The proposed approach to build a RIOT application using `riotdocker` is to execute `make` with `BUILD_IN_DOCKER=1` on host system, e.g.,
```
BUILD_IN_DOCKER=1 make -C examples/gnrc_networking flash term
```

However, very often the compilation has to be controlled by the variables `USEMODULE` and `CFLAGS`, for example:

```
USEMODULE=esp_wifi \
CFLAGS='-DESP_WIFI_SSID=\"ssid\" -DESP_WIFI_PASS=\"pass\"' \
make -C examples/gnrc_networking flash
```
Unfortunately, environment variables `USEMODULE` and `CFLAGS` are not passed docker. That is, using command
```
USEMODULE=esp_wifi \
CFLAGS='-DESP_WIFI_SSID=\"ssid\" -DESP_WIFI_PASS=\"pass\"' \
BUILD_IN_DOCKER=1 make -C examples/gnrc_networking flash
```
doesn't work.

At the moment, the only way is to run `riotdocker`, execute the `make` inside docker, and to execute `make flash-only` after that on host.

#### Expected results

Most important environment variables like `USEMODULE` and `CFLAGS` are passed to docker.

gschorcht, 2019-08-17:
@cladmi I guess you know the best.

cladmi, 2019-08-19:
Yes I know exactly :)

There are somehow nested reasons that cause this.

### Cannot access the value given from environment

The root of the issue is for me that there are *global* `export` for everything.
`USEMODULE`, `CFLAGS`, `LINKFLAGS`.
Which means that, if a value is modified during the makefile parsing, only the updated value would be accessible to sub-processes

If it was not the case, passing to docker could be done by using `CFLAGS=$${CFLAGS}` to be sure only the one that is in the environment is seen and not the one updated during parsing. (for the one from command line, it would need its specific handling too).

I am slowly trying to remove them along the other things I do https://github.com/RIOT-OS/RIOT/issues/10850

### How is it normally worked around

To know what was set on the command and should be given to docker, the current workaround is to, early in the parsing, look at the `origin` of the variables:

https://github.com/RIOT-OS/RIOT/blob/5a953a21375b8a288f58965a3645711169a28613/Makefile.include#L99-L102

Which is then parsed at https://github.com/RIOT-OS/RIOT/blob/b6832050984407a0017df53fd0af8f04af504792/makefiles/docker.inc.mk#L25-L84

### Why does not this work?

Because an application is not a normal module, that has a `Makefile.dep` and a `Makefile.include`.
Which then requires to modify `CFLAGS` and `USEMODULE` **before** parsing `RIOT/Makefile.include`.

This could be updated to parse the application `Makefile.dep`, `Makefile.include` and `Makefile.features` if they exist.

It would be more a workaround for the `docker` case but also makes sense in the global parsing.

I could PR this if you want.

#### What is the problem with this?

The issue with this, is that currently, some boards/cpu need to have some dependency declared before parsing the board/cpu Makefile.include as `Makefile.dep` is parsed only after. Like the `esp` for example 

https://github.com/RIOT-OS/RIOT/blob/b6832050984407a0017df53fd0af8f04af504792/cpu/esp32/Makefile.include#L29-L38

I am in process of fixing this with https://github.com/RIOT-OS/RIOT/pull/12014 which is part of https://github.com/RIOT-OS/RIOT/issues/9913
But will not be instant.

### Another workaround

Another workaround is to allow what you have in your cpu/esp32 Makefile.include:

https://github.com/RIOT-OS/RIOT/blob/b6832050984407a0017df53fd0af8f04af504792/cpu/esp32/Makefile.include#L24-L27

and allow adding variables to `DOCKER_ENV_VARS`.

So from the command line you could do `DOCKER_ENV_VARS+="USE_MODULES" USE_MODULES+="esp_wifi" make`

It is a simple update that makes globally sense and that I could PR.

cladmi, 2019-08-19:
I have another HACK, would be to allow manually adding directly to the docker arguments. Which is also something I want in general.

I prepare a PR for this and the `DOCKER_ENV_VARS`.

cladmi, 2019-08-19:
I opened the PR for allowing setting the variables from command line https://github.com/RIOT-OS/RIOT/pull/12030

The solution for your specific case is in the comment https://github.com/RIOT-OS/RIOT/pull/12030#issuecomment-522637580

Escaping characters correctly is tricky but works :D

fjmolinas, 2020-01-10:
@gschorcht I believe this one was fixed a while ago with  #12030, can we close?

gschorcht, 2020-01-10:
Yes. PR #12030 provides a way to set environment variables, even though it is not very easy and intuitive, it works.

gschorcht, 2020-01-10:
@fjmolinas Thanks for pointing that out.

closed 2020-01-10 by gschorcht

fjmolinas, 2020-01-10:
> Yes. PR #12030 provides a way to set environment variables, even though it is not very easy and intuitive, it works.Yes. PR #12030 provides a way to set environment variables, even though it is not very easy and intuitive, it works.

I agree this is not intuitive, but for now until all exports are removed on `USEMODULE`  and `CFLAGS` it will have to remain this way.