RIOT-OS/RIOT #12037: cpu/sam0_common: i2c baudrate calculation fails if CLOCK_CORECLOCK > 48 MHz

benpicco, 2019-08-20:
#### Description
The I2C baudrate is always derived from `CLOCK_CORECLOCK`, even if a different `.gclk_src` is selected. If `CLOCK_CORECLOCK` is too high, the resulting baudrate will is too big and a check in [`i2c_init()`](https://github.com/RIOT-OS/RIOT/blob/master/cpu/sam0_common/periph/i2c.c#L144) silently fails, leaving the I2C uninitialized.

Any resulting access to the i2c bus will fail.

#### Steps to reproduce the issue
in `examples/default`:
 - `USEMODULE += i2c_scan`
 - `make BOARD=same54-xpro flash`
 - run `i2c_scan 0`

#### Expected results
```
2019-08-20 14:01:01,264 - INFO #  > i2c_scan 0
2019-08-20 14:01:01,270 - INFO # Scanning I2C device 0...
2019-08-20 14:01:01,279 - INFO # addr not ack'ed = "-", addr ack'ed = "X", addr reserved = "R", error = "E"
2019-08-20 14:01:01,283 - INFO #      0 1 2 3 4 5 6 7 8 9 a b c d e f
2019-08-20 14:01:01,286 - INFO # 0x00 R R R R R R R R R R R R R R - -
2019-08-20 14:01:01,287 - INFO # 0x10 - - - - - - - - - - - - - - - -
2019-08-20 14:01:01,288 - INFO # 0x20 - - - - - - - - X - - - - - - -
2019-08-20 14:01:01,296 - INFO # 0x30 X - - - - - X - - - - - - - - -
2019-08-20 14:01:01,298 - INFO # 0x40 - - - - - - - - - - - - - - - -
2019-08-20 14:01:01,304 - INFO # 0x50 X - - - - - X - X - - - - - X -
2019-08-20 14:01:01,309 - INFO # 0x60 X - - - - - - - - - - - - - - -
2019-08-20 14:01:01,313 - INFO # 0x70 - - - - - - - - R R R R R R R R
```

#### Actual results
#### Expected results
```
2019-08-20 13:56:22,893 - INFO # > i2c_scan 0
2019-08-20 13:56:22,895 - INFO # Scanning I2C device 0...
2019-08-20 13:56:22,901 - INFO # addr not ack'ed = "-", addr ack'ed = "X", addr reserved = "R", error = "E"
2019-08-20 13:56:22,906 - INFO #      0 1 2 3 4 5 6 7 8 9 a b c d e f
```
the program hangs indefinitely in an `-EAGAIN` loop.

#### Versions
RIOT master

dylad, 2019-08-20:
Indeed, this is true for SAMD51. The baud calculation should be slightly rework.
We should also use the SERCOM GCLK input clock instead of CLOCK_CORECLOCK (which is a legacy from old SAMD21 inplementation).
I can have a deeper look after the summit if you want.

benpicco, 2019-08-20:
I think it helpful to look at how [Zephyr is handling it](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/i2c/i2c_sam0.c#L495). But it looks like we need to use a slower GCLK.

PR #12969 (2020-02-05): cpu/sam0: provide interface to query GCLK frequency
### Contribution description

The sam0 family offers multiple GCLKs that can be sourced from different timers.
However, all peripherals were effectively always clocked by the hard-coded `CLOCK_CORECLOCK`.

This causes several issues:

 - on high CPU frequencies (> 51 MHz) the available I2C divider is too small to produce the required I2C baud rate 
- on low CPU frequencies the available frequency is too low to produce high UART baudrates.

This introduces a `sam0_gclk_freq()` function to query the frequency of  the individual GCLKs, so other  [... cut]
commit 4fbca251 (2020-02-05): boards/same54-xpro: use 48 MHz GCLK6 for I2C

We can't run I2C off the 120 MHz main clock as the availiable dividers are too small.
Use the 48 MHz GCLK 6 instead which offers an appropriate frequency.

fixes #12037
closed 2020-02-05 by dylad