RIOT-OS/RIOT #12039: Why are we using -fno-delete-null-pointer-checks?

jcarrano, 2019-08-20:
#### Description

I was trying to investigate using [`-Wnull-dereference`](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html) which according to the docs "is only active when -fdelete-null-pointer-checks is active".

In RIOT the opposite is true: -fno-delete-null-pointer-checks is active (since #628):

https://github.com/RIOT-OS/RIOT/blob/f0272ed262f9aa9fcea5c37268f63647572484f8/makefiles/cflags.inc.mk#L9-L11

This is wrong and may be contributing to hide bugs. From the [C FAQ](http://www.c-faq.com/null/accessloc0.html):

>  Since whatever is at location 0 is obviously machine dependent, you're free to use whatever machine-dependent trick will work to get there.

In other words, a null pointer is not valid from standard code, and if you need to access 0 then go outside the standard (with inline ASM for example).

### Question

Can we remove that flag and add `-fdelete-null-pointer-checks -Wnull-dereference`?

miri64, 2019-08-28:
> Can we remove that flag and add `-fdelete-null-pointer-checks -Wnull-dereference`?

:+1:

miri64, 2019-08-28:
(probably needs some preliminary work though before activating).

closed 2020-03-31 by stale[bot]

sir-ragna, 2025-12-18:
Here is an explanation on why you likely want to keep this compiler flag: https://github.com/ossf/wg-best-practices-os-developers/blob/main/docs/Compiler-Hardening-Guides/Compiler-Options-Hardening-Guide-for-C-and-C%2B%2B.md

> Synopsis
If a code defect references a potentially-null pointer, compilers are allowed to remove the null pointer check, under the theory that since developers never make mistakes, the pointer check is unnecessary.
>
> Since developers do make mistakes, without this option, the result is that the source code may appear to request a check, one that is necessary for security, but the check will not be done by the compiled executable. This option is one of several that are recommended and used by various sources to address real-world errors [89](https://github.com/ossf/wg-best-practices-os-developers/blob/main/docs/Compiler-Hardening-Guides/Compiler-Options-Hardening-Guide-for-C-and-C%2B%2B.md#user-content-fn-Wang2012-f8ed62c3af26bef7417bfa6ed8d33f61).

I can also recommend Eskil Steenberg's explanation on undefined behavior as further reading on this topic  https://dependablec.org/#undefined%20behavior

maribu, 2025-12-31:
I agree with @sir-ragna that we want to keep the flag for saner behavior in presence of bugs in RIOT code. There is also another reason given in the code:

https://github.com/RIOT-OS/RIOT/blob/cf494ebf9949f0b37c3dc8be5c33fc7d6de1d182/makefiles/cflags.inc.mk#L9-L14

So even if we would say "YOLO, RIOT code never has bugs", we still would need to add the flag as address `0x0` is valid on many MCUs we support. (E.g. IO memory on AVR, flash is mapped read-only to `0x0` on many Cortex M MCUs.)

It would be nice if all MCU vendors would have acknowledged the special meaning of `NULL` and wouldn't mapped anything into at least the first 256 byte of address space. But alas, we have to work with the hardware we have.