RIOT-OS/RIOT #12045: floats and doubles being used all over the place.

jcarrano, 2019-08-20:
#### Description

Most platforms do not have HW support for single-precisions floating point and none (aside from native) have support for double precision.

`grep`ing through the source code I can find instances of float and double. These are a problem because:

- They increase code size if soft-float support has to be linked-in.
- Soft float is slow.
- Even if hard float is available, it should normally not be used in interrupts or kernel code, as this would involve saving/restoring fp registers.
  - With drivers it is hard to know what runs in which context.
- The user may not even suspect that, for example the `sx127x` driver is using doubles.

From what I can see, most of these usages are unjustified. In particular _all_ uses of double could be replaced by float.

#### Steps to reproduce the issue

A quick way to get an approximate search is to use these regexes:
```
$ git grep  "^\([^/]\|\(/[^*/]\)\)*\bdouble\b"  -- "*.[ch]"
$ git grep  "^\([^/]\|\(/[^*/]\)\)*\bfloat\b"  -- "*.[ch]"
```
They will catch some false positives and probably miss some false negatives.

#### Actual results

After some manual cleanup, here are the results:

<details>
 <summary>Double trouble</summary>

```
drivers/bmp180/bmp180.c:    return (int16_t)(44330.0 * (1.0 - pow((double)p / pressure_0, 0.1903)));;
drivers/bmp180/bmp180.c:    return (uint32_t)((double)p / pow(1.0 - (altitude / 44330.0), 5.255));;
drivers/sx127x/sx127x_getset.c:    channel = (uint32_t)((double) channel / (double)LORA_FREQUENCY_RESOLUTION_DEFAULT);
drivers/sx127x/sx127x_getset.c:            double bw = 0.0;
drivers/sx127x/sx127x_getset.c:            double rs = bw / (1 << dev->settings.lora.datarate);
drivers/sx127x/sx127x_getset.c:            double ts = 1 / rs;
drivers/sx127x/sx127x_getset.c:            double t_preamble = (dev->settings.lora.preamble_len + 4.25) * ts;
drivers/sx127x/sx127x_getset.c:            double tmp =
drivers/sx127x/sx127x_getset.c:                    / (double) (4 * dev->settings.lora.datarate
drivers/sx127x/sx127x_getset.c:            double n_payload = 8 + ((tmp > 0) ? tmp : 0);
drivers/sx127x/sx127x_getset.c:            double t_payload = n_payload * ts;
drivers/sx127x/sx127x_getset.c:            double t_on_air = t_preamble + t_payload;
drivers/sx127x/sx127x_internal.c:    initial_freq = (double) (((uint32_t) sx127x_reg_read(dev, SX127X_REG_FRFMSB) << 16)
drivers/sx127x/sx127x_internal.c:                             | ((uint32_t) sx127x_reg_read(dev, SX127X_REG_FRFLSB))) * (double)LORA_FREQUENCY_RESOLUTION_DEFAULT;
drivers/tmp006/tmp006.c:    *tamb = (double)rawt / 128.0;
drivers/tmp006/tmp006.c:    double tdie_k = *tamb + 273.15;
drivers/tmp006/tmp006.c:    double sens_v = (double)rawv * TMP006_CCONST_LSB_SIZE;
drivers/tmp006/tmp006.c:    double tdiff = tdie_k - TMP006_CCONST_TREF;
drivers/tmp006/tmp006.c:    double tdiff_pow2 = pow(tdiff, 2);
drivers/tmp006/tmp006.c:    double s = TMP006_CCONST_S0 * (1 + TMP006_CCONST_A1 * tdiff
drivers/tmp006/tmp006.c:    double v_os = TMP006_CCONST_B0 + TMP006_CCONST_B1 * tdiff
drivers/tmp006/tmp006.c:    double f_obj = (sens_v - v_os) + TMP006_CCONST_C2 * pow((sens_v - v_os), 2);
drivers/tmp006/tmp006.c:    double t = pow(pow(tdie_k, 4) + (f_obj / s), 0.25);
sys/crypto/modes/ocb.c:       L_$ = double(L_*)
sys/crypto/modes/ocb.c:       L_0 = double(L_$)
sys/crypto/modes/ocb.c:       L_i = double(L_{i-1}) for every integer i > 0
sys/include/cb_mux.h: * @param[in] head   double pointer to first list entry
sys/include/cb_mux.h: * @param[in] head   double pointer to first list entry
sys/include/net/gnrc/rpl/structs.h:    double link_metric;             /**< metric of the link */
sys/include/random.h:double random_real(void);
sys/include/random.h:double random_real_inclusive(void);
sys/include/random.h:double random_real_exclusive(void);
sys/include/random.h:double random_res53(void);
sys/include/ubjson.h:static inline ssize_t ubjson_get_double(ubjson_cookie_t *__restrict cookie, ssize_t content, double *dest)
sys/include/ubjson.h:        double f;
sys/include/ubjson.h:ssize_t ubjson_write_double(ubjson_cookie_t *__restrict cookie, double value);
sys/net/routing/nhdp/iib_table.c:static const double const_dat = (((double)DAT_CONSTANT) / DAT_MAXIMUM_LOSS);
sys/net/routing/nhdp/iib_table.c:    double sum_total, sum_rcvd, loss;
sys/net/routing/nhdp/iib_table.c:                loss = (((double)ls_elt->hello_interval) * ((double)ls_elt->lost_hellos))
[... 126 lines cut ...]
tests/driver_hih6130/main.c:        float hum = 0.f;
tests/driver_hih6130/main.c:        float temp = 0.f;
tests/driver_hih6130/main.c:        float integral = 0.f;
tests/driver_hih6130/main.c:        float fractional;
tests/driver_io1_xplained/main.c:    float temperature;
tests/driver_tmp006/main.c:    float tamb, tobj;
tests/pkg_cmsis-dsp/main.c:    float int_part;
tests/pkg_cmsis-dsp/main.c:    float frac_part = fabsf(modff(variance, &int_part));
tests/rng/test.c:    float entropy = 0.0;
tests/rng/test.c:            float count = (float) buffer[i] / (float) length;
tests/rng/test.h:#define log2f(x) (logf (x) / (float) 0.693147180559945309417)
tests/thread_float/main.c:    float f, init;
tests/thread_float/main.c:    float f, init;
tests/unittests/tests-scanf_float/tests-scanf_float.c:    float x = 0;\
tests/unittests/tests-scanf_float/tests-scanf_float.c:    float x = 0;
tests/unittests/tests-scanf_float/tests-scanf_float.c:    float x = 0;
```

</details>

### What now?

The doubles should go away. I believe that's pretty clear.

There is float usage in sensor drivers. This is in my opinion wrong. The driver should have a layer which is responsible for retrieving data from the device. That layer has no business doing floating point computations. In any case the float is not needed even in cases of sensors like the mpu9150 where values can be perfecly expressed and manipulated using integer math exclusively.

garrettluu, 2019-10-06:
Interested! I would like to work on this issue.

aabadie, 2019-10-08:
Thanks @garrettluu! Looking forward for your PRs. A good approach would be to open a single PR for each module fixed/updated.
And don't forget to have a look at [our contributor guide](https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md).

garrettluu, 2019-10-08:
New here, what is considered a module? For example, is the whole drivers folder a module, or is each folder within the drivers folder a module? @aabadie

aabadie, 2019-10-08:
> is each folder within the drivers folder a module?

Mostly yes. If you open a PR for each driver it will be easier to verify nothing is broken: only one driver to review/test per PR.

garrettluu, 2019-10-08:
Draft PR created for sx127x drivers, but I'm unsure of how to test or at least compile my changes without a proper board.

maribu, 2023-05-18:
IMO this is more a rant than an issue. I was so free to create a tracking issue: https://github.com/RIOT-OS/RIOT/issues/19614

I would like to close this in favor of that tracking issue. Please reopen, if anyone disagrees.

closed 2023-05-18 by maribu