RIOT-OS/RIOT #12052: security: policy for merging code in Master

emmanuelsearch, 2019-08-21:
This is a venue to discuss evolving our security-related policies while merging code into Master. 

#10143 is related (but this issue is overarching, about all code, not only about crypto code).

kaspar030, 2019-08-21:
Many of us will meet soon at the Summit. I propose we collect agenda points here and discuss there.

bergzand, 2019-08-21:
As a start:
 - What is our definition of security-sensitive code

jcarrano, 2019-08-21:
> What is our definition of security-sensitive code

Any code whose purpose is to perform a security sensitive task. In principle, any piece of code could have security implications, that does not make it security-sensitive. What I think we are referring to here is code whose primary function is related to security. Crypto clearly falls into this category. Anything related to access control too.

bergzand, 2019-08-21:
> Any code whose purpose is to perform a security sensitive task. In principle, any piece of code could have security implications, that does not make it security-sensitive. What I think we are [...]

I was hoping to have my bullet point added to the agenda points and at the summit agree together on a definition. I don't want to just dismiss your answer, maybe we can use it as a starting point for the discussion at the summit

emmanuelsearch, 2019-08-21:
other points we could aim to synch on:
- what complementary expertise do we expect/need from security experts? (related: who are such people in practice?)
- what level of security, or what security characteristics do we aim for?

jia200x, 2019-10-18:
how to proceed with this?

miri64, 2020-07-05:
Ping @kaspar030 @bergzand @emmanuelsearch @jia200x?