RIOT-OS/RIOT #12057: ESP32 + DHT + SAUL reading two endpoints causes freeze.

mmaxus35, 2019-08-21:
BOARD=ESP32
Related features: esp32, saul, dht sensor.
I have a definitions as follows:

Using SAUL  i try to gather temperature and humidity values. However, in dht there are two and points and i tried to gather both of them. Using saul_reg_find_type(SAUL_SENSE_HUM); i can switch to gathering humidity but this time i cannot acquire temperature. So, i tried to do following logic. However, then i try to read two values at " **HERE ** "  chip stops running and led never blinks. I couldn't understand why and if there is a way to acquire both temperature and humidity values using one  saul_reg_read( ) function. Because if you have more than one for one device for two endpoints it freeze the chip, i think.

static saul_reg_t* saul_dht_temp;
static saul_reg_t* saul_dht_humid;

saul_dht = saul_reg_find_type(SAUL_SENSE_HUM);
saul_dht2 = saul_reg_find_type(SAUL_SENSE_TEMP);

phydat_t phy;
phydat_t phy2;

while(1){
gpio_toggle(g2); --> stops after this

saul_reg_read(saul_dht_temp, &phy); ** HERE **
saul_reg_read(saul_dht_humid, &phy2);

printf("temperature: %d %d %d Humid: %d %d %d \n", phy.val[0],phy.unit,phy.scale,phy2.val[0],phy2.unit,phy2.scale);

}

mmaxus35, 2019-08-21:
Sorry, it is my bad. I connected to sensor to wrong pin. However, freezing problem is a big deal. Maybe a timeout can be beneficial for saul_reg_read.

mmaxus35, 2019-08-22:
I had to re-open this issue because if during the sensor reading if sensor is somehow lose wiring connection from any pin or it is forgetten to be wired this causes somehow freeze in chip at the " saul_reg_read  " function called.

gschorcht, 2019-08-24:
@mmaxus35 The question is, whether it is really a problem of ESP32/ESP8266 or rather a problem caused by the driver. The platform shouldn't be the problem if basic functionalities like GPIOs and Interrrupts are working stable what they seem to do. So it sounds for me rather a problem of the driver. Are you able to check it with RIOT on another platform, e.g., Arduino? As I remember, you were starting your work with some Arduino sketch.

mmaxus35, 2019-08-24:
@gschorcht I haven't run Riot on Arduino board e.g Uno. However, to test another platform I tried to edit my code as sketch according to Arduino hello world example. I successfully flash and built the code. However, it caused another issue, the values cannot prompt to Serial port. I also built and flash Arduino hello world example and it prompts nothing to Serial. As a result, Arduino is another issue but highly the problem is about DHT driver or SAUL.

Eleftheria-99, 2021-01-06:
@mmaxus35 Which pins did you use ?

maribu, 2022-09-23:
I recall the that DHT driver didn't had a timeout for unsuccessful reads at some point in time. Likely this in combination with a cabling / configuration issue cause this.

Anyway, the DHT driver does now reliably report back communication issues rather than freezing and without further input from the reporter, we cannot provide further assistance. Thus, I'm closing this now.

@mmaxus35: If you want to give this another spin, I would gladly help out you run into issues again.

closed 2022-09-23 by maribu