RIOT-OS/RIOT #12071: tests/pkg_semtech-loramac insufficient memory for arduino-mega2560

jcarrano, 2019-08-23:
#### Description

`tests/pkg_semtech-loramac` is failing to link for "arduino-mega2560" due to insufficient memory. The weird thing is that it was failing before but now the CI decided it wants to link that application, causing unrelated PRs to fail.

#### Expected results

The board is skipped for that test.

#### Actual results

The CI tries to link tests/pkg_semtech-loramac for arduino-mega2560 and fails.

See: https://ci.riot-os.org/RIOT-OS/RIOT/10635/1a2a239ebfd4b048621dec1d04d22d35526b5e99/output.html#error0

aabadie, 2019-08-23:
Which unrelated PR ? #10635 ? e.g. yours ?
The pkg_semtech-loramac application is using the shell and just fits on arduino-mega2560 (see the discussion in #11917 for example).
So this is not a CI issue as you suggest (other PRs just build fine) but certainly an issue with your PR. Indeed, it modifies the shell so it probably increase the code size. Have you checked that before opening this issue ?

aabadie, 2019-08-23:
I spoke too fast, I can reproduce the issue on master locally.

Git bisect points to this commit 41e440a0efa58f92706fa69ffef65fb47cb460a4 from pr #11880. If we want to avoid problems with this board and this application. I propose to disable the semtech_loramac_rx module, this will save a lot of RAM/ROM and the package will still be built for this platform.

It would be great to know what caused the CI to start linking this application for this board.

jcarrano, 2019-08-26:
> It would be great to know what caused the CI to start linking this application for this board.

That's precisely why I raised this issue. The failure itself is not a big deal, just disable the tests. What concerns me is the CI's behaviour.

aabadie, 2019-08-26:
> The failure itself is not a big deal, just disable the tests

In case you missed it, #12073 does the job nicely, without having to disable the test for this board.

PR #12073 (2019-08-26): tests/pkg_semtech-loramac: disable rx thread for arduino-mega2560

### Contribution description

Fixes #12071 by disabling the rx extra thread in the application for arduino-mega2560.

### Testing procedure

A green Murdock should be ok

### Issues/PRs references

fixes #12071 and will help merging #11917
closed 2019-08-26 by jcarrano