RIOT-OS/RIOT #12076: How to use w5100 with stm32f4

jhx2002, 2019-08-25:
Is there any difference of pin wiring and ip address setting between w5100 and enc28j60 ? 
I try to use w5100, but can not ping6 successfully.

maribu, 2019-08-28:
Does it initialize? For me it only worked on a cold boot. Maybe I have time to PR a fix for that today.

Some boards (e.g. some Arduinos) reboot when the shell is connected. Surprisingly, on Arduinos I didn't experience the issue. I assume that the reset pin on the ISP pin header of my W5100 shield is connected to the reset pin of the W5100, reproducing the cold boot experience on reboot.

On Nucleos and the nRF52840DK the reset pin seems to not emit a signal when the board reboots, preventing a hardware reset of the shield. (At least that is my theory.)

jhx2002, 2019-08-28:
Can you share  experience on how to use w5100 ?  Thank you @maribu

closed 2020-04-13 by stale[bot]