RIOT-OS/RIOT #12077: usbdev support for SAML21

ant9000, 2019-08-25:
I would like to use periph_usbdev on SAML21 based boards. Given that this is a SAM0 variant, i have naively activated support like this:

```
diff --git a/boards/saml21-xpro/Makefile.features b/boards/saml21-xpro/Makefile.features
index 53cbc6967..4e062523a 100644
--- a/boards/saml21-xpro/Makefile.features
+++ b/boards/saml21-xpro/Makefile.features
@@ -6,6 +6,7 @@ FEATURES_PROVIDED += periph_rtt
 FEATURES_PROVIDED += periph_spi
 FEATURES_PROVIDED += periph_timer
 FEATURES_PROVIDED += periph_uart
+FEATURES_PROVIDED += periph_usbdev
 
 # Put other features for this board (in alphabetical order)
 FEATURES_PROVIDED += riotboot
diff --git a/boards/saml21-xpro/include/periph_conf.h b/boards/saml21-xpro/include/periph_conf.h
index 95c4b8b2e..353857da0 100644
--- a/boards/saml21-xpro/include/periph_conf.h
+++ b/boards/saml21-xpro/include/periph_conf.h
@@ -172,6 +172,19 @@ static const adc_conf_chan_t adc_channels[] = {
 #define ADC_0_REF_DEFAULT                  ADC_REFCTRL_REFSEL_INTVCC2
 /** @} */
 
+/**
+ * @name USB peripheral configuration
+ * @{
+ */
+static const sam0_common_usb_config_t sam_usbdev_config[] = {
+    {
+        .dm     = GPIO_PIN(PA, 24),
+        .dp     = GPIO_PIN(PA, 25),
+        .d_mux  = GPIO_MUX_G,
+        .device = &USB->DEVICE,
+    }
+};
+/** @} */
 #ifdef __cplusplus
 }
 #endif
```

Unfortunately, this is not enough - I guess there is some SAML21 related quirk in the driver that needs to be ironed out. What I can see is that on the board, the _usbdev_set function is called for USBOPT_ADDRESS, but with the invalid address 0.

Is there anyone that made it work already, or that is willing to share some suggestion about how to debug the driver?

Thanks a lot,

Antonio

ant9000, 2019-08-29:
Thanks to @dylad suggestion on RIOT mailing list, I've managed to make it work. USB stack needs a 48MHz clock, while SAML21 is clocked at 16MHz (silly me...).

My attempt at solving this in the driver is here:

    https://github.com/ant9000/RIOT/tree/pr/saml21_usbdev_48mhz_clock

I can open a PR with this patch, just ping me if you feel I should refine it in any way.

Antonio

emmanuelsearch, 2019-08-29:
@bergzand any additional opinions on this?

dylad, 2019-08-29:
Great to hear that you solve your issue !
I think your solution is good but could be a bit more generic.
In general, I would like to rework the clock management of SAM0 families but I didn't find time yet to work on it.

ant9000, 2019-08-30:
If you think my solution is already good enough for inclusion, I will open a PR. Once integrated it could be worked on by people more skilled than me, and with access to more boards.

PR #12132 (2020-06-16): cpu/saml21: enable 48mhz clock for usbdev
### Contribution description

This patch for SAML21 boards enables the 48MHz clock used by the usbdev driver, and the adds the driver configuration for a couple of boards (saml21-xpro, samr30-xpro).

### Testing procedure

I've tested a SAML21 board with both example/usbus_minimal and tests/usbus_cdc_ecm, and it works correctly. Also tested on samr34-xpro using PR #11250.

### Issues/PRs references

Closes #12077. Code adapted from #10075 as suggested by @dylad.
closed 2020-06-16 by dylad