RIOT-OS/RIOT #12108: `make term` output is inconsistent between boards, `ethos` and `native`

cladmi, 2019-08-28:
#### Description

The behavior of the terminal is inconsistent with `native` and `ethos` compared to a normal board.

The output is missing the timestamping, typing an empty line does not repeat the last command as expected and there is no command history.

#### Steps to reproduce the issue

Execute default example with `native`, and `examples/gnrc_border_router` with a `samr21-xpro` and type a command, an empty line, and try to get back in history.

I typed 'help+enter', 'enter', 'up+enter'.

#### Expected results

It should have the same behavior as using `examples-default` with `samr21-xpro`.

<details><summary>BOARD=samr21-xpro RIOT_CI_BUILD=1 make --no-print-directory -C examples/default/ flash term</summary>

```
BOARD=samr21-xpro RIOT_CI_BUILD=1 make --no-print-directory -C examples/default/ flash term
Building application "default" for "samr21-xpro" with MCU "samd21".

   text    data     bss     dec     hex filename
  40864     508    6048   47420    b93c /home/harter/work/git/RIOT/examples/default/bin/samr21-xpro/default.elf
/home/harter/work/git/RIOT/dist/tools/edbg/edbg  -t atmel_cm0p -b -v -p -f /home/harter/work/git/RIOT/examples/default/bin/samr21-xpro/default.bin
Debugger: ATMEL EDBG CMSIS-DAP ATML2127031800004678 01.1A.00FB (S)
Clock frequency: 16.0 MHz
Target: SAM R21G18 (Rev C)
Programming..................................................................................................................................................................... done.
Verification..................................................................................................................................................................... done.
/home/harter/work/git/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200" 
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-08-28 18:07:19,903 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
help
2019-08-28 18:07:23,348 - INFO # help
2019-08-28 18:07:23,351 - INFO # Command              Description
2019-08-28 18:07:23,354 - INFO # ---------------------------------------
2019-08-28 18:07:23,357 - INFO # reboot               Reboot the node
2019-08-28 18:07:23,363 - INFO # ps                   Prints information about running threads.
2019-08-28 18:07:23,367 - INFO # random_init          initializes the PRNG
2019-08-28 18:07:23,372 - INFO # random_get           returns 32 bit of pseudo randomness
2019-08-28 18:07:23,377 - INFO # rtc                  control RTC peripheral interface
2019-08-28 18:07:23,381 - INFO # ifconfig             Configure network interfaces
2019-08-28 18:07:23,387 - INFO # txtsnd               Sends a custom string as is over the link layer
2019-08-28 18:07:23,393 - INFO # saul                 interact with sensors and actuators using SAUL
> 
2019-08-28 18:07:24,083 - INFO #  help
2019-08-28 18:07:24,086 - INFO # Command              Description
2019-08-28 18:07:24,090 - INFO # ---------------------------------------
2019-08-28 18:07:24,093 - INFO # reboot               Reboot the node
2019-08-28 18:07:24,099 - INFO # ps                   Prints information about running threads.
2019-08-28 18:07:24,102 - INFO # random_init          initializes the PRNG
2019-08-28 18:07:24,108 - INFO # random_get           returns 32 bit of pseudo randomness
2019-08-28 18:07:24,112 - INFO # rtc                  control RTC peripheral interface
2019-08-28 18:07:24,117 - INFO # ifconfig             Configure network interfaces
2019-08-28 18:07:24,123 - INFO # txtsnd               Sends a custom string as is over the link layer
2019-08-28 18:07:24,129 - INFO # saul                 interact with sensors and actuators using SAUL
> help
2019-08-28 18:07:29,087 - INFO #  help
2019-08-28 18:07:29,089 - INFO # Command              Description
2019-08-28 18:07:29,093 - INFO # ---------------------------------------
2019-08-28 18:07:29,096 - INFO # reboot               Reboot the node
2019-08-28 18:07:29,102 - INFO # ps                   Prints information about running threads.
2019-08-28 18:07:29,106 - INFO # random_init          initializes the PRNG
2019-08-28 18:07:29,111 - INFO # random_get           returns 32 bit of pseudo randomness
2019-08-28 18:07:29,116 - INFO # rtc                  control RTC peripheral interface
2019-08-28 18:07:29,120 - INFO # ifconfig             Configure network interfaces
2019-08-28 18:07:29,126 - INFO # txtsnd               Sends a custom string as is over the link layer
[... 87 lines cut ...]
All up, running the shell now
> help
help
Command              Description
---------------------------------------
reboot               Reboot the node
ps                   Prints information about running threads.
ping6                Ping via ICMPv6
random_init          initializes the PRNG
random_get           returns 32 bit of pseudo randomness
nib                  Configure neighbor information base
ifconfig             Configure network interfaces
fibroute             Manipulate the FIB (info: 'fibroute [add|del]')
6ctx                 6LoWPAN context configuration tool
> 

> ^[[A
shell: command not found: [A
> 
```
</details>

#### Versions

Current master, so 2019-07.

miri64, 2019-08-28:
There was some initial work towards that done in the past: https://github.com/RIOT-OS/RIOT/pull/2751

miri64, 2019-08-28:
BTW: `BOARD=iotlab-m3 IOTLAB_NODE=auto-ssh make -C examples/gnrc_networking` has the same issue.

cladmi, 2019-08-28:
> BTW: `BOARD=iotlab-m3 IOTLAB_NODE=auto-ssh make -C examples/gnrc_networking term` has the same issue.

It was implemented with `ssh` and `socat` on purpose to behave without the additional `pyterm` behavior. So it indeed does not work the same way.

cladmi, 2019-08-29:
For only the `readline` behavior, `rlwrap` could be used as tried in https://github.com/RIOT-OS/RIOT/pull/11099

maribu, 2022-09-23:
Isn't this mostly a duplicate of https://github.com/RIOT-OS/RIOT/pull/11099 ? IMO, we should close one of the two.

miri64, 2022-09-23:
The tests already use `cleanterm` (which seems to do the same what #11099 was aiming for). So maybe we can even close both. But in general: if there is a descriptive issue and a PR addressing it, I tend to say they are not duplicates of each other, but the one fixes the other (of course there are exceptions).

maribu, 2022-09-23:
Oh, the `[TRACKING]` in the PR title let me assume it is actually an issue...

> So maybe we can even close both.

Agreed!

closed 2022-09-23 by maribu