RIOT-OS/RIOT #12111: cpu/msp430: compiler is outdated, newer one available.

jcarrano, 2019-08-28:
#### Description

The GCC version used to compile for the MSP430 is very outdated:

```
msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Meanwhile, TI has released [GCC 8.2.0.52](http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/MSPGCC/latest/index_FDS.html#), along with binutils, gdb and newlib.

I stumbled upon this when trying to enable debug info compression.

Old toolchains make maintenance harder than it should be by requiring us to consider lots of corner cases and incompatible compiler options.

Also, AFAIK newlib is not being used for MSP430, requiring special handling, but there is one available with the TI toolchain.

#### Expected results

We should use an up to date compiler and library.

kaspar030, 2019-08-29:
I have a branch (make_msp430_use_newlib) that completely updates msp430 to use current gcc and shares newlib with arm. The only issue I had (with the ARCH linux msp430 gcc) was that newlib needed to be recompiled to use the same compilation flags regarding syscalls as ARM, and I didn't get to provide correctly compiled toolchain binaries...

Will check if the official TI binaries use the correct settings.

jcarrano, 2019-08-29:
>  same compilation flags regarding syscalls as ARM

Which flags?

kaspar030, 2019-08-29:
> Which flags?

IIRC these had to be set:

```
    --disable-newlib-supplied-syscalls
    --enable-newlib-reent-small
```

I just tried, my branch seems to work fine with the TI toolchain. I also tried re-basing, but the latest changes to msp430 malloc & heap handling break it again. Will investigate.

closed 2020-04-01 by stale[bot]