RIOT-OS/RIOT #12125: i2c_scan not working on ESP32

maribu, 2019-08-30:
#### Description

The `i2c_scan` application is not working correctly: For non existent addresses the application shows `E` instead of `-`. This indicates that the I2C driver does not correctly return `-ENXIO` for "address not acknowledged", but returns a different error instead.

#### Steps to reproduce the issue

Run `USEMODULE=i2c_scan BOARD=<ESP32-BOARD> make flash term -C examples/default` (replace `<ESP32-BOARD>` with your ESP32 board) to flash the default app including the `i2c_scan`utility and enter the shell. Type `i2c_scan 0` to perform an I2C scan.

#### Expected results

The app should give an output similar to this:

```
> i2c_scan 0
Scanning I2C device 0...
Legend: no device = "-", device responded = "X", error = "E"
     0 1 2 3 4 5 6 7 8 9 a b c d e f
0x00 - - - - - - - - - - - - - - - -
0x10 - - - - - - - - - - - - - - - -
0x20 - - - - - - - - - - - - - - - -
0x30 - - - - - - - - - - - - - - - -
0x40 - - - - X - - - - - - - - - - -
0x50 X - - - - - - - - - - - - - - -
0x60 - - - - - - - - - - - - - - - -
0x70 - - - - - - X - - - - - - - - -
```

#### Actual results

```
i2c_scan 0
2019-08-30 13:17:06,514 - INFO #  i2c_scan 0
2019-08-30 13:17:06,516 - INFO # Scanning I2C device 0...
2019-08-30 13:17:06,519 - INFO # addr not ack'ed = "-", addr ack'ed = "X", addr reserved = "R", error = "E"
2019-08-30 13:17:06,520 - INFO #      0 1 2 3 4 5 6 7 8 9 a b c d e f
2019-08-30 13:17:06,521 - INFO # 0x00 R R R R R R R R R R R R R R E E
2019-08-30 13:17:06,521 - INFO # 0x10 E E E E E E E E E E E E E E E E
2019-08-30 13:17:06,522 - INFO # 0x20 E E E E E E E E E E E E E E E E
2019-08-30 13:17:06,530 - INFO # 0x30 E E E E E E E E E E E E E E E E
2019-08-30 13:17:06,531 - INFO # 0x40 E E E E E E E E E E E E E E E E
2019-08-30 13:17:06,532 - INFO # 0x50 X E E E E E E E E E E E E E E E
2019-08-30 13:17:06,532 - INFO # 0x60 E E E E E E E E E E E E E E E E
2019-08-30 13:17:06,533 - INFO # 0x70 E E E E E E E E R R R R R R R R
```

#### Versions

Current RIOT master.

maribu, 2019-08-30:
@gschorcht:

Maybe the issue is here:

https://github.com/RIOT-OS/RIOT/blob/866b126f366bbd3c489d4be187939d7a6e4a8502/cpu/esp32/periph/i2c_hw.c#L291-L309

If I understand the code correctly, in line 302 an error is returned to indicate that no ack was received. The error code for that case should be `-ENXIO`, not `-EIO`.

maribu, 2019-08-30:
OK, wrong file.

https://github.com/RIOT-OS/RIOT/blob/866b126f366bbd3c489d4be187939d7a6e4a8502/cpu/esp32/periph/i2c_sw.c#L657-L678

Replacing the `return !bit ? 0 : -EIO` with `return !bit ? 0 : -ENXIO` resulted in correct output for me. But I think a proper fix would be more involved, as a missing ACK only during the address should result in returning `-ENXIO`, during any other byte `-EIO` should be returned. [i2c_read_bytes() API doc](http://api.riot-os.org/group__drivers__periph__i2c.html#ga196b4508511c41822eddf6b43b008e90)

maribu, 2019-08-30:
I think this fixes the issue for software I2C driver. (But I think the hard driver is also affected.)

```patch
diff --git a/cpu/esp32/periph/i2c_sw.c b/cpu/esp32/periph/i2c_sw.c
index 1209898411..82e201a929 100644
--- a/cpu/esp32/periph/i2c_sw.c
+++ b/cpu/esp32/periph/i2c_sw.c
@@ -225,7 +225,7 @@ int /* IRAM */ i2c_read_bytes(i2c_t dev, uint16_t addr, void *data, size_t len,
                 (res = _i2c_write_byte (bus, addr2)) != 0) {
                 /* abort transfer */
                 _i2c_abort (bus, __func__);
-                return res;
+                return -ENXIO;
             }
         }
         else {
@@ -233,7 +233,7 @@ int /* IRAM */ i2c_read_bytes(i2c_t dev, uint16_t addr, void *data, size_t len,
             if ((res = _i2c_write_byte (bus, (addr << 1 | I2C_READ))) != 0) {
                 /* abort transfer */
                 _i2c_abort (bus, __func__);
-                return res;
+                return -ENXIO;
             }
         }
     }
@@ -286,7 +286,7 @@ int /* IRAM */ i2c_write_bytes(i2c_t dev, uint16_t addr, const void *data, size_
                 (res = _i2c_write_byte (bus, addr2)) != 0) {
                 /* abort transfer */
                 _i2c_abort (bus, __func__);
-                return res;
+                return -ENXIO;
             }
         }
         else {
@@ -294,7 +294,7 @@ int /* IRAM */ i2c_write_bytes(i2c_t dev, uint16_t addr, const void *data, size_
             if ((res = _i2c_write_byte (bus, addr << 1)) != 0) {
                 /* abort transfer */
                 _i2c_abort (bus, __func__);
-                return res;
+                return -ENXIO;
             }
         }
     }
```

gschorcht, 2019-08-30:
@maribu Thanks for figuring out. Indeed, it is the same for the ESP32 hardware driver as well as for ESP8266. I will provide a fix for all of them.

gschorcht, 2019-08-30:
@maribu Hm, it seems to be a bit more complicated for ESP32 hardware implementation.

In ESP32 hardware implementation, a command pipeline is filled with START, WRITE, READ and STOP commands to prepare a complete transfer from START to STOP condition. Once the command pipeline is filled, it is executed to realize the transfer.

After a complete transfer, a status register contains the errors that happened during the execution of the complete transfer. However, it is not possible to determine for which byte a certain error occurred. For I2C read access it is clear that an ACK error can only happen for the address byte. But for I2C write access, the address byte is just one byte of a number of sent bytes. We can only say that an ACK error happened, but not whether for the address byte or any data byte.

What are the options?

1. We distinguish two cases:
   - For `i2c_read_bytes`, we return `-ENXIO` for ACK errors.
   - For `i2c_write_bytes`, we return `-EIO` for ACK errors.

2. We return always `-ENXIO` for ACK errors.

What do you think would be the best from your point of view. From my point of view, option 1 is most conformant with the API doc.

maribu, 2019-08-30:
The command pipeline has to be filled with the whole sequence, right? There is no possibility to fill it up to sending the address, and than just in time add more commands?

Or can the information be extracted from a side channel? E.g. maybe the I2C command pipeline has not been cleared and the size of the remaining commands in the pipe at failure can be accessed somehow?

gschorcht, 2019-08-31:
> The command pipeline has to be filled with the whole sequence, right? There is no possibility to fill it up to sending the address, and than just in time add more commands?

I tried to split the sequence in START+ADDR and DATA+STOP phase. Unfortunatly, it doesn't work in that way.

> Or can the information be extracted from a side channel? E.g. maybe the I2C command pipeline has not been cleared and the size of the remaining commands in the pipe at failure can be accessed somehow?

Unfortunately, the documentation for hardware I2C is really poor.

But, I think there could be a workaround. An ACK error is indicated by an interrupt. If the FIFO is used (what is the case by default), there is a FIFO TX counter which holds the number of remaining bytes to be sent. For `i2c_read_bytes` an ACK error can only happen for the address. For `i2c_write_bytes` the ACK error would be in address phase when this FIFO TX counter corresponds to the number of data bytes, or the number of data bytes + 1 for 10 bit addresses. I will try this approach

benpicco, 2019-09-10:
This should be fixed now.

closed 2019-09-10 by maribu