RIOT-OS/RIOT #12137: tests/driver_sx127x: missing initializer for member 'iolist::iol_next'?

mrusme, 2019-09-01:
#### Description

Copied the following piece of code from the test:

```
    iolist_t iolist = {
        .iol_base = argv[1],
        .iol_len = (strlen(argv[1]) + 1)
    };
```

Getting the following error:

```
error: missing initializer for member 'iolist::iol_next' [-Werror=missing-field-initializers]
     };
     ^
```

Either test is not implemented correctly or something else is odd here. However, unfortunately I couldn't find a clear documentation on the iolist approach. Other implementations do specify the `iol_next` parameter.

#### Versions

Current master.

miri64, 2019-09-10:
AFAIK, dou can't use functions. in a static initializer.

miri64, 2019-09-10:
If my suspicion is correct, you need to do it this way:

```C
    iolist_t iolist = {
        .iol_base = argv[1],
    };
    iolist.iol_len = strlen(argv[1]) + 1;
```

However, since that includes a hidden `memset(&iolist, 0, ...);`, I'd suggest you do it like this:

```C
    iolist_t iolist;

    iolist.iol_base = argv[1]
    iolist.iol_len = strlen(argv[1]) + 1;
```

miri64, 2019-09-10:
> Other implementations do specify the `iol_next` parameter.

Don't confuse `iolist_t` with `struct iovec` ;-)! They are similar, but different types.

mrusme, 2019-09-10:
Thanks! Turns out it was possible by altering the order:

```
    iolist_t iolist = {
        .iol_next = NULL,
        .iol_base = payload,
        .iol_len = length
    };
```

🤷‍♂

closed 2019-09-10 by mrusme