RIOT-OS/RIOT #12139: boards/nucleo-l412kb

mrusme, 2019-09-01:
#### Description

While boards like the L031K6 and the L432KC are being supported (👍) unfortunately the "in between" board L412KB is not. Maybe someone fluent with Nucleo boards could implement it or alternatively pair up with me (have the hardware ready) in order to implement it?

### Useful links

[https://www.st.com/en/evaluation-tools/nucleo-l412kb.html](https://www.st.com/en/evaluation-tools/nucleo-l412kb.html)

aabadie, 2019-09-02:
Indeed ST released quite a few new nucleo boards recently. There's a tracking issue about this already: #6629. I'll update it with this one and maybe we can close this issue when it's done ?

aabadie, 2019-09-02:
@mrusme, if you have the hardware then you can try to port this board. I can help you if you have problems.
You will first have to add support for the CPU: add the CMSIS header in `cpu/stm32l4/include/vendor` (you can get it from the STM32 Cube) and update the vectors.c file in cpu/stm32l4.
Then add the new board and configure it in the `boards` directory.

mrusme, 2019-09-02:
@aabadie thanks for the support! 👍 It looks like it's a little bit more that needs to be done for that specific board. Maybe you can have a look at #12144 ?

aabadie, 2020-03-05:
I can test this when time permits.

aabadie, 2020-05-26:
Support for this board was added in #13660. Closing.

closed 2020-05-26 by aabadie