RIOT-OS/RIOT #12143: There are unsatisfied feature requirements: periph_i2

JakeSCahill, 2019-09-02:
#### Description

Cannot compile code that uses I2C on nrg52832-mdk board

#### Steps to reproduce the issue

I'm using a Linux VirtualBox 6.0.10 with an nrf52832 MDK. I have Python3, PIP, pyOCD, openOCD, ARM toolchain.
1. `git clone https://github.com/Citrullin/RIOT.git`
2. `git checkout ble_sensor_node`
3 `cd examples/saul`
4. Add `USEMODULE bmx280` to the Makefile
5. `BOARD=nrf52832-mdk PORT=/dev/ttyUSB0 make flash term`

#### Expected results

The application compiles

#### Actual results
There are unsatisfied feature requirements: periph_i2c

EXPECT ERRORS!

Building application "saul_example" for "nrf52832-mdk" with MCU "nrf52".

"make" -C /home/jake/BLE-environment-sensor/boards/nrf52832-mdk
"make" -C /home/jake/BLE-environment-sensor/core
"make" -C /home/jake/BLE-environment-sensor/cpu/nrf52
"make" -C /home/jake/BLE-environment-sensor/cpu/cortexm_common
"make" -C /home/jake/BLE-environment-sensor/cpu/cortexm_common/periph
"make" -C /home/jake/BLE-environment-sensor/cpu/nrf52/periph
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c:45:22: error: 'I2C_NUMOF' undeclared here (not in a function); did you mean 'ADC_NUMOF'?
 static mutex_t locks[I2C_NUMOF];
                      ^~~~~~~~~
                      ADC_NUMOF
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c: In function 'bus':
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c:50:12: error: 'i2c_config' undeclared (first use in this function); did you mean 'i2c_conf_t'?
     return i2c_config[0].dev;
            ^~~~~~~~~~
            i2c_conf_t
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c:50:12: note: each undeclared identifier is reported only once for each function it appears in
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c: In function 'i2c_init':
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c:92:15: error: 'i2c_config' undeclared (first use in this function); did you mean 'i2c_conf_t'?
     gpio_init(i2c_config[dev].scl, GPIO_IN_PU);
               ^~~~~~~~~~
               i2c_conf_t
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c: In function 'bus':
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c:51:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
At top level:
/home/jake/BLE-environment-sensor/cpu/nrf52/periph/i2c.c:45:16: error: 'locks' defined but not used [-Werror=unused-variable]
 static mutex_t locks[I2C_NUMOF];
                ^~~~~
cc1: all warnings being treated as errors
/home/jake/BLE-environment-sensor/Makefile.base:83: recipe for target '/home/jake/BLE-environment-sensor/examples/saul/bin/nrf52832-mdk/periph/i2c.o' failed
make[3]: *** [/home/jake/BLE-environment-sensor/examples/saul/bin/nrf52832-mdk/periph/i2c.o] Error 1
/home/jake/BLE-environment-sensor/Makefile.base:20: recipe for target 'ALL--/home/jake/BLE-environment-sensor/cpu/nrf52/periph' failed
make[2]: *** [ALL--/home/jake/BLE-environment-sensor/cpu/nrf52/periph] Error 2
/home/jake/BLE-environment-sensor/Makefile.base:20: recipe for target 'ALL--/home/jake/BLE-environment-sensor/cpu/nrf52' failed
make[1]: *** [ALL--/home/jake/BLE-environment-sensor/cpu/nrf52] Error 2
/home/jake/BLE-environment-sensor/examples/saul/../../Makefile.include:481: recipe for target '/home/jake/BLE-environment-sensor/examples/saul/bin/nrf52832-mdk/application_saul_example.a' failed
make: *** [/home/jake/BLE-environment-sensor/examples/saul/bin/nrf52832-mdk/application_saul_example.a] Error 2

#### Versions

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

aabadie, 2019-09-02:
This is not a bug but a missing feature: there's no I2C configuration for this board. It should be rather simple to add it though.

aabadie, 2019-09-02:
@JakeSCahill can you try with #12148 and tell if this fixes your problem ?

JakeSCahill, 2019-09-02:
Thanks @aabadie ! It's working with your fix 😄

PR #12148 (2019-09-03): boards/nrf52832-mdk: enable I2C and use default configuration

### Contribution description

This PR enables I2C on nrf52832-mdk and use the I2C default configuration for nrf boards. The pinout of the nrf52832-mdk is the same as the nrf52840-mdk so we can do that.

### Testing procedure

- Plug an I2C sensor on P26/27 and verify that it works.

### Issues/PRs references

fixes #12143
closed 2019-09-03 by fjmolinas