RIOT-OS/RIOT #12168: pkg/libb2: blake2s doesn't work on AVR

gschorcht, 2019-09-04:
#### Description

Function `blake2s` returns a wrong hash value on AVR architecture. The problem was found when testing PR #12135 

#### Steps to reproduce the issue

Flash `tests/pkg_libb2` on an Arduino Mega2560 board and test it.
```
make BOARD=arduino-mega2560 -C tests/pkg_libb2 PORT=/dev/ttyUSB0 flash term
```

#### Expected results

The test should succeed with following output:
```
This is RIOT! (Version: 2019.10-devel-556-g93fa4)
..
OK (2 tests)
```

#### Actual results

The test fails with following output:
```
This is RIOT! (Version: 2019.10-devel-556-g93fa4)
.
2019-09-04 14:19:43,660 - INFO # blake2_tests.test_blake2s (tests/pkg_libb2/main.c 56) memcmp(b2s, hash, sizeof hash) == 0
.
```

aabadie, 2020-07-02:
I can reproduce the problem on atmega256rfr2-xpro. Compared to other archiectures (native, ARM, RISCV) the generated firmware of the test application is insanely big and it seems the test is crashing (I get the prompt for starting the test after the failure):

<details><summary>atmega256rfr2-xpro</summary>

```
make BOARD=atmega256rfr2-xpro -C tests/pkg_libb2
make: Entering directory '/work/riot/RIOT/tests/pkg_libb2'
Building application "tests_pkg_libb2" for "atmega256rfr2-xpro" with MCU "atmega256rfr2".

[INFO] updating libb2 /work/riot/RIOT/build/pkg/libb2/.pkg-state.git-downloaded
echo 60ea749837362c226e8501718f505ab138e5c19d  > /work/riot/RIOT/build/pkg/libb2/.pkg-state.git-downloaded
[INFO] patch libb2
"make" -C /work/riot/RIOT/pkg/libb2
"make" -C /work/riot/RIOT/build/pkg/libb2/src -f /work/riot/RIOT/pkg/libb2/Makefile.libb2
"make" -C /work/riot/RIOT/boards/atmega256rfr2-xpro
"make" -C /work/riot/RIOT/boards/common/atmega
"make" -C /work/riot/RIOT/core
"make" -C /work/riot/RIOT/cpu/atmega256rfr2
"make" -C /work/riot/RIOT/cpu/atmega_common
"make" -C /work/riot/RIOT/cpu/atmega_common/avr_libc_extra
"make" -C /work/riot/RIOT/cpu/atmega_common/periph
"make" -C /work/riot/RIOT/drivers
"make" -C /work/riot/RIOT/drivers/periph_common
"make" -C /work/riot/RIOT/sys
"make" -C /work/riot/RIOT/sys/auto_init
"make" -C /work/riot/RIOT/sys/embunit
"make" -C /work/riot/RIOT/sys/isrpipe
"make" -C /work/riot/RIOT/sys/stdio_uart
"make" -C /work/riot/RIOT/sys/test_utils/interactive_sync
"make" -C /work/riot/RIOT/sys/tsrb
   text	   data	    bss	    dec	    hex	filename
 218874	    840	    979	 220693	  35e15	/work/riot/RIOT/tests/pkg_libb2/bin/atmega256rfr2-xpro/tests_pkg_libb2.elf
make: Leaving directory '/work/riot/RIOT/tests/pkg_libb2'
```

</details>

<details><summary>stm32f429i-disc1</summary>

```
make BOARD=stm32f429i-disc1 -C tests/pkg_libb2
make: Entering directory '/work/riot/RIOT/tests/pkg_libb2'
Building application "tests_pkg_libb2" for "stm32f429i-disc1" with MCU "stm32".

"make" -C /work/riot/RIOT/pkg/libb2
"make" -C /work/riot/RIOT/build/pkg/libb2/src -f /work/riot/RIOT/pkg/libb2/Makefile.libb2
"make" -C /work/riot/RIOT/boards/stm32f429i-disc1
"make" -C /work/riot/RIOT/core
"make" -C /work/riot/RIOT/cpu/stm32
"make" -C /work/riot/RIOT/cpu/cortexm_common
"make" -C /work/riot/RIOT/cpu/cortexm_common/periph
"make" -C /work/riot/RIOT/cpu/stm32/bootloader
"make" -C /work/riot/RIOT/cpu/stm32/periph
"make" -C /work/riot/RIOT/cpu/stm32/stmclk
"make" -C /work/riot/RIOT/cpu/stm32/vectors
"make" -C /work/riot/RIOT/drivers
"make" -C /work/riot/RIOT/drivers/periph_common
"make" -C /work/riot/RIOT/sys
"make" -C /work/riot/RIOT/sys/auto_init
"make" -C /work/riot/RIOT/sys/embunit
"make" -C /work/riot/RIOT/sys/isrpipe
"make" -C /work/riot/RIOT/sys/newlib_syscalls_default
"make" -C /work/riot/RIOT/sys/pm_layered
"make" -C /work/riot/RIOT/sys/stdio_uart
"make" -C /work/riot/RIOT/sys/test_utils/interactive_sync
"make" -C /work/riot/RIOT/sys/tsrb
   text	   data	    bss	    dec	    hex	filename
  32352	    140	   2388	  34880	   8840	/work/riot/RIOT/tests/pkg_libb2/bin/stm32f429i-disc1/tests_pkg_libb2.elf
make: Leaving directory '/work/riot/RIOT/tests/pkg_libb2'
```
[... 65 lines cut ...]
"make" -C /data/riotbuild/riotbase/cpu/esp_common/vendor/xtensa
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/periph_common
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/auto_init
"make" -C /data/riotbuild/riotbase/sys/embunit
"make" -C /data/riotbuild/riotbase/sys/isrpipe
"make" -C /data/riotbuild/riotbase/sys/log
"make" -C /data/riotbuild/riotbase/sys/luid
"make" -C /data/riotbuild/riotbase/sys/newlib_syscalls_default
"make" -C /data/riotbuild/riotbase/sys/pm_layered
"make" -C /data/riotbuild/riotbase/sys/random
"make" -C /data/riotbuild/riotbase/sys/random/tinymt32
"make" -C /data/riotbuild/riotbase/sys/stdio_uart
"make" -C /data/riotbuild/riotbase/sys/test_utils/interactive_sync
"make" -C /data/riotbuild/riotbase/sys/tsrb
   text	   data	    bss	    dec	    hex	filename
  88155	   3656	   6408	  98219	  17fab	/data/riotbuild/riotbase/tests/pkg_libb2/bin/esp32-wroom-32/tests_pkg_libb2.elf
make: Leaving directory '/work/riot/RIOT/tests/pkg_libb2'
```

</details>

AVR: +200kB on flash, other architectures between 30kB and 100kB. I always thought that firmware generated for 8bit architectures should be smaller.
So I suspect that libb2 is just not compatible with 8bit architectures. Would you agree to simply blacklist `arch_8bit` with this package ? If yes, I can open a PR for that (and also an update of the package).

chudov, 2021-02-19:
Seems this issue shall be closed as expected behaviour due to #15513

closed 2021-02-19 by aabadie