RIOT-OS/RIOT #12179: "sorry, unimplemented: non-trivial designated initializers not supported"

mrusme, 2019-09-07:
Including a C `USEMODULE` into a C++ (`FEATURES_REQUIRED += cpp`) project which contains of a header file that uses this kind of array initialisers doesn't compile:

```c
#ifdef __cplusplus
extern "C" {
#endif

// ...

enum ENUM_VALUES {
  ENUM_VAL_1,
  ENUM_VAL_2,
};
typedef enum ENUM_VALUES ENUM_VALUES;

static const uint8_t BLA[256][8] = {
  [ENUM_VAL_1] = { 0, 0, 0, 0, 0, 0, 0, 0 },
  [ENUM_VAL_2] = { 0, 0, 0, 0, 0, 0, 0, 0 },
  // ...
};

// ...

#ifdef __cplusplus
}
#endif
```

`make` output is: `sorry, unimplemented: non-trivial designated initializers not supported`.

However, a `test.cpp` program which includes the specific header and accesses its values can be compiled and run without any issues using `g++ ./test.cpp -o ./test`.

mrusme, 2019-09-07:
Some more info:

```
arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027] 
```

When using the dedicated compiler instead of the native one, I get the same issue. However, by using `arm-none-eabi-gcc ./test.c -o ./test` instead of the g++ it works.

It looks like the arm-none-eabi-g++ does not (yet?) support this, but `arm-none-eabi-gcc` has no issues.

kaspar030, 2019-09-07:
The provided example doesn't have that issue, but I stumbled across that issue with our periph_conf.h initializers. g++ needs the initializers ordered as in the type declaration.

mrusme, 2019-09-07:
@kaspar030 sorry, my example was probably over-simplified, here's what I'm actually doing:

```c
static uint8_t BLA[256][256][8] = {
    [(ENUM_VAL_1 | ENUM_OPT_1)] = {
        [(ENUM_ATTR_1 | ENUM_ATTR_OPT_1)] = { 8, 0, 0, 0, 0, 0, 0, 0 },
    },
// ...
};
```

mrusme, 2019-09-07:
However, as mentioned, it looks like if order is kept, this issue can be worked around. Thanks for the hint!

closed 2019-09-07 by mrusme