RIOT-OS/RIOT #12189: board suggestion: nRF52840-dongle

chrysn, 2019-09-10:
#### Description

The nRF52840-dongle is similar to the already supported nRF5238-MDKv2 in that it offers the plug-the-PCB-into-the-USB-port form factor -- but with the advantages of being shorter and cheaper (10€ rather than 25€), and being SMD solderable (2.54mm spaced half-holes at the outline)

It offers direct chip access to the USB port at the cost of relying on the on-chip bootloader (no programmer in-place) -- not sure if that can be bricked, but if so, you're back to a 10-pin tag connect programmer, populating a 1.27mm port on the backside, or possibly using the SWDIO/SWCLK pins on the USB collar.

Most of its features it inherits from the nRF52840 -- BLE/802.15.4/... radio and the usual on-chip peripherals; the board features two LEDs (green, RGB) and a two buttons (reset, user) and is has 32MHz and 32768kHz oscillators.

### Useful links
* [User guide](https://infocenter.nordicsemi.com/pdf/nRF52840_Dongle_User_Guide_v1.1.pdf)
* [nRF website](https://www.nordicsemi.com/?sc_itemid=%7BCDCCA013-FE4C-4655-B20C-1557AB6568C9%7D)

benemorius, 2019-09-10:
I was just looking at this too.

It looks like we can use the pre-programmed USB bootloader with Nordic's `nrfutil` but the usage is a bit nonobvious mostly because it supports signing and the documentation isn't focused on command-line flashing. I found [this flashing guide](http://www.d0d0.pro/index.php/2018/09/23/programming-the-new-nordic-chip-nrf52840-and-usb-dongle-board-pca10059-featuring-it/) that makes quick work of it but I have yet to install python2 so I can try it out.

Also here's a [post](https://www.rototron.info/circuitpython-nrf52840-dongle-openocd-pi-tutorial/) on using openocd with it that includes some good information about things you might encounter when wiping the bootloader.

It's my understanding that one could easily backup the bootloader with openocd and restore it later but I haven't confirmed that.

chrysn, 2019-09-13:
The abovementioned branch can now be used to run examples like nimble_gatt or usbus_minimal, and `make flash` orchestrates all the required `nrfutil` incantations.

Unless someone tries this out and places more trust in it than I do, I'll leave this sitting around until #11085 can be used to work with its console. With that, I can get a better impression of whether things work as they should, and prep it for an actual PR.

A few questions popped up so far that may help get this into better shape even before it enters the PR phase:

* Is there a checklist of things a new board should be able to do, or general "new board" tips?
* What's the naming guidance between the vendor's marketing name ("nRF52840-Dongle") vs. its PCB identifier ("PCA10059")?
* The nRF chips appear to be quite flexible in their pinout, and as there're neither on-PCB devices that use SPI or I2C nor designated headers for use with those protocols. Is there any point in this board providing the feature `periph_i2c` with arbitrary pins like those in `boards/common/nrf52/include/cfg_spi_default.h`? Any application of those would need to redefine them anyway, pratically creating a fork/copy of the board with an explicit peripheral device for whichever purpose.
* Due to the flashing tool / bootloader involved, there is currently a bespoke linker script and a number of custom definitions in `Makefile.include` that orchestrate the use of the flasher. Does that need to be generalized into its own tool makefiles or can that stay here until something else (hint: Thingy:52 OTA bootloader) uses it?

maz3max, 2019-09-19:
About bricking the bootloader: You can use the bootloader to overwrite itself within nRF connect. Tested for you.
Maybe one could use MCUboot for this like [zephyr](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10059/doc/index.html?highlight=pca10059) can.

chrysn, 2019-09-19:
> ou can use the bootloader to overwrite itself within nRF connect.

That does have its upsides: A more versatile bootloader could be installed on boards without ever having to solder programmer headers to the board or owning a (10-pin ... why?) tag connect probe.

With a linker script in place that leaves the shipped bootloader intact, I think it's a good starting point for a default board configuration. Having a link to a tutorial on how to restore the nrf bootloader would be a big plus. (You'd be in the ideal position to explore that route ;-) )

chrysn, 2019-09-19:
With #11085 getting in good shape, I've added a [branch](https://github.com/chrysn-pull-requests/RIOT/tree/cdcacm-on-nrf52840-dongle-4) called "cdcacm-on-nrf52840-dongle-4" that merges current master, #12269 and #11085 in that sequence to get a setup for practically usable nRF52840-dongle board.

On top of that, I've added a bit of Makefile restructuring for common USB stuff (loosely related to #12273), and board file extensions that make USB the default serial and enable an Ethernet interface if `gnrc_netdev_default` is picked. With that, examples like saul run out of the box (`make flash` and `make term` still need to be called separately pending a proper solution to #12270). Examples like `gcoap` need the manual additon of the `gnrc_sixlowpan_border_router_default` module to work around #12002, or need their IPv6 addresses configured manually.

From here, I intend to assemble a minimal board support PR (without any USB automagic at first), to which later (when #11085 is merged) an auto serial console can be added in a separate PR.

maz3max, 2019-09-19:
This board with USB Serial in RIOT would be awesome!

chrysn, 2019-09-25:
Small update on the programmer side: As the nrfutil family is a bit slow to develop ([still no complete Python 3 port](https://github.com/NordicSemiconductor/pc-nrfutil/issues/246)) and in parts nonfree (see https://github.com/RIOT-OS/RIOT/issues/12189#issuecomment-530809698), I was glad to find that adafruit has [their own version](https://github.com/adafruit/Adafruit_nRF52_nrfutil) which appears to be forked from back when there was a free license to it, and already ported to Python 3. It looks like it only works for serial flashing, but that's all I care about for now (with the Bluetooth DFU still being interesting for the Thingy:52).

I'll look into that more deeply, possibly they can both be used as pluggable executables in the nrfutil programmer make scripts.

chrysn, 2019-10-01:
The adafruit-nrfutil is unfortunately very far from the current nrfutil (package generation differs), so I'm abandoning that for now. (a `FLASHER = $(shell if which adafruit-nrfutil > /dev/null; then echo adafruit-nrfutil; else echo nrfutil; fi)` is far from enough, and just `dfu genpkg` instead of `pkg generate --hw-version 52` doesn't cut it; it seems it can't even communicate with the serial port, "possible causes: wrong bootloader version, wrong baud rate, target not in DFU mode").

With the merging of #11085, I moved over all work from the cdcacm-on-nrf52840 branch back to the nrf52840-dongle branch, which now has a single merge in from master instead of several historic merges, and has CDC-ACM and Ethernet coming with the shell and auto-netif modules, respectively.

In terms of next steps, I'd like to prepare this for PRs. Any preferences on whether this should be a single one, or an "introduce the board" (which should be uncontroversial except for the custom flasher) PR and a "make the board pull in USB stuff by default because USB is its flasher" PR (which may get hung up in the VID/PID discussion)?

miri64, 2020-01-09:
Thinking this board is already supported, I bought this over the holidays. If anyone wants to port this, I'd be happy to test (or if I find the time, try my hands on my first board port :D)

miri64, 2020-01-09:
> It's far from working or being a pull request, but whoever 's curious or wants to join in can peek at my https://github.com/chrysn-pull-requests/RIOT/tree/nrf52840-dongle branch.

Ah, there is already a port :-).

miri64, 2020-01-09:
What is missing to make it a PR, @chrysn?

chrysn, 2020-01-09:
The main stopper was that I wanted to have a good story on USB PID/VID as those'd be required to run practically any example. With [my request at pid.codes](https://github.com/pidcodes/pidcodes.github.com/pull/468) accepted a few days ago, https://github.com/RIOT-OS/RIOT/issues/12273 is back in my court, and I just need a bit of time to tidy things up.

chrysn, 2020-01-09:
The other Open Question is as to how far the nrfutil steps should be abstracted, but that can be fleshed out in the PR process.

trickeydan, 2020-01-16:
Hi,

I'd be willing to contribute towards adding compatibility for this board.

chrysn, 2020-01-16:
I've just rebased my branch and pushed as https://github.com/chrysn-pull-requests/RIOT/tree/nrf52840-dongle-2 to circumvent some compatibility issues with current master that have arisen. PR will follow soon, I'd just like to give the current state a bit of testing.

closed 2020-05-08 by aabadie